JSBSim-Team/jsbsim #224: Compilation of jsbsim python packages

galleon, 2019-10-15:
All,

I have a question on the jsbsim build process. I recently find out this package on Pypi (https://pypi.org/project/JSBSim/#files), it seems to be different from what is proposed in this github repo. Is there any connection between both of them (beyond the maintainers ;-)) ?

Cheers
Guillaume

bcoconni, 2019-10-26:
Hi Guillaume,

You are correct: the package you mentioned from PyPI is indeed produced by our project. This is a version that is currently not maintained hence the differences you might have observed between the GitHub project and the package in PyPI.

At some point, I plan to push automatically the wheel packages produced by Travis CI and AppVeyor to PyPI but I cannot tell you when this will happen. It is yet another item on my (very long) TODO list.

In the meantime, you can download the binary Python wheel packages from the [Rolling Release v2019](https://github.com/JSBSim-Team/jsbsim/releases/tag/Rolling-release-v2019).

bcoconni, 2019-11-16:
@galleon Do you need further information ? If not, can this issue be closed ?

galleon, 2019-11-18:
Yes the issue can be closed. Thanks @bcoconni

closed 2019-11-18 by galleon