JSBSim-Team/jsbsim #225: JSBSIM build problem

Oleksandr1977, 2019-10-17:
Hello, JSBSIM-TEAM!

I am trying to build Jsbsim under cygwin64 platform by following instructions https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-the-program/
But I got error:

$ make
[  1%] Building CXX object src/simgear/io/iostreams/CMakeFiles/IOStreams.dir/sgstream.cxx.o
/home/Igor/jsbsim/src/simgear/io/iostreams/sgstream.cxx: In constructor ‘sg_ifstream::sg_ifstream(const SGPath&, std::ios_base::openmode)’:
/home/Igor/jsbsim/src/simgear/io/iostreams/sgstream.cxx:39:41: error: no matching function for call to ‘sg_ifstream::open(const wchar_t*, std::ios_base::openmode&)’

bcoconni, 2019-10-26:
Could you please specify the following items:
* Where did you get the source code from ?
* What are the commands you issued before `make` ?

Oleksandr1977, 2019-10-26:
I issued the following commands:

git clone git://github.com/JSBSim-Team/jsbsim.git
cd jsbsim
mkdir build
cd build
cmake -DCMAKE_CXX_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_C_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_BUILD_TYPE=Release ..
make

bcoconni, 2019-10-31:
OK thanks.

I'm a bit puzzled by the error message which reports that `sg_ifstream::open` is called at line 39 in `sgstream.cxx` while it should not be the case (`std::ifstream::open` is explicitly called).

Could you please check that the code in `/home/Igor/jsbsim/src/simgear/io/iostreams/sgstream.cxx` is identical to the code below (especially at line 39) ?
https://github.com/JSBSim-Team/jsbsim/blob/4840a621dafe1ce64970990ddaec3e86a1c7e082/src/simgear/io/iostreams/sgstream.cxx#L32-L40

Oleksandr1977, 2019-11-01:
bcoconni, thank you for the response.
I compared. The files are identical.

rega0051, 2019-11-20:
I get the a similar error trying to compile with mingw on Windows.

error: no matching function for call to 'sg_ifstream::open(const wchar_t*, std::ios_base::openmode&)'                                                                                       std::ifstream::open(ps.c_str(), io_mode);

bcoconni, 2019-11-27:
OK that's weird. At least there is a common element : this happens on the Windows platform but for 2 different compilers (cygwin and mingw).
At some point, we shall check the compilation with these 2 compilers on AppVeyor. Yet another item on the TODO list.

bcoconni, 2019-11-28:
@Oleksandr1977 could you please try building with the following commands from your `build`directory ?
```bash
cmake -G "MinGW Makefiles" ..
mingw32-make
```
If it does not work, could you please report here the output of the `cmake` command ?

rega0051, 2019-11-30:
Sorry the format is fun...

PS D:\chris\Simulation\JSBSim\build-mingw> cmake -G "MinGW Makefiles" ..\jsbsim-repo\
-- The C compiler identification is GNU 8.1.0
-- The CXX compiler identification is GNU 8.1.0
-- Check for working C compiler: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/bin/gcc.exe
-- Check for working C compiler: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/bin/g++.exe
-- Check for working CXX compiler: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Found PythonInterp: C:/Users/Chris/Anaconda3/python.exe (found version "3.7.3")
-- Could NOT find Cython (missing: CYTHON_EXECUTABLE)
CMake Warning at CMakeLists.txt:92 (message):
  JSBSim Python module and test suite will not be built

-- Could NOT find CxxTest (missing: CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: D:/chris/Simulation/JSBSim/build-mingw
PS D:\chris\Simulation\JSBSim\build-mingw>
PS D:\chris\Simulation\JSBSim\build-mingw>
PS D:\chris\Simulation\JSBSim\build-mingw> mingw32-make.exe
Scanning dependencies of target IOStreams
[  0%] Building CXX object src/simgear/io/iostreams/CMakeFiles/IOStreams.dir/sgstream.cxx.obj
D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.cxx: In constructor 'sg_ifstream::sg_ifstream(const SGPath&, std::ios_base::openmode)':
D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.cxx:39:41: error: no matching function for call to 'sg_ifstream::open(const wchar_t*, std::ios_base::openmode&)'
  std::ifstream::open(ps.c_str(), io_mode);
                                         ^
In file included from D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.hxx:36,
                 from D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.cxx:25:
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:625:7: note: candidate: 'void std::basic_ifstream<_CharT, _Traits>::open(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]'
       open(const char* __s, ios_base::openmode __mode = ios_base::in)
       ^~~~
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:625:7: note:   no known conversion for argument 1 from 'const wchar_t*' to 'const char*'
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:645:7: note: candidate: 'void std::basic_ifstream<_CharT, _Traits>::open(const string&, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::__cxx11::string = std::__cxx11::basic_string<char>; std::ios_base::openmode = std::_Ios_Openmode]'
       open(const std::string& __s, ios_base::openmode __mode = ios_base::in)
       ^~~~
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:645:7: note:   no known conversion for argument 1 from 'const wchar_t*' to 'const string&' {aka 'const std::__cxx11::basic_string<char>&'}
D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.cxx: In member function 'void sg_ifstream::open(const SGPath&, std::ios_base::openmode)':
D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.cxx:49:44: error: no matching function for call to 'sg_ifstream::open(const wchar_t*, std::ios_base::openmode&)'
     std::ifstream::open(ps.c_str(), io_mode);
                                            ^
In file included from D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.hxx:36,
                 from D:\chris\Simulation\JSBSim\jsbsim-repo\src\simgear\io\iostreams\sgstream.cxx:25:
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:625:7: note: candidate: 'void std::basic_ifstream<_CharT, _Traits>::open(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]'
       open(const char* __s, ios_base::openmode __mode = ios_base::in)
       ^~~~
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:625:7: note:   no known conversion for argument 1 from 'const wchar_t*' to 'const char*'
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:645:7: note: candidate: 'void std::basic_ifstream<_CharT, _Traits>::open(const string&, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::__cxx11::string = std::__cxx11::basic_string<char>; std::ios_base::openmode = std::_Ios_Openmode]'
       open(const std::string& __s, ios_base::openmode __mode = ios_base::in)
       ^~~~
[... cut ...]
C:/PROGRA~2/MINGW-~1/I686-8~1.0-P/mingw32/lib/gcc/i686-w64-mingw32/8.1.0/include/c++/fstream:871:7: note:   no known conversion for argument 1 from 'const wchar_t*' to 'const string&' {aka 'const std::__cxx11::basic_string<char>&'}
mingw32-make[2]: *** [src\simgear\io\iostreams\CMakeFiles\IOStreams.dir\build.make:63: src/simgear/io/iostreams/CMakeFiles/IOStreams.dir/sgstream.cxx.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:906: src/simgear/io/iostreams/CMakeFiles/IOStreams.dir/all] Error 2
mingw32-make: *** [Makefile:151: all] Error 2
PS D:\chris\Simulation\JSBSim\build-mingw>

bcoconni, 2019-12-01:
Thanks @rega0051, that's helpful.

It seems the problem is that MingW (and Cygwin) do not have the C++ function `std::ifstream::open(const wchar_t*, std::ios_base::openmode)` ?

I am not familiar myself with MingW, do you how if it can handle `wstring` as Visual C++ does or if we should use `string` as g++ on Linux ?

rega0051, 2019-12-03:
That's a tough question. I know MinGW is built to leverage the WinAPI, Cygwin tries to create a complete Posix environment.  However, g++ STL doesn't handle wchar, so I'm not sure...

I did a quick hack to force the "path.local8BitStr()" definitions and that file compiles. I don't know the full effect of SG_WINDOWS across the system, so I didn't want to mess with the logic switches.

The next issue I ran into was JSBSim.cpp line 282. Needed to execute the: #if defined(_MSC_VER) branch. Again, I hacked a fix just to get it to compile. Not sure on all the details of _MSC_VER, so I didn't want to propose a complete solution.

With those two hacks I get complete compilation. And successfully ran a basic test. But there wasn't anything strange in the file paths, so I probably didn't stress the character set.

bcoconni, 2019-12-03:
> I don't know the full effect of SG_WINDOWS across the system, so I didn't want to mess with the logic switches.

`SG_WINDOWS` is defined in `src/simgear/compiler.h` and it deliberately includes MingW and Cygwin in Windows systems:
https://github.com/JSBSim-Team/jsbsim/blob/7cfffba2a1b4815796ac79fa7402c8ab59e52e34/src/simgear/compiler.h#L40-L49
I understand from your tests that, at least for the code in `sg_stream.cxx` these conditions are not relevant. But since `SG_WINDOWS` is used in some other source files (namely `strutils.cxx` and `sg_path.cxx`), then rather than modifying the definition of `SG_WINDOWS`in `compiler.h`, our best option might be to explicitly exclude MingW and Cygwin from the `#if defined (SG_WINDOWS)` statements of `sg_stream.cxx`.
> The next issue I ran into was JSBSim.cpp line 282. Needed to execute the: #if defined(_MSC_VER) branch. Again, I hacked a fix just to get it to compile.

The same logic should hopefully apply to the code you mentioned in `JSBSim.cpp`: we will have to add new conditions to the statement `#if defined (MSC_VER)`  so that MingW and Cygwin are excluded from that branch of the preprocessor condition.
https://github.com/JSBSim-Team/jsbsim/blob/7cfffba2a1b4815796ac79fa7402c8ab59e52e34/src/JSBSim.cpp#L282-L288
I'll check on the internet which flags should be used to detect MSVC++ and reject MinGW and Cygwin.
> With those two hacks I get complete compilation. And successfully ran a basic test. But there wasn't anything strange in the file paths, so I probably didn't stress the character set.

Thanks, that's good to know. Once the 2 preprocessor conditionals that you mentioned above will be fixed, I hope this issue will be fixed as well.

bcoconni, 2019-12-07:
@rega0051 and @Oleksandr1977

I have just pushed the commit 9b15e85 that should hopefully fix the issues with the compilation of JSBSim with MingW and Cygwin. Could you please test the last revision of the `master` branch and let me know if that fixes the problem for you ?

Oleksandr1977, 2019-12-09:
@bcoconni and @rega005,
Thank you for the help. I am trying to build the latest revision of the master branch. I have some progress.

I didn't more get error:
[ 1%] Building CXX object src/simgear/io/iostreams/CMakeFiles/IOStreams.dir/sgstream.cxx.o
/home/Igor/jsbsim/src/simgear/io/iostreams/sgstream.cxx: In constructor ‘sg_ifstream::sg_ifstream(const SGPath&, std::ios_base::openmode)’:
/home/Igor/jsbsim/src/simgear/io/iostreams/sgstream.cxx:39:41: error: no matching function for call to ‘sg_ifstream::open(const wchar_t*, std::ios_base::openmode&)’

But I got new error:
[ 92%] Built target Aeromatic++
Scanning dependencies of target aeromatic
[ 93%] Building CXX object utils/aeromatic++/CMakeFiles/aeromatic.dir/aeromatic.cpp.o
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx: In member function ‘void SGPath::validate() const’:
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:333:16: error: aggregate ‘SGPath::validate() const::_stat buf’ has incomplete type and cannot be defined
   struct _stat buf ;
                ^~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:340:7: error: ‘_wstat’ was not declared in this scope
   if (_wstat(statPath.c_str(), &buf ) < 0) {
       ^~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:340:7: note: suggested alternative: ‘_stat’
   if (_wstat(statPath.c_str(), &buf ) < 0) {
       ^~~~~~
       _stat
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:346:20: error: aggregate ‘SGPath::validate() const::_stat parentBuf’ has incomplete type and cannot be defined
       struct _stat parentBuf;
                    ^~~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:348:43: error: ‘_S_IWRITE’ was not declared in this scope
           _canWrite = parentBuf.st_mode & _S_IWRITE;
                                           ^~~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:348:43: note: suggested alternative: ‘S_IWRITE’
           _canWrite = parentBuf.st_mode & _S_IWRITE;
                                           ^~~~~~~~~
                                           S_IWRITE
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:358:13: error: ‘_S_IREAD’ was not declared in this scope
  _canRead = _S_IREAD & buf.st_mode;
             ^~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:358:13: note: suggested alternative: ‘S_IREAD’
  _canRead = _S_IREAD & buf.st_mode;
             ^~~~~~~~
             S_IREAD
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:359:14: error: ‘_S_IWRITE’ was not declared in this scope
  _canWrite = _S_IWRITE & buf.st_mode;
              ^~~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:359:14: note: suggested alternative: ‘S_IWRITE’
  _canWrite = _S_IWRITE & buf.st_mode;
              ^~~~~~~~~
              S_IWRITE
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx: In static member function ‘static SGPath SGPath::fromEnv(const char*, const SGPath&)’:
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:551:23: error: ‘_wgetenv’ was not declared in this scope
  const wchar_t* val = _wgetenv(wname.c_str());
                       ^~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:551:23: note: suggested alternative: ‘getenv’
  const wchar_t* val = _wgetenv(wname.c_str());
                       ^~~~~~~~
                       getenv
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx: In static member function ‘static std::vector<SGPath> SGPath::pathsFromEnv(const char*)’:
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:569:23: error: ‘_wgetenv’ was not declared in this scope
  const wchar_t* val = _wgetenv(wname.c_str());
                       ^~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:569:23: note: suggested alternative: ‘getenv’
  const wchar_t* val = _wgetenv(wname.c_str());
                       ^~~~~~~~
                       getenv
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx: In member function ‘SGPath SGPath::realpath() const’:
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:618:50: error: ‘_MAX_PATH’ was not declared in this scope
     wchar_t *buf = _wfullpath( NULL, ws.c_str(), _MAX_PATH );
                                                  ^~~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:618:20: error: ‘_wfullpath’ was not declared in this scope
     wchar_t *buf = _wfullpath( NULL, ws.c_str(), _MAX_PATH );
                    ^~~~~~~~~~
/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx:618:20: note: suggested alternative: ‘realpath’
     wchar_t *buf = _wfullpath( NULL, ws.c_str(), _MAX_PATH );
                    ^~~~~~~~~~
                    realpath

bcoconni, 2019-12-09:
@Oleksandr1977 thanks for the feedback.

Could you please perform a search of every `SG_WINDOWS` in the file `/home/Igor/jsbsim/src/simgear/misc/sg_path.cxx` and replace them by `_MSC_VER` ? Then let me know if the code compiles.

Oleksandr1977, 2019-12-10:
@bcoconni 
I replaced SG_WINDOWS with _MSC_VER and got some progress. The file sg_path.cxx was compiled. 
But I got another error:

Scanning dependencies of target JSBSim
[ 81%] Building CXX object src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o
/home/Igor/jsbsim/src/JSBSim.cpp: In function ‘int main(int, char**)’:
/home/Igor/jsbsim/src/JSBSim.cpp:287:3: error: ‘feenableexcept’ was not declared in this scope
   feenableexcept(FE_DIVBYZERO | FE_INVALID);
   ^~~~~~~~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:287:3: note: suggested alternative: ‘feraiseexcept’
   feenableexcept(FE_DIVBYZERO | FE_INVALID);
   ^~~~~~~~~~~~~~
   feraiseexcept
make[2]: *** [src/CMakeFiles/JSBSim.dir/build.make:63: src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: src/CMakeFiles/JSBSim.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

bcoconni, 2019-12-10:
@Oleksandr1977 
I have just pushed some additional changes to the branch `master` according to your feedback (you might need to reverse the changes you made to the source file `sg_path.cxx`  to successfully update your local git clone).
Could you please test it and let me know if that fixes your issues ?

Oleksandr1977, 2019-12-11:
@bcoconni

I updated my local git clone and got a similar error:

Scanning dependencies of target JSBSim
[ 81%] Building CXX object src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o
/home/Igor/jsbsim/src/JSBSim.cpp: In function ‘int main(int, char**)’:
/home/Igor/jsbsim/src/JSBSim.cpp:283:3: error: ‘_clearfp’ was not declared in this scope
   _clearfp();
   ^~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:283:3: note: suggested alternative: ‘clearerr’
   _clearfp();
   ^~~~~~~~
   clearerr
/home/Igor/jsbsim/src/JSBSim.cpp:284:14: error: ‘_controlfp’ was not declared in this scope
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
              ^~~~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:284:35: error: ‘_EM_INVALID’ was not declared in this scope
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
                                   ^~~~~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:284:35: note: suggested alternative: ‘FE_INVALID’
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
                                   ^~~~~~~~~~~
                                   FE_INVALID
/home/Igor/jsbsim/src/JSBSim.cpp:284:49: error: ‘_EM_ZERODIVIDE’ was not declared in this scope
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
                                                 ^~~~~~~~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:284:66: error: ‘_EM_OVERFLOW’ was not declared in this scope
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
                                                                  ^~~~~~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:284:66: note: suggested alternative: ‘FE_OVERFLOW’
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
                                                                  ^~~~~~~~~~~~
                                                                  FE_OVERFLOW
/home/Igor/jsbsim/src/JSBSim.cpp:285:12: error: ‘_MCW_EM’ was not declared in this scope
            _MCW_EM);
            ^~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:284:3: error: ‘_controlfp’ was not declared in this scope
   _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
   ^~~~~~~~~~
make[2]: *** [src/CMakeFiles/JSBSim.dir/build.make:63: src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: src/CMakeFiles/JSBSim.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

bcoconni, 2019-12-11:
OK after a few searches on the internet, it seems that `__GNU_VISIBLE` must be defined to use `feenableexcept` with Cygwin ([see Cgwin code for `fenv.h`](https://github.com/Alexpux/Cygwin/blob/b61dc22adaf82114eee3edce91cc3433bcd27fe5/winsup/cygwin/include/fenv.h#L156)). So I pushed a new commit a2089b9 to fix that.

Could you please test and let me know if that fixes your remaining problem ?

Oleksandr1977, 2019-12-12:
@bcoconni

Unfortunelly, I got error again:

Scanning dependencies of target JSBSim
[ 81%] Building CXX object src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o
/home/Igor/jsbsim/src/JSBSim.cpp:55:0: warning: "__GNU_VISIBLE" redefined
 #  define __GNU_VISIBLE

In file included from /usr/include/sys/config.h:5:0,
                 from /usr/include/_ansi.h:11,
                 from /usr/include/wchar.h:4,
                 from /usr/lib/gcc/x86_64-pc-cygwin/7.4.0/include/c++/cwchar:44,
                 from /usr/lib/gcc/x86_64-pc-cygwin/7.4.0/include/c++/bits/postypes.h:40,
                 from /usr/lib/gcc/x86_64-pc-cygwin/7.4.0/include/c++/iosfwd:40,
                 from /usr/lib/gcc/x86_64-pc-cygwin/7.4.0/include/c++/memory:72,
                 from /home/Igor/jsbsim/src/FGFDMExec.h:44,
                 from /home/Igor/jsbsim/src/initialization/FGTrim.h:53,
                 from /home/Igor/jsbsim/src/JSBSim.cpp:42:
/usr/include/sys/features.h:261:0: note: this is the location of the previous definition
 #define __GNU_VISIBLE  0

In file included from /usr/lib/gcc/x86_64-pc-cygwin/7.4.0/include/c++/fenv.h:36:0,
                 from /home/Igor/jsbsim/src/JSBSim.cpp:56:
/usr/include/fenv.h:132:18: error: #if with no expression
 #if __GNU_VISIBLE
                  ^
/usr/include/fenv.h:156:18: error: #if with no expression
 #if __GNU_VISIBLE
                  ^
In file included from /home/Igor/jsbsim/src/JSBSim.cpp:67:0:
/usr/include/sys/time.h:373:18: error: #if with no expression
 #if __GNU_VISIBLE
                  ^
/home/Igor/jsbsim/src/JSBSim.cpp: In function ‘int main(int, char**)’:
/home/Igor/jsbsim/src/JSBSim.cpp:288:3: error: ‘feenableexcept’ was not declared in this scope
   feenableexcept(FE_DIVBYZERO | FE_INVALID);
   ^~~~~~~~~~~~~~
/home/Igor/jsbsim/src/JSBSim.cpp:288:3: note: suggested alternative: ‘feraiseexcept’
   feenableexcept(FE_DIVBYZERO | FE_INVALID);
   ^~~~~~~~~~~~~~
   feraiseexcept
make[2]: *** [src/CMakeFiles/JSBSim.dir/build.make:63: src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: src/CMakeFiles/JSBSim.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

bcoconni, 2019-12-13:
Right. According to the error message and to the source file [`/usr/include/sys/time.h`](https://github.com/Alexpux/Cygwin/blob/b61dc22adaf82114eee3edce91cc3433bcd27fe5/newlib/libc/include/sys/features.h#L236), `__GNU_VISIBLE` must be set to 1 for the compilation to succeed.

Hopefully, this will work this time. Let me know.

Oleksandr1977, 2019-12-13:
@bcoconni
Wonderful.  Everything compiles and builds without errors under for Cygwin platform.  Many thanks.

bcoconni, 2019-12-13:
Great ! We've finally succeeded.
Thanks for the report.

closed 2019-12-13 by bcoconni