JSBSim-Team/jsbsim #227: Generate Linear Model from XML Script

benhager, 2019-10-22:
Does anyone have any experience with, or is it possible to generate a linear model via XML scripting? Looking to call into FGStateSpace to get x0, u0, A, B, C, D vectors. Thanks!

bcoconni, 2019-10-26:
Your request is too vague. What do you mean by "generate a linear model" ?

bcoconni, 2019-11-16:
No response from the issuer for 3 weeks: issue closed.

benhager, 2019-12-03:
Sorry for the delay, i was not notified of a comment on this ticket. I thought generating a linear model is pretty standard, and I believe it is a part of the source code already. It is implemented in [JSBSim Advanced Triming Branch](https://github.com/arktools/jsbsim) by calling into FGLinearization.cpp.

The output should be something like this:
`x0=..
[  3.3500000001e+002;
   1.0833708379e-001;
   1.0833708380e-001;
   0.0000000000e+000;
  -5.8423256804e-004;
   0.0000000000e+000;
   0.0000000000e+000;
   0.0000000000e+000;
   0.0000000000e+000;
   0.0000000000e+000;
   0.0000000000e+000;
   1.5000000000e+002];
u0=..
[  4.1507273608e-001;
  -3.5248398193e-003;
  -2.0575829692e-001;
  -2.9106407215e-003];
sys = syslin('c',..
[ -4.3427060360e-002,  -9.8096550611e+000,  -3.2087058179e+001,  -2.0560557019e-010,   3.3305725641e+002,  -1.8636402009e-002,  -2.0618344994e-010,  -2.1078986357e-010,   6.3607575515e-010,   2.0773530465e-010,  -2.1567803515e-001,   1.4236057475e-005;
  -8.1166258462e-005,  -1.3702931926e-001,   1.1657013512e-002,   1.0000000112e+000,  -2.3830098715e-007,   1.5667392399e-005,   5.7283788879e-004,   6.2980734972e-005,  -1.4375970466e-004,  -1.7085689860e-007,  -2.1476001587e-008,   3.9049977087e-007;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   1.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   6.0540241507e-036,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   4.8014340544e-005,  -2.6971631568e+000,  -1.9094256891e-002,  -2.2374890325e+000,   6.2364254569e-001,  -2.5663288773e-005,  -9.2158761873e-004,  -1.0519342025e-004,   2.3547930880e-004,   2.7986858924e-007,   3.0142651432e-008,  -5.6196960641e-007;
   3.2484363763e-007,   1.4609051075e-004,  -2.0180142393e-004,  -3.0944937006e-015,   7.3392437136e-006,   9.5220718213e-002,   1.0812528411e-001,  -9.9413727569e-001,  -6.0973949052e-013,  -2.4168673428e-015,   1.4546616224e-004,  -3.2639002901e-015;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   5.4871117678e-038,   0.0000000000e+000,   0.0000000000e+000,   1.0000000000e+000,   1.0876293119e-001,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   4.3186288961e-006,   3.6266092870e-002,  -2.3172110931e-012,  -1.5085023771e-005,  -1.3973860910e+001,  -3.2521186265e-011,  -2.8602884424e+000,   2.5709013825e-002,   4.8483072664e-010,  -1.7883824485e-011,  -1.6154612370e-013,  -2.1185981843e-008;
  -3.8410920579e-008,   2.7088800961e-003,   3.0836305031e-012,   1.5042210065e-005,   1.8357369617e+000,  -4.5386746125e-010,  -2.9805207306e-002,  -3.9335120276e-001,   4.1956396647e-009,   2.3490235331e-012,   1.4995306610e-014,   1.8686946609e-010;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   3.0448632692e-036,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   1.0058972985e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
  -2.7919215606e-011,  -1.7922246605e-020,   1.1029074834e-020,   0.0000000000e+000,   0.0000000000e+000,  -1.7309715278e-006,   0.0000000000e+000,   0.0000000000e+000,   1.6008943163e-005,   2.0413438883e-017,   0.0000000000e+000,   4.3213431698e-016;
   4.7787890441e-008,   2.4281611155e-016,   1.2423149893e-016,   1.4117215788e-016,   9.0350181040e-017,  -1.0112899773e-009,   1.4117215788e-016,   1.4117215788e-016,   9.3529470475e-009,   1.4117215788e-016,   9.3173624198e-017,  -7.6492439679e-013;
   1.3286536223e-010,  -3.3500000001e+002,   3.3499994284e+002,   1.2199442845e-010,   1.2267097060e-010,   1.9457045900e-001,   1.6344275470e-010,   2.0440824959e-010,   1.7561183988e-010,   2.0440824959e-010,   3.8202369509e-010,   2.0440824959e-010],..
[  2.5569127211e+001,   2.0899803878e-012,   1.5534093046e+000,  -1.5323390704e-012;
  -1.1952405922e-003,   1.0162385181e-004,  -8.7323683594e-003,   7.4279608648e-004;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   4.6836146026e-002,  -1.6649632294e-004,  -1.1821223004e+000,  -1.2000347358e-003;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   1.6676965113e+000,   1.2463432217e-005,   5.3727331192e-001;
   0.0000000000e+000,  -1.4953362323e-003,   9.5942719514e-007,  -3.6441188893e-001;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000],..
[  1.0000000104e+000,   1.3737159558e-009,   7.0106883262e-009,   5.3053857603e-009,   6.4422541376e-009,   7.2949054205e-009,   5.3053857603e-009,   5.3053857603e-009,   6.5369931690e-009,   5.3053857603e-009,   4.8316906032e-009,   5.3053857603e-009;
  -9.1362103068e-013,   9.9999999998e-001,   1.9463597400e-011,  -1.2143064332e-012,  -2.7755575616e-013,  -3.4694469520e-013,  -1.6884641833e-012,  -1.2143064332e-012,  -3.0068540250e-013,  -1.6653345369e-012,  -1.1564823173e-012,  -1.2143064332e-012;
[... cut ...]
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000;
   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000,   0.0000000000e+000]);
tfm = ss2tf(sys);
`

seanmcleod, 2019-12-03:
Taking a quick look I see `FGLinearization.h` and  `FGLinearization.cpp` are in the current source tree in the `initialization` directory, however they are not referenced in `CMakeLists.txt` so they aren't included in the build currently.

seanmcleod, 2019-12-03:
Also searching for `FGLinearization` in the Advanced Trimming Branch see https://github.com/arktools/jsbsim/search?q=FGLinearization&unscoped_q=FGLinearization comes up without any hits at all. I was trying to see how the linearization code is typically called.

agodemar, 2019-12-03:
This code comes from the great work by James Goppert, back in 2012.
https://github.com/arktools/jsbsim

bcoconni, 2019-12-03:
> Taking a quick look I see `FGLinearization.h` and `FGLinearization.cpp` are in the current source tree in the `initialization` directory, however they are not referenced in `CMakeLists.txt` so they aren't included in the build currently.

Well they were at some point during JSBSim history (with the commit messages because they are quite enlightening):
* Commit 71e66e7f77447af0c0493fa0d52a659721a3f8d1 *Added (FINALLY) James Goppert's trimming and state space determination routines* (Oct 25,2012)
* Commit d724427f1117794cf43662ec442c0f11477226e1 *Incorporated most of or all of JGoppert's trim improvements* (Jan 25, 2013)

and then at some point Jon decided to remove all the code altogether:
* Commit 0c739604464f0f3eaceda676d200f33c89b20180 *Removed unused, undocumented, and not working trim analysis and linearization code* (Dec 9, 2015)

Interestingly, the linearization code was supposed to be run by setting a `mode` value in the property `/simulation/do_linearization`
https://github.com/JSBSim-Team/jsbsim/blob/d724427f1117794cf43662ec442c0f11477226e1/src/FGFDMExec.cpp#L157
and the linearization was made by calling `FGLinearization`
https://github.com/JSBSim-Team/jsbsim/blob/d724427f1117794cf43662ec442c0f11477226e1/src/FGFDMExec.cpp#L1203-L1211
Another interesting point is that the documentation of `DoLinearization` stipulated that it should be called after the aircraft was trimmed
https://github.com/JSBSim-Team/jsbsim/blob/d724427f1117794cf43662ec442c0f11477226e1/src/FGFDMExec.h#L470-L473
The good news are that we are not starting from a blank sheet. The bad news are that Jon's comment made it clear in the commit 0c739604464f0f3eaceda676d200f33c89b20180 (the one that removed James Goppert's code) that there were some problems with the code and I infer from that that we'd better know what we are doing before reintroducing this code in the current code base.

benhager, 2019-12-04:
I am not surprised that you should trim the aircraft before running the linearization routine. This is a common stipulation, as linearization is only valid at a stable trim point.

seanmcleod, 2020-04-08:
@jonsberndt can you shed any light on what the problems were in the code when you removed it via commit https://github.com/JSBSim-Team/jsbsim/commit/0c739604464f0f3eaceda676d200f33c89b20180?

In particular I guess if you remember any particular issues with the linearization code. If the issue was only/mainly with the trim code then maybe the linearization code could be used with the current trim code?

bcoconni, 2020-05-01:
I have come across [automatic differentiation](https://en.wikipedia.org/wiki/Automatic_differentiation) which should also allow to generate a linear model with much more flexibility than what was done by `FGLinearization`. With that technique you can virtually derive any quantity with respect to any other quantity without needing to hard code the formula for that particular differentiation.

The concept is quite simple, you need to implement a new class for [dual numbers](https://en.wikipedia.org/wiki/Dual_number#Differentiation) and replace most of, if not all, occurrences of `double` in JSBSim by that new class. Nothing really difficult but that would need a decent, although not necessarily extensive, knowledge of C++ (at the very least class design and templating) and no particular familiarity with JSBSim overall architecture since it would be a low level modification.

Any thoughts ?

rega0051, 2020-05-09:
Very interested in bringing this feature back. Along with more control of trim options. The two features go hand in hand, relying on generation of jacobians, etc. One issue will be dealing with certain classes of nonlinearities in the actuator and sensor models. Disabling freeplay, deadband, error models, noise in the subsystems makes the results more representative of the desired system.

I'd like to help if help is needed.

seanmcleod, 2020-05-10:
@rega0051 what additional control of the trim options are you looking for?

In terms of freeplay, deadband etc. if you don't want them included while generated the linearized model are you suggesting that the linearizing process automatically detect them and exclude them as opposed to you removing them from the model during the linearization process?

rega0051, 2020-05-11:
@seanmcleod As far as I know there isn't a means of either programmatically detecting or removing the presence of nonlinearities.

I've had many issues with the trim routines being hardcoded, I'm hopeful these issues won't propagate to the Linearization. A few of the issues off the top of my head: Trimming a flying wing without yaw control: can't use tFull, but tLongitudinal doesn't trim yaw at all (the desire is to have yaw trimmed with sideslip). The other recent issue I ran into is trimming a quad copter model in hover. The tCustom trim type would likely be useful, but there is no means of setting it up through a script or Python binding.

Goppert's Simplex linearization was also hardcoded if I recall (at least some of the parameters anyway, I think the GUI interface had ability to modify). The real difference with FGTrim is that the simplex algorithm will attempt to simultaneously minimize the objectives simultaneously.

seanmcleod, 2020-05-11:
@agodemar I'm trying to remember/figure out the relationship between your paper - **'A General Solution to the Aircraft Trim Problem'** - http://wpage.unina.it/agodemar/DSV-DQV/AIAA-2007-6703-905_DeMarco_Duke_Berndt.pdf and the trim code you mention in the paper to James Goppert's trimming code in -  https://github.com/arktools/jsbsim

Was James's code an updated/improved version of what you describe in the paper?

And do you remember why @jonsberndt removed it in 2015? Had something changed between 2013 and 2015 with lack of maintenance etc. to stop it working?

bcoconni, 2020-05-14:
@rega0051 
> Very interested in bringing this feature back. Along with more control of trim options. The two features go hand in hand, relying on generation of jacobians, etc. One issue will be dealing with [...]

Of course, help is **always** very appreciated :smile: You're very welcome to submit proposals. Just make sure that we are on the same line before writing too much code. First things first, I would like to clarify what you are volunteering for ? Is it bringing back James Goppert's `FGLinearization` code as the original poster @benhager asked? Or is it about coding automatic differentiation as I have suggested above ? Not trying to push either way but we need to discuss the same topic to avoid misunderstandings.
> As far as I know there isn't a means of either programmatically detecting or removing the presence of nonlinearities.

Well, it might not be possible to detect non linearities in general but we can detect the usage of `FGActuator`, `FGDeadBand` and the likes. Special handling can then be coded for these particular flight controls.

bcoconni, 2020-05-14:
@seanmcleod 
There were at some point 3 different trim algorithms in JSBSim.
#### Legacy trim routines
These are the older (and the last still active) trim routines. They were designed by Tony Peden. The algorithm is quite simple: it trims one axis at the time. Of course this fails when local axis minima do not match the global minimum, hence the development of algorithms that targeted the global minimum by trying to minimize all axes simultaneously.

**Files**: `src/initialization/FGTrim.h`, `src/initialization/FGTrim.cpp`, `src/initialization/FGTrimAxis.h` and `src/initialization/FGTrimAxis.cpp`.
#### De Marco/Durke/Berndt's trim algorithm
These are implementing the algorithms described in the paper '**A General Solution to the Aircraft Trim Problem**'. Even though its source code is still in JSBSim repository, it is no longer compiled and linked with JSBSim making it dead, unmaintained code.

**Files**: `src/initialization/FGTrimAnalysis.h`, `src/initialization/FGTrimAnalysis.cpp`, `src/initialization/FGTrimAnalysisControl.h` and `src/initialization/FGTrimAnalysisControl.cpp`.
#### Goppert's trim routines
As @rega0051 mentioned, these are using the Simplex algorithm to optimize all the axis simultaneously. The differentiation (aka linearization) is indeed hardcoded in `FGLinearization` and the trim routines themselves are managed by `FGTrimmer` and `FGSimplexTrim`. There again, even though the source code is still in JSBSim repository, it is no longer compiled and linked with JSBSim making it dead, unmaintained code.

**Files**: `src/initialization/FGTrimmer.h`, `src/initialization/FGTrimmer.cpp`,`src/initialization/FGSimplexTrim.h`,  `src/initialization/FGSimplexTrim.cpp`, `src/initialization/FGLinearization.h` and `src/initialization/FGLinearization.cpp`.

___
In the case where some or all of these routines would be brought back to JSBSim, keep in mind that they also need some "glue" code in `FGFDMExec` to be summoned from scripts.

seanmcleod, 2020-05-14:
@bcoconni thanks for the good summary.

I'm interested in trim routines that can potentially handle more cases than the current trim routines in JSBSim so I'll spend some time in a local branch adding Goppert's trim routines back in and testing them.

bcoconni, 2020-05-14:
> I'm interested in trim routines that can potentially handle more cases than the current trim routines in JSBSim so I'll spend some time in a local branch adding Goppert's trim routines back in and testing them.

Sure. That would be most interesting.

For the record, James Goppert has documented his algorithm in a document that is still accessible from the internet: [On Some Trim Strategies for Nonlinear Aircraft Flight Dynamics Models with the Open Source Software JSBSim](http://wpage.unina.it/agodemar/CEAS-2011/CEAS2011_Trim_JSBSim1.pdf)

Also note that James Goppert (@jgoppert) is on GitHub, no doubt he has some recollections about the algorithm he wrote almost 10 years ago :smiley:

seanmcleod, 2020-05-14:
When I was searching yesterday I also came across a couple of YouTube videos he posted demonstrating the trim functionality via a GUI he produced and also generating linearized models and loading them up in Scicoslab.

https://www.youtube.com/watch?v=JCjRGJTWe5k

https://www.youtube.com/watch?v=-B0nQbcyo-o

jgoppert, 2020-05-14:
I got sick of hidden states in the c++ code. For instance trying to track down states in a jet engine model. I would push you toward automatic differentiation. Also bring able to obtain a human readable state name would be useful. Using AD and getting a 20 state model and not knowing why can be annoying. I found the Modelica language is a bit friendlier in this regard as it maps directly to a differential algebraic equation and supports AD, also see pymoca and casadi which supports AD. Casadi is a C++ AD library and maybe you could leverage it. Let me know how I can help.

agodemar, 2020-05-14:
> For the record, James Goppert has documented his algorithm in a document that is still accessible from the internet: [On Some Trim Strategies for Nonlinear Aircraft Flight Dynamics Models with the Open Source Software JSBSim](http://wpage.unina.it/agodemar/CEAS-2011/CEAS2011_Trim_JSBSim1.pdf)

For the record, that a presentation I gave of a paper that @jgoppert and me co-authored quite a few years ago ;-)

bcoconni, 2020-05-18:
> For the record, that's a presentation I gave of a paper that @jgoppert and me co-authored quite a few years ago ;-)

Oops ! Sorry. And for the sake of completeness there was also a third author: Inseok Hwang 😉

bcoconni, 2020-05-18:
After some more archeological unearthing, I found the following 2 discussions threads in the mailing list archives:
* [Announcement of the addition of the simplex code to JSBSim](https://sourceforge.net/p/jsbsim/mailman/jsbsim-devel/thread/000001cdb26e$60f15db0$22d41910$@net) dated Oct 25, 2012
* [Jon reported that something is broken in the simplex code](https://sourceforge.net/p/jsbsim/mailman/message/32359901/) dated May 20, 2014. Not giving much details unfortunately.

ermarch, 2020-09-27:
This is just a heads up that the latest version of AeromatiC++ now also outputs the linear coefficients in JSON format. This might be enough for some uses.

bcoconni, 2020-09-27:
> This is just a heads up that the latest version of AeromatiC++ now also outputs the linear coefficients in JSON format. This might be enough for some uses.

@ermarch  Just to clarify things : Aeromatic++ is not able to read an existing JSBSim model and build linear coefficients for any trimmed position, is it ?

ermarch, 2020-09-27:
Yes that's correct, it can only calculate them from geometric data.

aarondewindt, 2021-01-07:
I'm interested in using Goppert's linearization code (`FGLinearization` & `FGStateSpace`) to generate linear systems from the JSBsim models. Based on this thread I think there are more people who are interested in this as well. Since I'm planning on doing this either anyway, is there some specific way you'd like me to expose this functionality?

My plan for now was to add a `DoLinearization` function to `FGFDMExec`. This function would take in four `std::vector< std::vector<double>>` references to write the state space matrices to. I'm only planning on using this function through the python wrapper, so the next step for me would be to wrap this function so it returns 4 numpy arrays.

Do do this I want to do these modification to the `FGLinearization` class.

- Make the A, B, C and matrices private members of the class.
- Add a `GetStateSpace(...)` function that takes four `std::vector< std::vector<double>>` references.
- Move the scicoslab code generator to a seperate function, but since I don't need this I'm not planning planning on exposing this unless there is some interest.

This will 100% break any code that already uses this, but I don't think will be an issue considering these files are currently not being included in the build.

I'll make pull request once I have something working. Are these any requested features?

agodemar, 2021-01-07:
@aarondewindt That's a good start. Thanks. It would be great if you could double check the trimming features introduced by Goppert.

aarondewindt, 2021-01-07:
I've exposed the linearization function to python. My plan is to spend the next few days testing and checking it out. 

Are there any known issues with it by any chance? [Jon mentions](https://sourceforge.net/p/jsbsim/mailman/message/32359901/) that the state space and linearization code stopped
working, but I'm getting models out of it. I still need to see how valid they are though. @jgoppert and @jonsberndt, do either of you have any input or tips for me?

aarondewindt, 2021-01-07:
@agodemar I might have a look at those at some point in the future, for now I'm focusing on the linearization part of it. From what I understand the issue that ended up with the Goppert's code being removed was with the linearization.

jgoppert, 2021-01-07:
The biggest difficulty I had with this was hidden engine states etc. I remember having to trim and then having to simulate until the engine states settled down. The linearization was a rather straight forward finite difference approach. If there was a way to get it working with automatic differentiation it would be much better. Also automatically finding all states in the code would be a huge step forward.

bcoconni, 2021-01-09:
@aarondewindt 
> This function would take in four `std::vector< std::vector<double>>` references to write the state space matrices

I suggest you use `typedef` or `using` to replace that ugly C++ code with something a bit more legible (in the `JSBSim` namespace of course)
```c++
using Matrix = std::std::vector< std::vector<double>>;
```

> This will 100% break any code that already uses this, but I don't think will be an issue considering these files are currently not being included in the build.

Given that the feature is disabled for years and nobody complained until @benhager brought the topic 1 year ago, your feature is unlikely to break any code :smile:
> I still need to see how valid they are though.

Yes, tests would be very much appreciated (just as you initiated with `TestLinearization` in your PR #370)

bcoconni, 2021-01-09:
@jgoppert
> If there was a way to get it working with automatic differentiation it would be much better.

I have some work in progress in my branch `LinearSpace` https://github.com/bcoconni/jsbsim/tree/LinearSpace where I am using [dual numbers](https://en.wikipedia.org/wiki/Automatic_differentiation#Automatic_differentiation_using_dual_numbers) for automatic differentiation in JSBSim. At the moment, all the tests from the `master` branch pass and I can get the gradient of any property with respect to (almost) any property, but it remains to be checked if the gradient computations are correct. At the moment, the gradient of the dynamic pressure `qbar` with respect to the body velocity `U` is correct (i.e I get `rho*U`). That's a start :smile: I also plan to use Eugene L. Duke's [Derivation and Definition of a Linear Aircraft Model](https://www.nasa.gov/centers/dryden/pdf/88104main_H-1391.pdf) to check the results I'm getting.

Furthermore, since every `double` occurrence must be replaced by a dual number instance, all the code of JSBSim is impacted. This means that the code needs to be maturated before finding its way to the `master` branch. In addition, automatic differentiation increases the amount of computations (twice as much since you need to compute the value and its gradient) so it cannot be enabled permanently. I need to find a way to enable it optionally without making the code a complete nightmare to read. Finally, there remains much to be done for the API. At the moment, you set the gradient of a property (say `U` aka `velocities/u-fps`) to `1.0` and JSBSim computes the gradients of all the variables with respect to `U` in parallel to its "normal" computations and that's it. No matrices/jacobians are built. You need to collect "manually" the gradients if you plan to get a matrix. As I said, this is WIP.

The work of @aarondewindt is useful in that we will be able to cross check the results from 2 different algorithms which is always useful. In addition, it will be the only way to get linearization matrices for a while :smile:

aarondewindt, 2021-01-09:
@bcoconni 

> I suggest you use typedef or using to replace that ugly C++ code with something a bit more legible (in the JSBSim namespace of course)

Are you sure? Naming it matrix kinda implies it can be used for matrix calculations, but it's nothing more than an std::vector.

bcoconni, 2021-01-09:
> Are you sure? Naming it matrix kinda implies it can be used for matrix calculations, but it's nothing more than an std::vector.

Well, use whatever name you like which make the whole expression shorter !

bcoconni, 2021-01-15:
@aarondewindt's patch has just been committed to the branch `master`. Finite difference linearization is now back in JSBSim ! Thanks Aaron and thanks also James who authored the original linearization code :+1:

At the moment,the easiest way to run the feature is with Python:
```python
import jsbsim

fdm = jsbsim.FGFDMExec('.') # The path supplied to FGFDMExec is the location of the folders "aircraft", "engines" and "systems"

fdm.load_model('737') # Load the aircraft 737
fdm.load_ic('cruise_init.xml', True) # Load the initial conditions
fdm.run_ic() # Initialize the aircraft with initial conditions

# Set engines running
fdm['propulsion/engine[0]/set-running'] = 1
fdm['propulsion/engine[1]/set-running'] = 1
fdm.run()

# Trim
fdm['simulation/do_simple_trim'] = 1

# Linearization
linearization = jsbsim.FGLinearization(fdm)

print(linearization.system_matrix)
print(linearization.input_matrix)
print(linearization.output_matrix)
print(linearization.feedforward_matrix)
```

agodemar, 2021-01-15:
That's great news. Thanks

sthelia, 2021-05-27:
Hi, 
This is a great feature. I'm trying to replicate this but as soon as I include the linearization part I get an error. 

``` 
  File "/home/test/jsbsim-code2/test4.py", line 33, in <module>
    linearization = jsbsim.FGLinearization(fdm)
AttributeError: module 'jsbsim' has no attribute 'FGLinearization'
```

Do I have to do anything different to include the correct code? I'm running JSBSim 1.1.6.

seanmcleod, 2021-05-27:
When and how did you install the python module for JSBSim? Maybe you've downloaded and built JSBSim 1.1.6 but you've previously done a `pip install` of JSBSim of an earlier version before the linearization was added?

sthelia, 2021-05-27:
Yes, I have done a `pip install` of JSBSim, I don't remember when but sometime this spring, definitely after the Jan 15th comment. I did uninstall it and then reinstalled it today to see if that made any difference, but the error is still the same.

seanmcleod, 2021-05-27:
My spring is at a very different time to your spring 😉 

But I just double-checked and I can confirm that I see this issue after pip installing 1.1.6.

```sh
(base) C:\Users\Sean>pip install jsbsim
Collecting jsbsim
  Downloading JSBSim-1.1.6-320-cp37-cp37m-win_amd64.whl (767 kB)
     |████████████████████████████████| 767 kB 6.8 MB/s
Requirement already satisfied: numpy in c:\users\sean\anaconda3\lib\site-packages (from jsbsim) (1.18.1)
Installing collected packages: jsbsim
Successfully installed jsbsim-1.1.6
(base) C:\Users\Sean>
```

Trying to use `FGLinearization`, note the JSBSim build date of Apr 30 2021 and the confirmation of v1.1.6.

```sh
>>> import jsbsim
>>> fdm = jsbsim.FGFDMExec('.')

     JSBSim Flight Dynamics Model v1.1.6 [GitHub build 320/commit b477f6312bee2fd3af4c4e5ba1a3e732ed2b99d4] Apr 30 2021 12:00:59
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

>>> linearization = jsbsim.FGLinearization(fdm)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'jsbsim' has no attribute 'FGLinearization'
>>>
```

It looks to me @bcoconni something went awry with the tagging of v1.1.6?

The `master` branch shows the linearization support being added to `jsbsim.pyx.in` and 'jsbsim.pxd` on the 15th Jan, e.g.

https://github.com/JSBSim-Team/jsbsim/blob/master/python/jsbsim.pyx.in

However if you look at `jsbsim.pyx.in` and `jsbsim.pxd` for the v1.1.6 tag they're older than the 15th Jan, even though the v1.1.6 tag and version built is from Apr 2021?

https://github.com/JSBSim-Team/jsbsim/tree/v1.1.6/python

seanmcleod, 2021-05-27:
> However if you look at jsbsim.pyx.in and jsbsim.pxd for the v1.1.6 tag they're older than the 15th Jan, even though the v1.1.6 tag and version built is from Apr 2021?

Hmm, I assumed based on the build date of Apr 30 2021 and the following comment on https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.1.6 that 1.1.6 was based on code in master on Apr 30.

> github-actions released this 27 days ago · 164 commits to master since this release

However looking at the link to the 164 commits since - https://github.com/JSBSim-Team/jsbsim/compare/v1.1.6...master it looks like v1.1.6 is based on code from the beginning of Oct 2020, which would explain why it doesn't have any support for the linearization.

sthelia, 2021-05-27:
@aarondewindt Alright, uninstalling with pip seemed to work! Thanks.

bcoconni, 2021-05-28:
> However looking at the link to the 164 commits since - `v1.1.6...master` it looks like v1.1.6 is based on code from the beginning of Oct 2020, which would explain why it doesn't have any support for the linearization.

Yes, correct. Releases `v1.1.x` are maintenance releases. The linearization feature is in `1.2.0dev1` as @aarondewindt pointed out.

@sthelia if you want to save yourself the burden of compiling JSBSim, you can download a Python wheel package from our [Rolling Release](https://github.com/JSBSim-Team/jsbsim/releases/tag/Linux) and pick the file `*.whl` that match your platform and Python version. The file can then be installed with `pip`:
```bash
> pip install jsbsim --no-index -f /path/to/the/wheel/file.whl
```

seanmcleod, 2021-05-28:
@sthelia just to confirm I followed @bcoconni's suggestion of downloading a wheel file from the rolling release, pip installed it and I was able to successfully run the linearization example that @bcoconni posted earlier in this thread.

GolgeA, 2022-10-23:
> sys = syslin('c',.. [ -4.3427060360e-002, -9.8096550611e+000, -3.2087058179e+001, -2.0560557019e-010, 3.3305725641e+002, -1.8636402009e-002, -2.0618344994e-010, -2.1078986357e-010, [...]

I've been looking at linear models for a few days too, and I saw a problem with this 12-state system matrix A. First three states of twelve are normally positions and they should have no effect on each other, sometimes last three states are positions but here for both conditions, position states are dependent to other positions which is seems inaccurate. Here I would like to know the order of the state variable or state variables column. Maybe the order of the states are different?

seanmcleod, 2022-10-23:
> sometimes last three states are positions

Looking at the order here it does look like the last 3 states are positions (lat, lon, alt).

https://github.com/JSBSim-Team/jsbsim/blob/c943e7817b06e0fd4eea05241b32ff00de879f7a/src/initialization/FGLinearization.cpp#L28-L58

pd3a, 2022-12-01:
I tried doing the linearization but the trim failed. Could anyone help

 fdm['simulation/do_simple_trim'] = 1
  Full Trim

  Trim failed
  Trim Results:
       Angle of Attack:   7.50  wdot:  3.22e+01 Tolerance: 1e-03  Failed
              Throttle:   0.50  udot:  1.35e+00 Tolerance: 1e-03  Failed
            Pitch Trim:   0.00  qdot: -4.53e-03 Tolerance: 1e-04  Failed
            Roll Angle:  -0.00  vdot:  1.31e-15 Tolerance: 1e-03  Passed
              Ailerons:   0.00  pdot:  3.61e-22 Tolerance: 1e-04  Passed
                Rudder:   0.00  rdot: -1.01e-22 Tolerance: 1e-04  Passed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "jsbsim.pyx", line 1058, in jsbsim.FGFDMExec.__setitem__
  File "jsbsim.pyx", line 1232, in jsbsim.FGFDMExec.set_property_value
jsbsim.TrimFailureError: Trim Failed

seanmcleod, 2022-12-01:
> I tried doing the linearization but the trim failed

You haven't specified what aircraft you're using and what your initial conditions are. For example maybe you're trying to trim the 737 at 50kts, well you're never going to be able to generate a trim solution for that combination.

seanmcleod, 2022-12-01:
Have you looked at `reset00.xml`?

```sh
(base) C:\source\jsbsim\aircraft\f16>type .\reset00.xml
<?xml version="1.0"?>
<initialize name="reset00">
  <!--
    This file sets up the aircraft to start off
    from the runway in preparation for takeoff.
  -->
  <ubody unit="FT/SEC">    0.0  </ubody>
  <vbody unit="FT/SEC">    0.0  </vbody>
  <wbody unit="FT/SEC">    0.0  </wbody>
  <latitude unit="DEG">   47.0  </latitude>
  <longitude unit="DEG"> 122.0  </longitude>
  <phi unit="DEG">         0.0  </phi>
  <theta unit="DEG">       0.0  </theta>
  <psi unit="DEG">       180.0  </psi>
  <altitude unit="FT">     5.5  </altitude>
</initialize>
(base) C:\source\jsbsim\aircraft\f16>
```

So it's sitting on the ground with no airspeed. So the only trim option that would make sense would be `ground trim`.

Secondly the trim isn't going to work with an aircraft which has a FBW command law like the F-16 has. So try it with an aircraft without a FBW command law, e.g. the 737 model or the C172, A4 models for example.

pd3a, 2022-12-01:
ok understood. But i want to compute for f-16, how can i do it? Thanks

seanmcleod, 2022-12-01:
@pd3a are you trying to generate a linear model as per this discussion? If not, and you're just trying to trim the F-16 model then rather start a separate discussion dedicated to trimming FBW aircraft.

pd3a, 2022-12-01:
I'm trying to generate a linear model for f-16, yes.

seanmcleod, 2022-12-01:
Okay, if your aim is to generate a linear model for the F-16 around some trim point then I suggest you edit the F-16 model and remove the FBW command law implementation and replace it with a basic FCS like you see in the non-FBW aircraft.

JSBSim should then be able to generate a trim solution, assuming you have valid initial conditions, and then you can get a linear model as per above.

pd3a, 2022-12-01:
Even after removing FBW command law i'm getting same error. I gave initial conditions of speed 420 ft/s and 15000ft.

seanmcleod, 2022-12-01:
Post/attach a copy of your updated F-16 FDM and initial conditions file.

pd3a, 2022-12-01:
[Linear.zip](https://github.com/JSBSim-Team/jsbsim/files/10133205/Linear.zip)

I attached the zip file here

pd3a, 2022-12-01:
could you guide me how to replace the FBW with the FCS

seanmcleod, 2022-12-01:
Basically take a look at the 737's implementation and use that with the F-16.