JSBSim-Team/jsbsim #228: Interfacing JSBSim for "live" Control Inputs

davidosten, 2019-10-27:
In the header file src\FDM\JSBSim\FGFDMExec.h of FlightGear (attached
below), I find the exact description of what I would like to code,
however I am not clear about hot to code the copy_toJSBsim() and
copy_from_JSBsim() functions.

     @code
     copy_to_JSBsim(); // copy control inputs to JSBSim
     fdmex->RunIC(); // loop JSBSim once w/o integrating
     copy_from_JSBsim(); // update the bus
     @endcode

     Once initialization is complete, cyclic execution proceeds:

     @code
     copy_to_JSBsim(); // copy control inputs to JSBSim
     fdmex->Run(); // execute JSBSim
     copy_from_JSBsim(); // update the bus
     @endcode

 >>>

I am aiming to use JSBSim as a fast-paced 6DoF Simulation core, and plan
to feed - during runtime - control-inputs into JSBSim. The interfacing
approach should already be realised in FlightGear, however I do not
figure out the respective spot in the code of FlightGear to see how the
interfacing is performed.

My question is: does JSBSim already contain the feature of feeding
control-inputs via UDP Sockets (or any other means) from an external
program during runtime into the simulation, or if not, how can I code
this (as mentioned in the code snippet above)? As far as I understand, I can enter control inputs only via the XML-Runscript BEFORE running the JSBSim simulation.

Best Regards,
David

bcoconni, 2019-10-31:
> I am not clear about hot to code the copy_toJSBsim() and copy_from_JSBsim() functions

There are examples of these routines in the file `examples/flightgear.cxx`
https://github.com/JSBSim-Team/jsbsim/blob/4840a621dafe1ce64970990ddaec3e86a1c7e082/examples/FlightGear.cxx#L589-L730
https://github.com/JSBSim-Team/jsbsim/blob/4840a621dafe1ce64970990ddaec3e86a1c7e082/examples/FlightGear.cxx#L736-L995

> does JSBSim already contain the feature of feeding control-inputs via UDP Sockets (or any other means) from an external program during runtime into the simulation

Yes this is feasible. For example, the process to control JSBSim via UDP sockets from FlightGear is described in [this example](https://github.com/JSBSim-Team/jsbsim#how-can-i-interface-jsbsim-with-flightgear-). If you plan to code UDP sockets yourself, the data format is described in the file https://github.com/JSBSim-Team/jsbsim/blob/4840a621dafe1ce64970990ddaec3e86a1c7e082/src/input_output/net_fdm.hxx#L1-L121

opt12, 2019-11-12:
I'm stumbling into the very same problem and did not really get a hold at it:

I'm trying to use JSBSim as a flight model for some machine learning and interface it from Python.

I want to set controls, execute a step, read back the new state and figure out the next controls.

Unfortunately, if I do this too fast, JSBSim misses out control commands and if I do this too slow, JSBSim keeps on running while my Python script is still waiting.

Is there a way to synchronize JSBSim with external input/output and to synchronize the internal steps with some outside trigger? this is very similar to the 
 ```
 @code
 copy_to_JSBsim(); // copy control inputs to JSBSim
 fdmex->Run(); // execute JSBSim
 copy_from_JSBsim(); // update the bus
 @endcode
```
snippet given above. I'm just missing out how to realize this 
`copy_to_JSBsim(); // copy control inputs to JSBSim` and ` copy_from_JSBsim(); // update the bus`in a blocking (and thus synchronous) way from my Python script.

Any help is very much appreciated!

THX

bcoconni, 2019-11-27:
> big THX to the JSBSim team.

@opt12 Thank you on behalf of the team. That is much appreciated.

@davidosten Have your questions been answered ? Or do you need further support ?

bcoconni, 2019-12-14:
No response from the issuer means the ticket is closed.
Thanks.

closed 2019-12-14 by bcoconni