JSBSim-Team/jsbsim #230: Change of payload mass translates aircraft along the ground

Gijsrooy, 2019-10-31:
Hi,

With current FlightGear Git, if I increase the weight of a payload/tank while standing still on the ground, the aircraft moves a couple of feet into the direction where this weight is added. If I then decrease the weight, the aircraft moves back to its initial position. The movements are proportional to the magnitude of the weight that's added. It looks as if there's something wrong with the ground interactions/friction.

I'm sorry that I do not have the time for a more thorough analysis right now, but I will update this issue when more info becomes available. At least the issue has been confirmed by several others and it is not unique to my 747-400 (although it is very prominent there due to the large masses and arms).

andgi, 2019-10-31:
Hi Gijs,

I think that it may well be that JSBSim primarily keeps track of the centre of gravity - so when you move that the aircraft "moves" but without being seen as moving by JSBSim. Possibly.

bcoconni, 2019-10-31:
> it may well be that JSBSim primarily keeps track of the centre of gravity - so when you move that the aircraft "moves" but without being seen as moving by JSBSim.

According to our good old friend Isaac Newton, the CoG of a resting body only moves when that body is submitted to forces. As a consequence when the mass distribution is modified via the menus in FlightGear, no forces are produced on the aircraft body and the CoG does not move. However the alteration of the mass distribution results in a displacement of the CoG with respect to the structural frame and this produces the translation of the structural frame (and the 3D model) with respect to the World.

There is a hack that is designed in JSBSim to counteract that artificial displacement:
https://github.com/JSBSim-Team/jsbsim/blob/4840a621dafe1ce64970990ddaec3e86a1c7e082/src/models/FGMassBalance.cpp#L209-L210
Currently `FGPropagate->NudgeBodyLocation` is only triggered when the property `forces\hold-down` is set to `true` i.e. when `FDMExec->GetHoldDown()` returns `true`. I think we should also trigger this correction when there is Weight On Wheels i.e. when `FDMExec->GetGroundReactions()->GetWOW()`is `'true`.

The following modification to `src/models/FGMassBalance.cpp` should fix the issue:
```diff
- if (FDMExec->GetHoldDown())
+ if (FDMExec->GetHoldDown() || FDMExec->GetGroundReactions()->GetWOW())
    Propagate->NudgeBodyLocation(vDeltaXYZcgBody);
```
Could you please test it on FlightGear and let me know if that fixes the issue you reported ?

Gijsrooy, 2019-11-01:
Hi Bertrand,

Thanks for the swift reply! Adding the mod to FGMassBalance (and including FGGroundReactions.h) does indeed solve the issue. Great!

It does make me wonder why I didn't observe this behaviour before. Has something changed in the way JSBSim works?

bcoconni, 2019-11-10:
> Adding the mod to FGMassBalance (and including FGGroundReactions.h) does indeed solve the issue.

Thanks for the tests. The bug fix is now committed to JSBSim and FlightGear.
> It does make me wonder why I didn't observe this behaviour before. Has something changed in the way JSBSim works?

Yes, since the commit 8a0cff17c905fd18007180859656322c4998eee5, the function `NudgeBodyLocation` is no longer called unconditionally.

Indeed the CoG displacement can only be the result of the application of forces which we already account for in `FGAccelerations`. There are exceptions however where some forces are not modeled in JSBSim among which the modification of mass distribution while the aircraft is on ground.

Gijsrooy, 2019-11-11:
Excellent, thanks Bertrand!

closed 2019-11-11 by Gijsrooy