JSBSim-Team/jsbsim #231: Research paper on JSBSim

agodemar, 2019-11-06:
Dear all,
I believe this is a good place to ask about this subject. If someone feels this is an OT message, I apologize in advance.
If you are involved in a research or project that features flight simulations with JSBSim *and* can consider to (or are allowed to) write down a joint paper on how the software has been determinant to accomplish this or that goal, please let me know.

All sorts of suggestion and criticism are welcome.

As a scholar, I think it's time to produce something valuable that arises from the community of developers and users.

agodemar, 2019-12-15:
I am collecting infos on recent use cases of JSBSim.

It would be great to hear how JSBSim has been an effective support for your projects.
Which University or company? Which research group? Which project? Which was your achievement?

I want to mention the most relevant ones in a paper that I have in mind. It will be an updated version of this one: [AIAA-2009-5699_Berndt_DeMarco](http://wpage.unina.it/agodemar/DSV-DQV/AIAA-2009-5699_Berndt_DeMarco.pdf) that was written 10 years ago (2009).

Please, do feel free to contact me if you want to contribute.

seanmcleod, 2019-12-15:
@agodemar I've sent you an email regarding my use of JSBSim as part of a VSS simulator for test pilot training. Not sure if it will be of enough interest as part of an academic paper.

jonsberndt, 2019-12-15:
Sean,

 

I’d be interested to see this, too, if that’s possible.

 

Jon Berndt

seanmcleod, 2019-12-15:
@jonsberndt I've just emailed you the info I sent Agostino.

agodemar, 2019-12-15:
@bcoconni are you willing to appear as a co-author?
JSBSim state of the art means also how we have ported the original project on GitHub and how we manage our CI/CD workflow, how we support users and improve models and code quality.

bcoconni, 2019-12-16:
> @bcoconni are you willing to appear as a co-author?

Sure, although I am not a researcher myself but a pat in the back is always welcome 😄

Gijsrooy, 2020-01-09:
Hi Agostino, 

Nice initiative! I will check if some of my colleagues at Delft University of Technology are interested in contributing. In fact, I'm currently at AIAA's SciTech in Orlando, trying to attend all of the FlightGear/JSBsim presentations. Will see if I can get some additional contacts from that.

Gijs

agodemar, 2020-01-09:
Hi Gijs, yes you're in the perfect spot to get an idea of recent/advanced JSBSim use cases.
I'd be great if you could report your impressions.

jonsberndt, 2020-01-09:
How many papers is FlightGear and JSBSim mentioned in?

 

Jon

Gijsrooy, 2020-01-09:
Hi Jon,

There are 3 papers with JSBSim being mentioned (and actively used), 2 of those also used FlightGear, bringing the grand total for FlightGear on 7 papers this week.

seanmcleod, 2020-01-09:
@Gijsrooy can you post the abstracts of the 7 papers here? When I looked at AIAA's conference website I could only find the title of the talk/paper. I'm assuming as an attendee you might have online access to the list of abstracts?

Gijsrooy, 2020-01-09:
You can find the abstracts at ARC (they are free to access, but for the actual papers you must have a subscription):

- https://arc.aiaa.org/doi/10.2514/6.2020-0135
- https://arc.aiaa.org/doi/10.2514/6.2020-0136
- https://arc.aiaa.org/doi/10.2514/6.2020-0373
- https://arc.aiaa.org/doi/10.2514/6.2020-0409
- https://arc.aiaa.org/doi/10.2514/6.2020-0859
- https://arc.aiaa.org/doi/10.2514/6.2020-1401
- https://arc.aiaa.org/doi/10.2514/6.2020-1968
- https://arc.aiaa.org/doi/10.2514/6.2020-1970

seanmcleod, 2020-01-09:
@Gijsrooy perfect thanks!

seanmcleod, 2020-01-09:
And @Gijsrooy I see you authored one of the papers - https://arc.aiaa.org/doi/10.2514/6.2020-0373, although I can only read the abstract at the moment ;<

Gijsrooy, 2020-01-09:
That's correct Sean! I've presented my master thesis in Orlando: https://forum.flightgear.org/viewtopic.php?f=34&t=35902

seanmcleod, 2020-01-09:
@Gijsrooy congrats. Which programmable sidestick did you use? I've used one from Stirling Dynamics in the past and currently using the CLS-P joystick from Brunner.

Gijsrooy, 2020-01-10:
In our experiment we used a (rather old) Moog FCS Ecol-8000, that allows us to adjust the forces in the stick dynamically.

HanbumKo, 2020-02-12:
Hi, I've made dogfight Reinforcement Learning environment using JSBSim and also have been using, do you all think that this would be helpful to the project also?
https://www.youtube.com/watch?v=wi7mrOGmCXc

closed 2020-12-11 by bcoconni