JSBSim-Team/jsbsim #232: function: <random> and <urandom> fixed ... but crash the FGFS System ...

abassign, 2019-11-07:
I started using the <random> function in a system to generate random fluctuations in the flight.
If I call it only in one function, for the whole FGFS program it works regularly, but if I have another function that calls it ... the FGFS system freezes.

For example:

```
<channel name="Airbrake control" execrate="8">
     ...
     <fcs_function name="systems/airbrake/induced-turbulence">
            <function>
                <product>
                    <product>
                        <p>systems/airbrake/speedbrake-pos-norm</p>
                        <p>aero/qbar-psf</p>
                        <random/>
                        <p>systems/airbrake/induced-turbulence-intensity</p>
                    </product>
                    <table>
                        <independentVar>aero/qbar-psf</independentVar>
                        <tableData>
                            0.0 0
                            6.53 0.008
                            26.1 0.006
                            104.0 0.002
                            418.0 0.001
                            1671.0 0.0005
                        </tableData>
                    </table>
                </product>
            </function>
        </fcs_function>
```
If I repeat the same code, with some minor variations, in another program block (different channel or different XML) I have the immediate freeze of the FGFS program (since JSBSim is executed inside FGFS I think it has blocked JSBSim. If isolo the <random /> statement entering for example <v> 0.0 </1> everything works properly.

```
<channel name="Landing Gear control" execrate="8">
        ...
        <fcs_function name="systems/gear/induced-turbulence">
            <function>
                <product>
                    <product>
                        <p>gear/gear-cmd-norm</p>
                        <p>aero/qbar-psf</p>
                        
                        <!-- ERRORE DA CHIARIRE! se inserito si blocca se c'é un'altro random
                        <random/>
                        -->
                        <v>0.0</v>
                        
                        <p>systems/gear/induced-turbulence-intensity</p>
                    </product>
                    <table>
                        <independentVar>aero/qbar-psf</independentVar>
                        <tableData>
                            0.0 0
                            6.53 0.008
                            26.1 0.006
                            104.0 0.002
                            418.0 0.001
                            1671.0 0.0005
                        </tableData>
                    </table>
                </product>
            </function>
        </fcs_function>
```
I think I will insert many random generators along the G91R1B program so I would like to know if it is possible to solve the problem.

Thanks for collaboration

bcoconni, 2019-11-10:
Sorry but I cannot reproduce the issue you are reporting. Can you be more specific about the set of functions that lead to the freeze ?

abassign, 2019-11-10:
Hi, I did a further check, I verified that the system does not crash, but slows down about 6-7 times. The loading of the G91R1B aircraft program in FGFS instead of taking 33 seconds ... if active the second command <random> becomes 2 minutes and 28 seconds long which increase to 3 minutes and 18 seconds if execrete is entered = 8 ... Obviously I did not expect such an abnormal expansion in the loading time and therefore I had blocked the program before with CTRL-C thinking it had stopped. I then activated the multi-threading system parameter:
<Multithreading-mode> AutomaticSelection </ multithreading-mode>
with the <random> command entered the time became 4 minutes and 40 seconds! If I remove the second command <random> the time has been reduced to 45 seconds (It is normal that with <multithreading-mode> AutomaticSelection </multithreading-mode>  active the loading time of FGFS is higher.)

bcoconni, 2019-11-10:
> [...] Obviously I did not expect such an abnormal expansion in the loading time and therefore I had blocked the program before with CTRL-C thinking it had stopped.

While conducting my tests, I have observed neither a program freeze nor a longer time to initiate. The test case `TestFunctions` runs in approximately 0.5 seconds whether I request multiple occurences of `<random>` or not.
> I then activated the multi-threading system parameter:
AutomaticSelection </ multithreading-mode>
with the command entered the time became 4 minutes and 40 seconds! If I remove the second command the time has been reduced to 45 seconds (It is normal that with AutomaticSelection active the loading time of FGFS is higher.)

I'm sorry but this is a matter that should be discussed with FG developers, not here.
> My feeling is that there is an interference between the random number generation system with some FGFS program used during the upload.

I need hard facts to track down and fix bugs ; ideally a test case that allows to reproduce the bug occurrence locally. Gut feelings are not helping in any way.

abassign, 2019-11-14:
Hi, @bcoconni 

Hello,

I did a quick check of the slow-motion problem with the random number generator I reported to you in this thread.
The problem of slow start-up  running is been reported in various parts and is due to a slowness in the code when the rand () function is called!

For example look this: [https://stackoverflow.com/questions/21237905/how-do-i-generate-thread-safe-uniform-random-numbers](url)

At this point I did a test by inserting the call to the random engine defined in <random>

std :: default_random_engine generator;

For example:

```
    } else if (operation == "random") {
      auto f = [](const decltype(Parameters)& p)->double {
                 // return GaussianRandomNumber();
                 // abassign
                 std::default_random_engine generator;
                 std::normal_distribution<double> distribution(1.0,1.0);
                 return distribution(generator);
               };
      Parameters.push_back(new aFunc<decltype(f), 0>(f, fdmex, element, Prefix, var));
```

So the problem is the call to the rand () function repeated hundreds of times per second. I don't know the reason, it could be lagato to the fact that the system clock is called, it is known that the call to the system clock is "relatively slow", while the #include <random> inserts a class of pseudorandom functions that are notoriously very faster.

At this point I would check the JSBSim code where the rand () function is used and replace it with the functions defined in <random>.

But one thing I think is important:

Being calls:

std :: default_random_engine generator;
ùstd :: normal_distribution <double> distribution (1.0.1.0);

a global action on the system ... it would be necessary to isolate them .. for this there are various techniques that I would like to analyze in order to definitively resolve the problem.

I think at this point it would be useful to build a separate random generator module in order to take advantage of the function with user-defined seeds and different distributions.

What do you think about it ?

bcoconni, 2019-11-16:
Interesting, I was not aware of the speed issues related to the random number generator `rand()`. Your approach seems fine to me.
> a global action on the system ... it would be necessary to isolate them .. for this there are various techniques that I would like to analyze in order to definitively resolve the problem.

What about using the following (untested) code ?
```c++
    } else if (operation == "random") {
      std::default_random_engine generator;
      std::normal_distribution<double> distribution(1.0,1.0);
      auto f = [generator, distribution](const decltype(Parameters)& p)->double {
                 return distribution(generator);
               };
```
The two variables `generator` and `distribution` are created once when the `random` function is generated and captured by value by the lambda function using the operator `[generator, distribution]`. That would mean that the 2 variables are unique to each occurrence of the `random` function.

On a side note, I am surprised that you provide `(1.0, 1.0)` to the constructor of `std::normal_distribution<double>` ? Should not it be `(0.0, 1.0)` ?

abassign, 2019-11-16:
@bcoconni Ciao,

I try with your code, 
```
 } else if (operation == "random") {
      //auto f = [](const decltype(Parameters)& p)->double {
      //           return GaussianRandomNumber();
      std::default_random_engine generator;
      std::normal_distribution<double> distribution(0.0,1.0);
      auto f = [generator, distribution](const decltype(Parameters)& p)->double {
                 return distribution(generator);
               };
      Parameters.push_back(new aFunc<decltype(f), 0>(f, fdmex, element, Prefix, var));

```

I like the solution because it builds a random generator for each call. But from this error which unfortunately, due to my limited knowledge of C ++, I cannot understand:

```
[ 31%] Building CXX object src/FDM/JSBSim/CMakeFiles/JSBSim.dir/models/FGFCS.cpp.o
/media/abassign/Dati/Dati/fgfs-rb/flightgear/src/FDM/JSBSim/math/FGFunction.cpp: In lambda function:
/media/abassign/Dati/Dati/fgfs-rb/flightgear/src/FDM/JSBSim/math/FGFunction.cpp:526:47: error: no match for call to ‘(const std::normal_distribution<double>) (const default_random_engine&)’
                  return distribution(generator);
                                               ^
In file included from /usr/include/c++/8/random:49,
                 from /media/abassign/Dati/Dati/fgfs-rb/flightgear/src/FDM/JSBSim/math/FGFunction.cpp:34:
/usr/include/c++/8/bits/random.h:2040:2: note: candidate: ‘std::normal_distribution<_RealType>::result_type std::normal_distribution<_RealType>::operator()(_UniformRandomNumberGenerator&) [with _UniformRandomNumberGenerator = const std::linear_congruential_engine<long unsigned int, 16807, 0, 2147483647>; _RealType = double; std::normal_distribution<_RealType>::result_type = double]’ <near match>
  operator()(_UniformRandomNumberGenerator& __urng)
  ^~~~~~~~
/usr/include/c++/8/bits/random.h:2040:2: note:   passing ‘const std::normal_distribution<double>*’ as ‘this’ argument discards qualifiers
/usr/include/c++/8/bits/random.h:2045:2: note: candidate: ‘template<class _UniformRandomNumberGenerator> std::normal_distribution<_RealType>::result_type std::normal_distribution<_RealType>::operator()(_UniformRandomNumberGenerator&, const std::normal_distribution<_RealType>::param_type&) [with _UniformRandomNumberGenerator = _UniformRandomNumberGenerator; _RealType = double]’
  operator()(_UniformRandomNumberGenerator& __urng,
  ^~~~~~~~
/usr/include/c++/8/bits/random.h:2045:2: note:   template argument deduction/substitution failed:
/media/abassign/Dati/Dati/fgfs-rb/flightgear/src/FDM/JSBSim/math/FGFunction.cpp:526:47: note:   candidate expects 2 arguments, 1 provided
                  return distribution(generator);
                                               ^
make[2]: *** [src/FDM/JSBSim/CMakeFiles/JSBSim.dir/build.make:375: src/FDM/JSBSim/CMakeFiles/JSBSim.dir/math/FGFunction.cpp.o] Error 1
make[2]: *** Attesa per i processi non terminati....
[ 31%] Linking CXX executable fgcom
[ 31%] Built target fgcom
make[1]: *** [CMakeFiles/Makefile2:1999: src/FDM/JSBSim/CMakeFiles/JSBSim.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

```
There could be an alternative solution, at least for the functions, to construct a single generator and two different "distribution" (uniform and Gaussian) to be called as a function. In this way the number of generators is only one.
What do you think about it ? Even if the solution you proposed seems preferable to me because each "<random>" method is independent of the others.

bcoconni, 2019-11-16:
> But from this error which unfortunately, due to my limited knowledge of C ++, I cannot understand: [...snip..]

Using the cursed `const_cast` allows the compilation to succeed (this is really a C++ privilege to get such an ugly and convoluted syntax :angry: )
```c++
    } else if (operation == "random") {
      std::default_random_engine generator;
      std::normal_distribution<double> distribution(0.0, 1.0);
      auto f = [generator, distribution](const decltype(Parameters)& p)->double {
                 auto d = const_cast<std::normal_distribution<double>&>(distribution);
                 auto g = const_cast<std::default_random_engine&>(generator);
                 return d(g);
               };
```
I'll let you check if that works.
> There could be an alternative solution, at least for the functions, to construct a single generator and two different "distribution" (uniform and Gaussian) to be called as a function. In this way the number of generators is only one.

Yes that could allow an easier to read syntax. If `generator` and `distribution` are made static members of `FGFunction` you could pass them with the `this` pointer
```c++
    } else if (operation == "random") {
      auto f = [this](const decltype(Parameters)& p)->double {
                 return this->distribution(this->generator);
               };
```

abassign, 2019-11-18:
Ciao @bcoconni 

I tried this version which should make the individual Random declarations independent, I made 4, in 4 independent functions, but the random value produced is always the same!

This is the code, the seed is necessary to prevent all random values for the 4 function are not identical ...

```
else if (operation == "random") {
        // abassign for random generator
        unsigned seed = std::chrono::system_clock::now().time_since_epoch().count();
        std::default_random_engine rnd_generator(seed);
        std::normal_distribution<double> rnd_distribution_normal(0.0,1.0);
        std::cout << "Random some Normal-distributed(0.0,1.0) seed:" << seed << std::endl;
        auto f = [rnd_generator,rnd_distribution_normal](const decltype(Parameters)& p)->double {
                    auto d = const_cast<std::normal_distribution<double>&>(rnd_distribution_normal);
                    auto g = const_cast<std::default_random_engine&>(rnd_generator);
                    std::map<int, int> hist{};
                    for(int n=0; n<10000; ++n) {
                        ++hist[std::round(d(g))];
                    }
                    for(auto p : hist) {
                        std::cout << std::setw(2) << p.first << ' ' << std::string(p.second/200, '*') << '\n';
                    }
                    std::cout << "Random some Normal-distributed(0.0,1.0) results:" << d(g) << std::endl;
                    return d(g);
      //auto f = [](const decltype(Parameters)& p)->double {
      //           return GaussianRandomNumber();
               };
```

This version make this output:

Only one time (creation step):

```
Random some Normal-distributed(0.0,1.0) seed:704209127
Random some Normal-distributed(0.0,1.0) seed:707451452
Random some Normal-distributed(0.0,1.0) seed:731604543
Random some Normal-distributed(0.0,1.0) seed:731667044
```

Prove that the 4 seeds created, for the four functions, are different. Or the builder seems to do his job well.

But unfortunately, during the execution phase ... each of the 4 random functions always generate the same result. Notice from the diagram. that the values produced within the single call <random> are well distributed, that is they are not constant:

```
Random some Normal-distributed(0.0,1.0) results:-1.09596
-4 
-3 
-2 **
-1 ************
 0 *******************
 1 ************
 2 **
 3 
 4 
Random some Normal-distributed(0.0,1.0) results:0.420707
-4 
-3 
-2 **
-1 ************
 0 *******************
 1 ***********
 2 **
 3 
 4 
Random some Normal-distributed(0.0,1.0) results:-0.0628796
-3 
-2 ***
-1 ************
 0 *******************
 1 ************
 2 **
 3 
 4 
Random some Normal-distributed(0.0,1.0) results:1.82365
-4 
-3 
-2 **
-1 ************
 0 *******************
 1 ***********
 2 ***
 3 
 4

This is a repetition:

Random some Normal-distributed(0.0,1.0) results:-1.09596
-4 
-3 
-2 **
-1 ************
 0 *******************
 1 ************
 2 **
 3 
 4 
```

At this point I have the feeling that the function is re-executed every time I call it.
As you know, I don't know C ++ well enough to understand its function calling methods, but I have the feeling that this is the problem.
I could do, for the function that is instanced, a single random generator, but I tried without ever succeeding. However I start to have doubts about the fact that we are working inside a Lambda function .. this could make the difference, even if I don't know how C ++ manages it.

What do you think about it ?

Regards
Adriano

bcoconni, 2019-11-19:
> But unfortunately, during the execution phase ... each of the 4 random functions always generate the same result.

Yes, right. I think the `const_cast` thing allows to cast the constness out but since the variables `d` and `g` are initialized at each iteration from const variables, we are getting back to square one each time.

In order to work around this limitation, our best option is to create the variables on the heap and pass pointers to the variables to the lambda function.
```c++
    } else if (operation == "random") {
      unsigned int seed = chrono::system_clock::now().time_since_epoch().count();
      shared_ptr<default_random_engine> generator(new default_random_engine(seed));
      shared_ptr<normal_distribution<double>> distribution(new normal_distribution<double>(0.0, 1.0));
      auto f = [generator, distribution](const decltype(Parameters)& p)->double {
                 return (*distribution.get())(*generator);
               };
      Parameters.push_back(new aFunc<decltype(f), 0>(f, fdmex, element, Prefix, var));
```
Usage of `shared_ptr<>` allows the variables to have exactly the same life span than the lambda function, avoiding us to write the code for their destruction.

It compiles and runs successfully on my laptop (i.e. a different number is returned each time the function is called). Let me know if that works for you too.

abassign, 2019-11-19:
@bcoconni FANTASTIC :) !!! 
I checked your code immediately, it works great! Finally I have a random number generator that can be used in various parts of the JSBSim program for the FIATG91R1B. In this case I tried it with the oscillation of the paraphrase that follows a typical Gaussian. I'm very happy.
I must admit that C ++ is a rather mysterious language for me, I am very sorry, but I don't have much time to go into this important programming language. But I see that it is possible to collaborate also in these parts of code that are difficult to understand. At this point I think you can complete the work with the other generator, but I wanted to ask you a question:
considering that the genrator of the <random> library has the possibility to insert 1 or two parameters .. do you think it is possible to do the same thing within the <random /> or <urandom /> function? Or you could make a more complete block that you can use as a random generator with a more complete approach and maybe with other distributions. I can do it too, it should be a style very similar to the block I made last year (linear actuator).
Tell me what you think.
Thank you for your cooperation!

closed 2019-11-19 by abassign