JSBSim-Team/jsbsim #235: Support: <ixz unit="SLUG*FT2"> inverted when flying

NikolaiVChr, 2019-11-13:
Hello. I have set:

`<ixz unit="SLUG*FT2"> 982 </ixz>`

But when I fly its has become inverted (-982).

So what should I set it to? The aerodynamic paper I took the value from, lists it as positive.
The paper uses same axes as JSBSIM XYZ and that is also the system all coefficients is listed in.

Here is the source ref.:
```
<reference refID="NASA TP-1538" author="Nguyen, L. T.; Ogburn, M. E.; Gilbert, W. P.; Kibler, K. S.; Brown, P. W.; Deal, P. L."
    title="Simulator study of stall/post-stall characteristics of a fighter airplane with relaxed longitudinal static stability." date="12/1979"/>
```

bcoconni, 2019-11-16:
Good catch ! The inertia `XZ` is used by JSBSim as read from the XML file (i.e. without reversing its sign):
https://github.com/JSBSim-Team/jsbsim/blob/82ea6c67b1b9dbb3139e8e1c1323c6a2158c6ae1/src/models/FGMassBalance.cpp#L117-L124
But the sign of the returned value of the `XZ` inertia is reversed
https://github.com/JSBSim-Team/jsbsim/blob/82ea6c67b1b9dbb3139e8e1c1323c6a2158c6ae1/src/models/FGMassBalance.h#L212
This is obviously incorrect.

In addition, I am concerned that the inverse of the inertia matrix is computed with the opposite of `mJ(1,3)` https://github.com/JSBSim-Team/jsbsim/blob/82ea6c67b1b9dbb3139e8e1c1323c6a2158c6ae1/src/models/FGMassBalance.cpp#L232-L252
I'll investigate that and push a commit once I'll have clarified the situation.

Also for the record, similar matters have been discussed in issue #125 and @rega0051 made a good point that the inertia tensor is not expressed in the structural frame as the inertia are computed with respect to the CG (which is not the origin of the structural frame).

bcoconni, 2019-11-17:
The bug fix has just been pushed to FlightGear. Please test and let me know if the bug is fixed.
> In addition, I am concerned that the inverse of the inertia matrix is computed with the opposite of `mJ(1,3)`

The computation of the inverse of the inertia matrix is correct. It uses the formula from Stevens & Lewis "Aircraft Control and Simulation" for an inertia matrix expressed in the body frame which is the case of the matrix `mJ` in JSBSim.

NikolaiVChr, 2019-11-17:
Thank you very much for fixing and clarifying. :)

bcoconni, 2019-11-17:
I guess the issue can now be closed ?

closed 2019-11-17 by NikolaiVChr

mike402, 2019-11-24:
By the way, why does the commit only change the sign of `GetIxz` and not `GetIxy` and `GetIyz` (apparently they output fine somehow, despite having inverted sign on the input as well)?

mike402, 2019-11-24:
Or is the right way to input inertia tensor in JSBSim to only invert the sign of Ixy and Iyz?

bcoconni, 2019-11-27:
@mike402 please read the discussion in the issue #125 to get answers about the signs in inertia tensors.