JSBSim-Team/jsbsim #238: Any possibility of getting a Python 3.6 wheel file for 64-bit Windows?

Phillip-M-Feldman, 2019-11-27:


bcoconni, 2019-11-27:
Yes, it is theoretically feasible. The AppVeyor setting file `appveyor.yml` should be modified for the library to be compiled with different versions of Python (and there are quite a number of them available in [AppVeyor](https://www.appveyor.com/docs/windows-images-software/#python)).

Another much more promising option would be to use [cibuildwheel](https://github.com/joerick/cibuildwheel) but this might need some tweaking in our `setup.py`.

bcoconni, 2019-11-30:
@Phillip-M-Feldman 
The wheel packages for 64 bits Windows Python 3.5, 3.6, 3.7 and 3.8 are now part of the [2019 rolling release](https://github.com/JSBSim-Team/jsbsim/releases/tag/Rolling-release-v2019).
Could you please check that it works ?

Phillip-M-Feldman, 2019-12-01:
The 64-bit Windows Python 3.6 package appears to work.  Thanks!

bcoconni, 2019-12-01:
Great ! Would you have any problem running this wheel package, please open a new issue.

closed 2019-12-01 by bcoconni