JSBSim-Team/jsbsim #240: "Testing JSBSim" in DeveloperDocs.md needs library path

kwan0xfff, 2019-12-01:
I've been following the instructions for DeveloperDocs.md to testing JSBSim.
https://github.com/JSBSim-Team/jsbsim/blob/master/doc/DevelopersDocs.md#testing-jsbsim
When I attempted to run `ctest` in the `build` directory, most of the tests failed.  In fact, the only ones succeeding were those in `tests/unit_tests`.

Peaking in Travis CI, I discovered the following a few steps before testing: 

    export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$PWD/src

When I insert this before `ctest`, the vast majority of tests pass. It turns out the `libJSBSim.so` symlink and its library reside under `build/src` at the end of the build.  Of course, this is specific to Linux; I don't know how this is handled in Windows.

 It may be worth including this line in a Linux example, such as:

    > export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$PWD/src
    > ctest -j4

The same description issue exists in the JSBSim Reference Manual, under "Testing JSBSim":
https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-the-program/

bcoconni, 2019-12-01:
Thanks for feedback @kwan0xfff

The instructions to test JSBSim in the developers docs have been modified to point to the FAQ question [When I try to run `ctest`, most of the tests fail](https://github.com/JSBSim-Team/jsbsim/blob/master/doc/DevelopersDocs.md#when-i-try-to-run-ctest-most-of-the-tests-fail).

Let me know if that is clear enough and I will update the JSBSim reference manual accordingly.

kwan0xfff, 2019-12-02:
I see. I think I would want to be "beat over the head" a bit more... :-)  A possible revision of the sentence:
"Make sure that the JSBSim shared library in `build/src` is in the path of the dynamic loader..."

For me, the term "dynamic loader" sounds like a function using run-time `dlopen()` rather than the load-time "dynamic linking" that `ld.so` does.  The term might be perfectly valid, but somehow I suspect some of us are going to miss it, which is why I threw in "shared library". (Curiously, the man pages for `dlopen(3)` and  `ld.so(8)` use the terms "dynamic linker" or "dynamic linker/loader", but not "dynamic loader".)

bcoconni, 2019-12-04:
I have just updated the docs according to your comments:
* Specified that the JSBSim library is shared.
* Use the term "dynamic linker" instead of "dynamic loader".

kwan0xfff, 2019-12-04:
Thank you. Much appreciated. (I assume I should close this as a happy issue reporter. :-) )

closed 2019-12-04 by kwan0xfff

bcoconni, 2019-12-04:
> I assume I should close this as a happy issue reporter. :-)

You are welcome 😄