JSBSim-Team/jsbsim #241: set airport in JSBSim-FlightGear

MrStick94, 2019-12-04:
hi, i'm trying to run a script with graphical output using flightgear but when i launch the script my aircraft is set in the sea. i changed the coordinates in initialization file to set they as the same coordinates of airport in flightgear but it still not work.
can you help me?

bcoconni, 2019-12-04:
Please check the [discussion "Spawn in the water"](https://forum.flightgear.org/viewtopic.php?f=17&t=32741) in the FlightGear forums.

kvnjonathan70, 2020-01-23:
i am also facing the same issue, auto download option is set ON for scenery. but i am not at airport. i copied KOAK's long and lat. still it didnt work.

bcoconni, 2020-02-10:
@kvnjonathan70 
Do you spawn in the water when using the very same coordinates with FlightGear alone (i.e. when not using a socket connection with JSBSim) ? If so, that means that the problem is with FlightGear and I would suggest that you request support on FlightGear forums.

Otherwise, could you please provide the exact command sequence that you are using to set up JSBSim and FlightGear ?

kvnjonathan70, 2020-04-10:
sorry for the late reply... the issue was simple,and was resolved by downloading the correct scenery file.. files got mismatch in fg root thus faced that issue.  i guess this issue can be closed now.

closed 2020-04-10 by bcoconni