JSBSim-Team/jsbsim #242: External force - Arduplane simulation

VinhA2018, 2019-12-05:
Hello all, I'm trying to add a new external force to my plane. I've added in the .xml configuration files the external force:

<external_reactions>
		<force name="rope" frame="BODY">
			 <location unit="M">
				 <x>0</x>
				 <y>0</y>
				 <z>0</z>
			 </location>
			 <direction>
				 <x>0</x>
				 <y>-999</y>
				 <z>0</z>
			 </direction>
		 </force>
</external_reactions>

This is intended to add a force to the center of mass of the drone, but when I'm simulating the flight, the plane doesn't seem to get that force applied.. The tragectory is the same as before adding this force... 

Am I doing/supposing anything wrong?

Thanks, Vinha

bcoconni, 2019-12-05:
Hi @VinhA2018,
> Am I doing/supposing anything wrong?

You did not specify the force magnitude which is 0 (zero) by default. So it is not surprising that you don't see any effect on your simulation. The force magnitude must be specified (in pound-force) via the property  `external_reactions/rope/magnitude`

The direction is normalized so the 2 specifications below are strictly equivalent and their norms are not interpreted as force magnitudes.
```xml
<direction>
  <x> 0.0 </x>
  <y> -1.0 </y>
  <z> 0.0 </z>
</direction>
```
```xml
<direction>
  <x> 0.0 </x>
  <y> -999.0 </y>
  <z> 0.0 </z>
</direction>
```

VinhA2018, 2019-12-06:
Hello again and thanks for your reply!

So, I need to write like this?
`<external_reactions>
		<force name="rope" frame="BODY">
			 <location unit="M">
				 <x>0</x>
				 <y>0</y>
				 <z>0</z>
			 </location>
			 <direction>
				 <x>0</x>
				 <y>-1</y>
				 <z>0</z>
			 </direction>
                         <magnitude>999</magnitude>
		 </force>
</external_reactions>`

bcoconni, 2019-12-07:
> So, I need to write like this?

No, the magnitude statement `<magnitude> 999 </magnitude>` will be ignored. You need to set the property `external_reactions/rope/magnitude` **after** the model is loaded. You cannot set it from the XML definition file.

I don't know how a JSBSim property can be set with ArduPilot, you should ask them.

VinhA2018, 2019-12-09:
In the SIM_JSBSIM.cpp file of ardupilot I added the property and it worked! Thanks guys!

```cpp
"    <event name=\"start force\">\n"
"      <condition> simulation/sim-time-sec le 0.01 </condition>\n"
"      <set name=\"external_reactions/rope/magnitude\" value=\"2\"/>\n"
"      <notify/>\n"
"    </event>\n"
```

However, I want to go further and I want this magnitude to be dependent of of the velocity of the airplane. Do I need to write this in the XML definitions?

bcoconni, 2019-12-09:
> In the SIM_JSBSIM.cpp file of ardupilot I added the property and it worked! Thanks guys!

Great ! You're welcome
> However, I want to go further and I want this magnitude to be dependent of of the velocity of the airplane. Do I need to write this in the XML definitions?

Yes. You need a function that computes the force magnitude.

There is an example in the file `jsbsim/aircraft/f16/Systems/pushback.xml`. The system is designed to apply a force which magnitude is driven by the property `external_reactions/pushback/magnitude`.
1. The first `<switch>` checks if a number of conditions are met before engaging the pushback. The result of the test is stored in `systems/pushback/linked`.
2. Then the `<summer>` computes the error between the current velocity and the target velocity. The value is stored in `systems/pushback/speed-error`.
3. The `<pid>` is a [PID controller](https://en.wikipedia.org/wiki/PID_controller) that computes how much force must be applied as a function of the error `systems/pushback/speed-error`.
4. The last `<switch>` finally copies the force computed by the PID to the external force magnitude `external_reactions/pushback/magnitude` if the condition `systems/pushback/linked` is true (different from 0)
https://github.com/JSBSim-Team/jsbsim/blob/9b15e85ce7c68d07c248619d60fcca41874a5f73/aircraft/f16/Systems/pushback.xml#L1-L41

bcoconni, 2020-02-10:
@VinhA2018 Has your request been answered ? Do you need further support ?

VinhA2018, 2020-02-11:
Yes! I has been answered, thank you!

closed 2020-02-11 by VinhA2018