JSBSim-Team/jsbsim #243: Get NaN for properties

opt12, 2019-12-12:
Currently I run into problems when trying to feed external controls into JSBSim.

I feed in commands for `fcs_elevator_cmd_norm`and `fcs_aileron_cmd_norm` which are indeed within the range of [-1...1].

The control signal are determined by a machine learning algorithm which behaves quite randomly in the beginning and admittedly, the flight is still far away from being "controlled".
Unfortunately, after a while I get `NaN` for all output properties from JSBSim which causes my ML algo to crash.

How does it come, that JSBSim retruns `NaN`for its properties?

Wouldn't it be good when `jsbsim.FGFDMExec().run()` would return `False` in that case?

Is there any way to catch or prevent the `NaN`-situations in advance? 

I don't really understand what happens here.

Regards,
Felix

bcoconni, 2019-12-14:
> The control signal are determined by a machine learning algorithm which behaves quite randomly in the beginning and admittedly, the flight is still far away from being "controlled".

Well, in that case my first guess would be that the aircraft spins out of control and crashes violently on the ground. JSBSim ground reactions code can handle correctly soft to mild impacts but it is not so good for hard/violent impacts since these would normally lead to the loss of structural integrity, something that JSBSim does not account for. 

I'd suggest for a start to rule out all the aircraft that are getting to close to the ground (property `position/h-agl-ft` close to 0.0).
> Is there any way to catch or prevent the NaN-situations in advance?

There is but it is not officially supported by JSBSim. During the compilation of the library, JSBSim builds a Python module named `fpectl` (it should be located in `build/tests/`). If you add the following lines to your code then an exception `FloatingPointError` will be raised whenever a NaN is generated.
```python
import fpectl

# Triggers the SIGFPE signal from the FPU
fpectl.turnon_sigfpe()

# Runs JSBSim code
fdm = jsbsim.FGFDMExec()

try:
  while fdm.run():
    pass
except FloatingPointError:
  print("Something wrong happened !")
```
The module `fpectl` might not work for your platform. The best way to tell is to check if the Python test `fpectl.py` succeeds when `ctest` is run. If the test fails then we'll have to tweak the module C++ source code https://github.com/JSBSim-Team/jsbsim/blob/master/python/fpectlmodule.cpp

opt12, 2020-01-08:
@bcoconni 
Thanks Bertrand for the sudden answer and the good support.

It looks like the NaN really occurred due to severe crashes my ML algo produced in the beginning of the learning phase.

After I added a check to end training episodes when the ground comes close (altitude approaches 0), I never had this problem again.

I did not try to catch the exception instead as the easy solution works for the moment.

bcoconni, 2020-01-09:
> Thanks Bertrand for the sudden answer and the good support.

You're welcome 😄 Should we consider this issue closed then ?

closed 2020-01-17 by opt12