JSBSim-Team/jsbsim #244: Turbine augmethod zero and reheat amount

NikolaiVChr, 2019-12-12:
We are modeling an older turbine jet from a supersonic plane.

In doubt what the best method is for controlling augmentation amount.

In the JA-37 I put a factor as product to the augmentation table inside engine file. But not sure that is overly smart, as the factor probably has to be a table found by experimentation. Is there a better way to control augmentation amount when using 'augmethod' 0? Property augmentation is a boolean, right?

For those interested here is how the 2 methods of augmentation in this engine works:

[image]
[image]

And here is a variant of that engine for JSBSim:
```xml
<?xml version="1.0"?>

<!--
    name:                    Adour 804
    type:                    Turbine Engine
    mil thrust:              5260.0 lbf
    idl thrust:              600.0 lbf
    aug thrust:              8000.0 lbf
    bypass ratio:            0.85:1
    overall pressure ratio:  30.0:1
    compression ratio:       11.0:1
    augmented?               yes
    injected?                no
-->

<turbine_engine name="Adour 804">
  <milthrust>    5260.0 </milthrust>
  <maxthrust>    8000.0 </maxthrust>
  <bypassratio>     0.850 </bypassratio>
  <tsfc>            0.650 </tsfc>
  <atsfc>           2.363 </atsfc>
  <bleed>           0.03</bleed>
  <idlen1>         30.0 </idlen1>
  <idlen2>         55.0 </idlen2>
  <maxn1>         100.0 </maxn1>
  <maxn2>         100.0 </maxn2>
  <!--<ignitionn1>        5.0 </ignitionn1>-->
  <!--<ignitionn2>       15.0 </ignitionn2>-->
  <n1spinup>        1.0 </n1spinup>
  <n2spinup>        3.0 </n2spinup>
  <n1startrate>     1.4 </n1startrate>
  <n2startrate>     2.0 </n2startrate>
  <augmented>         1 </augmented>
  <augmethod>         2 </augmethod>
  <injected>          0 </injected>

  <function name="IdleThrust">
    <product>
       <v>2.336</v>
       <table>
        <independentVar lookup="row">velocities/mach</independentVar>
        <independentVar lookup="column">atmosphere/density-altitude</independentVar>
        <tableData>
             -10000     0     10000   20000   30000   40000   50000   60000
         0.0  0.0430  0.0488  0.0528  0.0694  0.0899  0.1183  0.1467  0
         0.2  0.0500  0.0501  0.0335  0.0544  0.0797  0.1049  0.1342  0
         0.4  0.0040  0.0047  0.0020  0.0272  0.0595  0.0891  0.1203  0
         0.6 -0.0804 -0.0804 -0.0560 -0.0237  0.0276  0.0718  0.1073  0
         0.8 -0.2129 -0.2129 -0.1498 -0.1025  0.0474  0.0868  0.0900  0
         1.0 -0.2839 -0.2839 -0.1104 -0.0469 -0.0270  0.0552  0.0800  0
        </tableData>
       </table>
    </product>
  </function>

  <function name="MilThrust">
   <table>
    <independentVar lookup="row">velocities/mach</independentVar>
    <independentVar lookup="column">atmosphere/density-altitude</independentVar>
    <tableData>
          -10000       0   10000   20000   30000   40000   50000   60000
      0.0  1.2600  1.0000  0.7400  0.5340  0.3720  0.2410  0.1490  0.0000
      0.2  1.1491  0.9165  0.6840  0.4965  0.3484  0.2267  0.1403  0.0000
      0.4  1.1070  0.8866  0.6661  0.4871  0.3436  0.2243  0.1396  0.0000
      0.6  1.1147  0.8976  0.6806  0.5022  0.3568  0.2341  0.1454  0.0000
      0.8  1.1639  0.9437  0.7235  0.5385  0.3858  0.2544  0.1573  0.0000
      1.0  1.2410  1.0153  0.7896  0.5901  0.4306  0.2855  0.1768  0.0000
      1.2  1.3184  1.0920  0.8656  0.6605  0.4839  0.3232  0.1998  0.0000
      1.4  1.3854  1.1646  0.9438  0.7344  0.5458  0.3685  0.2286  0.0000
    </tableData>
   </table>
  </function>

  <function name="AugThrust">
   <table>
    <independentVar lookup="row">velocities/mach</independentVar>
    <independentVar lookup="column">atmosphere/density-altitude</independentVar>
    <tableData>
           -10000       0   10000   20000   30000   40000   50000   60000
     0.0    1.1816  1.0000  0.8184  0.6627  0.5280  0.3756  0.2327  0
     0.2    1.1308  0.9599  0.7890  0.6406  0.5116  0.3645  0.2258  0
     0.4    1.1150  0.9474  0.7798  0.6340  0.5070  0.3615  0.2240  0
     0.6    1.1284  0.9589  0.7894  0.6420  0.5134  0.3661  0.2268  0
     0.8    1.1707  0.9942  0.8177  0.6647  0.5309  0.3784  0.2345  0
     1.0    1.2411  1.0529  0.8648  0.7017  0.5596  0.3983  0.2467  0
     1.2    1.3287  1.1254  0.9221  0.7462  0.5936  0.4219  0.2614  0
     1.4    1.4365  1.2149  0.9933  0.8021  0.6360  0.4509  0.2794  0
     1.6    1.5711  1.3260  1.0809  0.8700  0.6874  0.4860  0.3011  0
     1.8    1.7301  1.4579  1.1857  0.9512  0.7495  0.5289  0.3277  0
     2.0    1.8314  1.5700  1.3086  1.0474  0.8216  0.5786  0.3585  0
     2.2    1.9700  1.6900  1.4100  1.2400  0.9100  0.6359  0.3940  0
     2.4    2.0700  1.8000  1.5300  1.3400  1.0000  0.7200  0.4600  0
     2.6    2.2000  1.9200  1.6400  1.4400  1.1000  0.8000  0.5200  0
    </tableData>
   </table>
  </function>

</turbine_engine>
```

mike402, 2019-12-14:
Hi, I think the only way so far would be to kludge it around by splitting the AB into a separate "engine", whose MilThrust would be (AugThrust - MilThrust).

bcoconni, 2019-12-14:
> In the JA-37 I put a factor as product to the augmentation table inside engine file. But not sure that is overly smart, as the factor probably has to be a table found by experimentation.

Might not be relevant but in the code you provided the factor is applied to the table `IdleThrust` and not to the table `AugThrust` as you mentioned ?
```xml
  <function name="IdleThrust">
    <product>
       <v>2.336</v>
       <table>
        <independentVar lookup="row">velocities/mach</independentVar>
        <independentVar lookup="column">atmosphere/density-altitude</independentVar>
        <tableData>
             -10000     0     10000   20000   30000   40000   50000   60000
         0.0  0.0430  0.0488  0.0528  0.0694  0.0899  0.1183  0.1467  0
         0.2  0.0500  0.0501  0.0335  0.0544  0.0797  0.1049  0.1342  0
         0.4  0.0040  0.0047  0.0020  0.0272  0.0595  0.0891  0.1203  0
         0.6 -0.0804 -0.0804 -0.0560 -0.0237  0.0276  0.0718  0.1073  0
         0.8 -0.2129 -0.2129 -0.1498 -0.1025  0.0474  0.0868  0.0900  0
         1.0 -0.2839 -0.2839 -0.1104 -0.0469 -0.0270  0.0552  0.0800  0
        </tableData>
       </table>
    </product>
  </function>
```

NikolaiVChr, 2019-12-14:
> Might not be relevant but in the code you provided the factor is applied to the table `IdleThrust` and not to the table `AugThrust` as you mentioned ?

Oh, sorry I should have explained. The JA37 is another aircraft I have worked on.
The factor on idle thrust here is to get idle thrust to 600.

closed 2020-12-11 by bcoconni