JSBSim-Team/jsbsim #245: Modelling a diamond kite ? Tether Qs ?

linuxguy123, 2019-12-15:
I want to model a tethered diamond kite in JSBSim.

I want to see how high it flies and how far downwind it gets pulled in various winds, with various loads. Is this doable in JSBSim ?

How is a tether modelled in JSBSim ? Does it sag and have drag like a real tether does ?

Is there a better place to ask JSBSim questions ?

Thanks

linuxguy123, 2019-12-15:
Lets consider an easier case - a helium balloon with fixed buoyancy being held by a string of negligible mass and drag, such that it is a straight line.  How would one model this is JSBSim ?

bcoconni, 2019-12-16:
> Lets consider an easier case - a helium balloon with fixed buoyancy being held by a string of negligible mass and drag, such that it is a straight line. How would one model this is JSBSim ?

I would suggest starting from the weather balloon example: https://github.com/JSBSim-Team/jsbsim/blob/1081fb99f3aeeb99afd44cb2083dc29434d99fcd/aircraft/weather-balloon/weather-balloon.xml#L1-L394
It probably needs a bit of clean up to remove the features you don't need. 

Then the action of the string could be modelled by an `<external_reactions>` that would act as a spring (i.e. a force that is proportional to the distance between the balloon and a fixed point).

bcoconni, 2020-02-10:
@linuxguy123 Has your request been answered ? Do you need further support ?

linuxguy123, 2020-02-10:
I figured it out.  Thanks for following up.

bcoconni, 2020-02-10:
Great ! Issue closed.

closed 2020-02-10 by bcoconni