JSBSim-Team/jsbsim #246: What exactly is a FGColumnVector3 ?

linuxguy123, 2019-12-16:
What exactly is a FGColumnVector3 ?     A vector from 1 point to another ?

bcoconni, 2019-12-16:
`FGColumnVector3` is a 3D [column vector](https://en.wikipedia.org/wiki/Row_and_column_vectors). Its class definition is available in the [developers documentation of the C++ classes](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGColumnVector3.html).

linuxguy123, 2019-12-16:
Thanks for the reply.

I saw the class definition, but I don't understand, in English, what it's
purpose is.  Is is a directional unit vector ?   Or is a magnitude vector
?  I think I've seen it used as both ?

LG

bcoconni, 2019-12-17:
> what it's purpose is.  Is is a directional unit vector ?   Or is a magnitude vector ?  I think I've seen it used as both ?

The class itself is nothing more than the mathematical object "vector" and it is used to store physical quantities such as positions, velocities, accelerations and all that sort of things

linuxguy123, 2019-12-18:
Thank you for clarifying that.  I get it now.

closed 2019-12-18 by bcoconni