JSBSim-Team/jsbsim #247: JSBSim fails to get host by number for flightgear port over LAN when run in WSL

brisk0, 2020-01-15:
Problem
---------
JSBSim reports "Could not get host address by number..." instead of sending packets when attempting to output using flightgear protocol

Set-up / Testing
----------------
I am running an aircraft including the line:
    <output name="{LAN ipv4 address of host computer}" type="FLIGHTGEAR" port="7127" protocol="UDP" rate="200"/>
Tested by having the host computer (at address listed) running `ncat -l -u -p 7127`
Data was sent successfully using `nc -u {host LAN address} 7127`
JSBSim running on the same machine reports "Could not get host net address by number..."
Tested running JSBSim through ardupilot SITL integration and running JSBSim directly (compiled from source) using the same aircraft file.

Environment
-------------
WSL running Ubuntu 18.04 (bionic) on Windows 10
Local area network on WiFi, private network with firewall off

Solution
---------
My local man pages report that the used `gethostfromaddr()` and related functions are deprecated in favour of `getaddrinfo()`. Doing a relatively simple drop in replacement successfully sends packets (seen both in netcat and in a personal visualiser).

Diff of my personal fix (haven't attempted to make robust):

[diff.txt](https://github.com/JSBSim-Team/jsbsim/files/4062563/diff.txt)

bcoconni, 2020-01-16:
Thanks @brisk0 for the detailed bug report.

Man pages indeed report that `gethostbyname` and `gethostbyaddr` are now obsolete (checked on Fedora 31). Now the question is: shall we continue to support `gethostbyname` and `gethostbyaddr` for platforms that might not have yet adopted the new functions `getaddrinfo` and `getnameinfo` ? Does someone know since when the new functions have been introduced ?

bcoconni, 2020-01-18:
@brisk0 I have just pushed some code that replaces `gethostbyname` and `gethostbyaddr` by `getaddrinfo` as per your suggestion.

Please test and let me know if that fixes the problem you have reported.

brisk0, 2020-01-20:
The push resolves the issue, cheers.

closed 2020-01-20 by brisk0