JSBSim-Team/jsbsim #248: Why there's no integration scheme such as rk4 in jsbsim?

2012ZGZYY, 2020-01-15:
And I find it difficult to add such a scheme into the current jsbsim code. Can anyone give me some advice？

bcoconni, 2020-01-16:
Regarding why there is no RK4 scheme in JSBSim, the short answer is because nobody has ever needed it (yet).

JSBSim uses Adams-Bashforth for the equations integration and Adams-Bashforth 4 is a 4th order explicit integrator just as RK4 is. The only restriction is that the initial conditions of Adams-Bashforth are a bit more involved since it is a multiple steps scheme. In most cases however this is not relevant since the errors introduced by the approximate initialization are quickly damped and have therefore a negligible effect on the results. Even for mildly "hard" problems such as ground contacts, Adams-Bashforth behaves quite well thanks to the use of the projected Gauss Seidel algorithm to solve friction forces.

Now, why is it difficult to implement RK4 in JSBSim ? This is mostly due to the fact that JSBSim is entirely designed around the assumption that time propagation is immediately processed at each iteration: assuming that the equation to solve is `ydot = f(t, y)` then the propagation of PIDs, accelerators, actuators immediately are taking place while `f(t, y)` is evaluated.
The problem is that RK4 needs 4 evaluations of `f(t, y)` without any propagation. Unfortunately JSBSim is not designed for that. There will be plenty of accumulators, PIDs, actuators, etc. that will propagate in your back each time you try to evaluate `f(t, y)`.

That being said, is it feasible to add RK4 to JSBSim ? Yes, but that will not be easy and that's an understatement. While you evaluate `f(t, y)`, you will need to tell all the integrators that they must not propagate. And when RK4 has completed the integration of the current time step then you will have to tell each integrator to propagate. This needs an observer pattern for each integrator which will make the code significantly more complex. Is it worth it ? I mean have you a compelling reason to add RK4 to JSBSim ? I'm intrigued by the motivation behind your request 😄

bcoconni, 2020-02-10:
@2012ZGZYY Has your request been answered ? Do you need further support ?

2012ZGZYY, 2020-02-11:
> @2012ZGZYY Has your request been answered ? Do you need further support ?

Thank you so much！

closed 2020-02-11 by 2012ZGZYY