JSBSim-Team/jsbsim #251: FGPropeller: compute partial derivative of thrust w.r.t. engine power

EliaTarasov, 2020-01-22:
Does it possible to compute partial derivative of thrust w.r.t. engine power?
From the code of FGPropeller the math is:

1. Thrust(omega_in) = C_T * omega_in^2 * D^4 * rho
2. omega_out = omega_in + [(EnginePower - RequiredPower(omega_in))/(2* pi* omega_in*I_xx) ]*dt
3. omega_in = omega_out

The main problem for me is that EnginePower does not directly relate to Thrust.
I don't have any idea how to compute it, symbolically or numerically.

@bcoconni could you please give any hint?
Thanks!

Prepar3DGuy, 2020-01-22:
EnginePower affects on dynamic of propeller rotation as you mentioned in 2 and 3.
<a href="https://www.codecogs.com/eqnedit.php?latex=I_{xx}\frac{d\omega}{dt}=\frac{1}{\omega}\left&space;(&space;P_{engine}&space;-&space;P_{req}\right&space;)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?I_{xx}\frac{d\omega}{dt}=\frac{1}{\omega}\left&space;(&space;P_{engine}&space;-&space;P_{req}\right&space;)" title="I_{xx}\frac{d\omega}{dt}=\frac{1}{\omega}\left ( P_{engine} - P_{req}\right )" /></a> 
Where 
<a href="https://www.codecogs.com/eqnedit.php?latex=\omega&space;=&space;2\pi&space;n" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\omega&space;=&space;2\pi&space;n" title="\omega = 2\pi n" /></a> - angular velocity of propeller in radians per second
<a href="https://www.codecogs.com/eqnedit.php?latex=P_{engine}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?P_{engine}" title="P_{engine}" /></a> - input power in Watts from power source  (free turbine, reciprocating engine, electric, etc.)
<a href="https://www.codecogs.com/eqnedit.php?latex=P_{req}&space;=&space;C_p\rho&space;n^3D^5" target="_blank"><img src="https://latex.codecogs.com/gif.latex?P_{req}&space;=&space;C_p\rho&space;n^3D^5" 
title="P_{req} = C_p\rho n^3D^5" /></a> - power that propeller absorbs (use for thrust and drag) on current advanced ratio and propeller rotation speed.
The propeller thrust at the same contidions are <a href="https://www.codecogs.com/eqnedit.php?latex=T&space;=&space;C_T\rho&space;n^2D^4" target="_blank"><img src="https://latex.codecogs.com/gif.latex?T&space;=&space;C_T\rho&space;n^2D^4" title="T = C_T\rho n^2D^4" /></a>
You must to know that propeller dimensionless coefficients depends from the current rpm <a href="https://www.codecogs.com/eqnedit.php?latex=C_t,&space;C_P&space;=&space;f(J,pitch,M_{tip})" target="_blank"><img src="https://latex.codecogs.com/gif.latex?C_t,&space;C_P&space;=&space;f(J,pitch,M_{tip})" title="C_T, C_P = f(J,pitch,M_{tip})" /></a>
Where <a href="https://www.codecogs.com/eqnedit.php?latex=J&space;=&space;\frac{V_a}{nD}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?J&space;=&space;\frac{V_a}{nD}" title="J = \frac{V_a}{nD}" /></a> - advanced ratio.
If and only if your C_T, C_P are constant you can express n for steady-state of rotation differencial equation as
<a href="https://www.codecogs.com/eqnedit.php?latex=n&space;=&space;\sqrt[3]{\frac{P_{engine}}{C_P&space;\rho&space;D^5}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?n&space;=&space;\sqrt[3]{\frac{P_{engine}}{C_P&space;\rho&space;D^5}}" title="n = \sqrt[3]{\frac{P_{engine}}{C_P \rho D^5}}" /></a>
Substitute this in the thrust equeation you get
<a href="https://www.codecogs.com/eqnedit.php?latex=T(P_{engine})&space;=&space;(P_{engine})^{2/3}C_T\sqrt[3]{\frac{\rho&space;D^2}{C_P^2}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?T(P_{engine})&space;=&space;(P_{engine})^{2/3}C_T\sqrt[3]{\frac{\rho&space;D^2}{C_P^2}}" title="T(P_{engine}) = (P_{engine})^{2/3}C_T\sqrt[3]{\frac{\rho D^2}{C_P^2}}" /></a>
And partial derivative with respect to (w.r.t.?) engine power will be
<a href="https://www.codecogs.com/eqnedit.php?latex=\frac{\partial&space;T}{\partial&space;P_{engine}}&space;=&space;\frac{2}{3}\frac{C_T&space;\rho^{1/3}}{(C_P)^{2/3}}D^{2/3}\frac{1}{\sqrt[3]{P_{engine}}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\frac{\partial&space;T}{\partial&space;P_{engine}}&space;=&space;\frac{2}{3}\frac{C_T&space;\rho^{1/3}}{(C_P)^{2/3}}D^{2/3}\frac{1}{\sqrt[3]{P_{engine}}}" title="\frac{\partial T}{\partial P_{engine}} = \frac{2}{3}\frac{C_T \rho^{1/3}}{(C_P)^{2/3}}D^{2/3}\frac{1}{\sqrt[3]{P_{engine}}}" /></a>

But C_T and C_P on very low J are not constant 
[image]
Diagram for propeller's coefficient of thrust and power for low value of advanced ratio. (Александров В. Л. Воздушные Винты М. 1951)

EliaTarasov, 2020-01-22:
Thanks!
I will check this, looks very promising.
Just two tips are not clear for me:
1. 

> If and only if your C_T, C_P are constant you can express n for steady-state of rotation differencial equation

Maybe the term 'instant' is better than steady-state?

2. Why do we need C_T and C_P to be constant for the whole domain of J? Given the fact we know instant J, we can get C_T and C_P and just substitute them into final partial differential equation.

bcoconni, 2020-01-25:
> Maybe the term 'instant' is better than steady-state?

@Prepar3DGuy assumes that

<a href="https://www.codecogs.com/eqnedit.php?latex=P_{engine}=P_{req}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?P_{engine}=P_{req}" title="P_{engine}=P_{req}" /></a>

which is only true if

<a href="https://www.codecogs.com/eqnedit.php?latex=I\frac{d\omega}{dt}=0" target="_blank"><img src="https://latex.codecogs.com/gif.latex?I\frac{d\omega}{dt}=0" title="I\frac{d\omega}{dt}=0" /></a>

i.e. if we are in a steady state.
> Why do we need C_T and C_P to be constant for the whole domain of J? Given the fact we know instant J, we can get C_T and C_P and just substitute them into final partial differential equation.

The relation
<a href="https://www.codecogs.com/eqnedit.php?latex=n&space;=&space;\sqrt[3]{\frac{P_{engine}}{C_P&space;\rho&space;D^5}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?n&space;=&space;\sqrt[3]{\frac{P_{engine}}{C_P&space;\rho&space;D^5}}" title="n = \sqrt[3]{\frac{P_{engine}}{C_P \rho D^5}}" /></a>
is only valid if C_P does not depend on `n`. However it does since C_P is a function of `J` which itself is a function of `n`. So ultimately C_P is a function of `n`.

However the strict condition is that C_P must be independent from `n` i.e. if

<a href="https://www.codecogs.com/eqnedit.php?latex=\frac{\partial&space;C_P}{\partial&space;n}=0" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\frac{\partial&space;C_P}{\partial&space;n}=0" title="\frac{\partial C_P}{\partial n}=0" /></a>

Also note that C_T does not need to be independent from `n` for the final relation to be valid.

EliaTarasov, 2020-01-25:
@Prepar3DGuy @bcoconni  Thanks for the explanation. 
So summarises it all:
It is possible to compute the partial derivative of the thrust w.r.t. engine power only if dynamics of the propeller is neglected and even in this case it is possible only when J is not too low to allow C_T to be constant? So in general it is not possible?

EliaTarasov, 2020-02-04:
I finally gave up and end up just with interpolation of the motor's table for power and torque.
Thanks!

closed 2020-02-04 by EliaTarasov