JSBSim-Team/jsbsim #252: Use of uninitialized variables for in-air starts

ermarch, 2020-02-01:
Hi,

It turns out that trimming using tFull instead of tGround results in the use of unitialized variables in FlightGear.

I've tracked the cause down to FGTrim::DoTrim not initializing fdmex.

The following small patch should fix it.
But since it is in the trimming code I thought it would be a good idea to let others double-check:
[http://adalin.com/ehtw.info/inair-init.diff](url)

bcoconni, 2020-02-01:
I am a bit skeptical that this patch would make any difference since `FGTrimAxis::Run() `calls unconditionally this sequence:
```c++
fdmex->Initialize(&fgic);
fdmex->Run();
```
If you look a few lines below to where you suggest some code modification, you will see that `FGTrimAxis::Run()` is called in line 227 below
https://github.com/JSBSim-Team/jsbsim/blob/2f2398cbafc0fc4f14dc6157d1923548ec8804ba/src/initialization/FGTrim.cpp#L220-L227
and that `FGTrimAxis::Run()` sets the initial conditions in lines 256 and 257 below
https://github.com/JSBSim-Team/jsbsim/blob/2f2398cbafc0fc4f14dc6157d1923548ec8804ba/src/initialization/FGTrimAxis.cpp#L245-L257

ermarch, 2020-02-03:
You are right, I've traced the calls to fdmex->Initialize() and it is being called when I expect it too.
Sorry for the confusion.

closed 2020-02-03 by ermarch