JSBSim-Team/jsbsim #253: [Feature Request] Quadrotor model

acxz, 2020-02-05:
Hello! I would like to preface this issue by saying you guys are doing some amazing work (and have been for a long while)! I have used many projects that use JSBSIM as a backend and now I would like to use it directly.

I would like to do some work on a multirotor configuration specifically a quadrotor. However when looking under the examples directory I could not find any prebuilt configurations. I think now that the industry is more and more focusing on rotorcraft, having a quadrotor configuration would be very useful and even open JSBSim to more users.

Thank you!

bcoconni, 2020-02-10:
@acxz Thanks for the positive feedback.
> I think now that the industry is more and more focusing on rotorcraft, having a quadrotor configuration would be very useful and even open JSBSim to more users.

I guess the first step would be to provide some specifications (dimensions, weight, performance, power, etc.) for the quadrotor model you are looking for. Such specifications need not to be much detailed for a start. Maybe someone could then kindly design the model of your quadrotor on that base ?

The other option would be to take over the task yourself in which case we can provide some guidance to help you create your model.

acxz, 2020-02-11:
Thx for responding! 

I didn't have a particular quadrotor model in mind. Although I think a popular one is (used to be apparently, not sold anymore) is a Parrot AR Drone: https://www.parrot.com/global/drones/parrot-ardrone-20-elite-edition This would be a good drone to model off of you.

As for doing it myself, I wouldn't mind but I have 0 experience in modeling in JSBSIM and the project that I am considering this for is still a far ways away. I thought putting a feature request as early as possible would be a good idea and also to gauge it's interest.

The reason I am posing this FR in the first place, is that I would like a high fidelity aero physics engine, but without a lot of overhead. Airsim and Flightgoggles appealed to me at first, but I do not desire photo-realism or vivid graphics, just a good physics engine for drones.

bcoconni, 2020-02-11:
> I thought putting a feature request as early as possible would be a good idea and also to gauge it's interest.

This topic is indeed an interesting subject but I have not myself much free time to work on this at the moment. So I have added the `help wanted` label to your issue which will hopefully attract candidates to work on this topic.
> I would like a high fidelity aero physics engine, but without a lot of overhead

The specifications you provided are much too vague to build a high fidelity model. High fidelity requires detailed specifications as well as performance data measurements from the real quadcopter. With the current state of your specifications, you can hope at best for an approximation/educated guess. But even such a simple model could be the foundation for a high fidelity model.

acxz, 2020-02-12:
I see, thank you!

rega0051, 2020-05-11:
@acxz I've made slow progress building an F450 quad model. I don't have a good way validate the model yet. The aero model is simple, ground contacts are as measured, mass properties are probably good enough based on measurement of components, the prop Ixx, CT and CP are likely good but I don't have a good estimate of p-factor for the prop.

The next step for the model is to complete the autopilot model.

https://github.com/UASLab/OpenFlightSim/tree/master/Simulation/aircraft/F450

Jaeyoung-Lim, 2020-10-02:
@acxz PX4 already provides a quadrotor model in px4-jsbsim-bridge

If you clone the PX4 Firmware, you can simply run the quadrotor with PX4 in JSBSim
```
make px4_sitl jsbsim_quadrotor_x
```

If you have never used PX4, you can follow the setup guide: https://dev.px4.io/master/en/setup/dev_env.html

rega0051, 2020-10-21:
I'm really grad that the bridge to px4 is being worked again. Badly needed as the other SITL/HITL options have been historically really poor quality and lacking realism. The link I referenced earlier for the F450 is part of our "bridge" to our own flight computer (the other part of that repo has to do with generating the source data for the models).

I've never understood why multi-rotor models tend to use "external forces" rather than the propeller model included with JSBSim. The quad model in paparazzi uses a similar approach, but I believe I've seen earlier examples of that approach, never any rationale.  Propeller CT and CP values are pretty easily available, and by using the JSBSim prop model the inertial and gyroscopic generalized forces are applied properly. Further, altering the models is much easier if the source data is easily reproduced.

bcoconni, 2020-10-21:
> I've never understood why multi-rotor models tend to use "external forces" rather than the propeller model included with JSBSim. The quad model in paparazzi uses a similar approach, but I believe [...]

@rega0051, issue #333 has been created about that very topic by @seanmcleod. He included a couple of testimonies in his initial post that shed some light on the motivation behind the usage of external forces over JSBSim built-in propellers.

seanmcleod, 2020-10-22:
@rega0051 I'm busy modifying the minimum rotor inertia allowed as mentioned in issue #333 and I'm using your [F450 drone](https://github.com/UASLab/OpenFlightSim/tree/master/Simulation/aircraft/F450) for some of my testing and noticed that your propeller inertia you have specified for the DJI 9450 propellers is smaller than the current minimum of `0.001 SLUG*FT2`.

```xml
<ixx unit="KG*M2"> 6.05e-05 </ixx>
```
Works out to `4.4618493443662699e-05 SLUG*FT2`

seanmcleod, 2020-10-22:
Which makes a massive difference in performance 😉 

Using your `jsb_F450_Launch.xml` script with the propeller inertia limited to `0.001 SLUG*FT2` versus the specified value of `4.46e-05 SLUG*FT2`.

At time 1.0 for both cases:

```sh
Repeating Notify (Event 4) executed at time: 1.000000
    fcs/throttle-cmd-norm = 0.500000
    fcs/throttle-cmd-norm[1] = 0.500000
    fcs/throttle-cmd-norm[2] = 0.500000
    fcs/throttle-cmd-norm[3] = 0.500000
    propulsion/engine[0]/thrust-lbs = 0.000000
    propulsion/engine[1]/thrust-lbs = 0.000000
    propulsion/engine[2]/thrust-lbs = 0.000000
    propulsion/engine[3]/thrust-lbs = 0.000000
    position/h-agl-ft = 0.487024
    velocities/vc-kts = 0.093671
    attitude/phi-rad = 0.000000
    attitude/theta-rad = 0.000000
    attitude/psi-rad = 1.570796
```

1s later the drone has climbed to 3.7ft with a propeller inertia of `0.001`

```sh
Repeating Notify (Event 4) executed at time: 2.005000
    fcs/throttle-cmd-norm = 0.500000
    fcs/throttle-cmd-norm[1] = 0.500000
    fcs/throttle-cmd-norm[2] = 0.500000
    fcs/throttle-cmd-norm[3] = 0.500000
    propulsion/engine[0]/thrust-lbs = 1.121508
    propulsion/engine[1]/thrust-lbs = 1.121508
    propulsion/engine[2]/thrust-lbs = 1.121508
    propulsion/engine[3]/thrust-lbs = 1.121508
    position/h-agl-ft = 3.752105
    velocities/vc-kts = 5.001246
    attitude/phi-rad = 0.000000
    attitude/theta-rad = -0.000000
    attitude/psi-rad = 1.570796
```

Versus 223.6ft with a propeller inertia of `4.46e-05`.

```sh
Repeating Notify (Event 4) executed at time: 2.005000
    fcs/throttle-cmd-norm = 0.500000
    fcs/throttle-cmd-norm[1] = 0.500000
    fcs/throttle-cmd-norm[2] = 0.500000
    fcs/throttle-cmd-norm[3] = 0.500000
    propulsion/engine[0]/thrust-lbs = 0.587068
    propulsion/engine[1]/thrust-lbs = 0.587068
    propulsion/engine[2]/thrust-lbs = 0.587068
    propulsion/engine[3]/thrust-lbs = 0.587068
    position/h-agl-ft = 223.603379
    velocities/vc-kts = 103.069004
    attitude/phi-rad = 0.000040
    attitude/theta-rad = -0.000002
    attitude/psi-rad = 1.570793
```

seanmcleod, 2020-10-23:
The results are quite sensitive to the simulation rate. The results shown above were based on the 200Hz rate specified in `jsb_F450_Launch.xml`, when I increase the simulation rate to 1,000Hz these are the results with a propeller inertia of `4.46e-05`.

So ~55ft versus 224ft AGL roughly 1s after power is applied.

```sh
Repeating Notify (Event 4) executed at time: 2.001000
    fcs/throttle-cmd-norm = 0.500000
    fcs/throttle-cmd-norm[1] = 0.500000
    fcs/throttle-cmd-norm[2] = 0.500000
    fcs/throttle-cmd-norm[3] = 0.500000
    propulsion/engine[0]/thrust-lbs = 1.353860
    propulsion/engine[1]/thrust-lbs = 1.353860
    propulsion/engine[2]/thrust-lbs = 1.353860
    propulsion/engine[3]/thrust-lbs = 1.353860
    position/h-agl-ft = 55.441271
    velocities/vc-kts = 40.229328
    attitude/phi-rad = 0.000000
    attitude/theta-rad = -0.000000
    attitude/psi-rad = 1.570796

Repeating Notify (Event 4) executed at time: 2.101000
    fcs/throttle-cmd-norm = 0.500000
    fcs/throttle-cmd-norm[1] = 0.500000
    fcs/throttle-cmd-norm[2] = 0.500000
    fcs/throttle-cmd-norm[3] = 0.500000
    propulsion/engine[0]/thrust-lbs = 1.331312
    propulsion/engine[1]/thrust-lbs = 1.331312
    propulsion/engine[2]/thrust-lbs = 1.331312
    propulsion/engine[3]/thrust-lbs = 1.331312
    position/h-agl-ft = 62.400123
    velocities/vc-kts = 41.023862
    attitude/phi-rad = 0.000000
    attitude/theta-rad = -0.000000
    attitude/psi-rad = 1.570796
```

bcoconni, 2020-10-23:
Interesting. This confirms how important are the time step and propellers inertia.

seanmcleod, 2020-10-23:
@rega0051 I pushed a commit today to JSBSim to lower the allowable minimum propeller inertia so that your F450 model's propeller inertia will be used rather than being artificially capped at 0.001 slugs*ft2.

moreba1, 2020-11-27:
does it possible to run px4 simulation with jsbsim in windows?

Jaeyoung-Lim, 2020-11-27:
@moreba1 I don't think so, as far as I am concerned nobody tried

seanmcleod, 2020-11-27:
@moreba1 are you asking specifically about px4-jsbsim-bridge? A work colleague of mine has used PX4 in SITL mode on Windows with AirSim in the past.

moreba1, 2020-11-28:
> @moreba1 are you asking specifically about px4-jsbsim-bridge? A work colleague of mine has used PX4 in SITL mode on Windows with AirSim in the past.

I will simulate Flight of a plane in px4 with jsbsim using cgywin in windows not linux.

moreba1, 2020-11-28:
> 
> 
> @moreba1 I don't think so, as far as I am concerned nobody tried

can you guide me?

Jaeyoung-Lim, 2020-11-28:
@moreba1 Unfortunately not, I dont have access to windows

bcoconni, 2020-11-28:
@moreba1, we need to keep this issue focused on its original topic which is to provide a model of a quadrotor UAV.

If you have questions related to PX4, please submit them as issues to the [JSBSim bridge for PX4 project](https://github.com/Auterion/px4-jsbsim-bridge).
If you have questions related to JSBSim, you are welcome to open new issues in our project just as you did for issue #350. 

Thanks.

OpenUAS, 2021-03-18:
Maybe of help https://github.com/paparazzi/paparazzi/tree/master/conf/simulator/jsbsim/aircraft

pbecchi, 2021-12-23:
> he results are quite sensitive to the simulation rate. The results shown above were based on the 200Hz rate specified in jsb_F450_Launch.xml, when I increase the simulation rate to 1,000Hz these are the results with a propeller inertia of 4.46e-05.
So ~55ft versus 224ft AGL roughly 1s after power is applied.

Comparing to the new test with BLDC motor (#333), the F450 climb at 2.0 sec. 3.6 mt (12 ft). This is done with standard intergration time interval.
It seams to me a quite reasonable number  (50+ft is really to much for 1 second climb!).
This behaviour depend a lot on motor propeller acceleration therefore depend on inertia but also on torque available . Torque available now is limited to the phisical torque the motor is able to provide that depend on voltage and actual RPMs.

bcoconni, 2022-02-13:
The quadcopter `DJI F450` has been added to JSBSim by the PR #574.
This issue is now closed. Please open a new issue, if you encounter any issue with the FDM of the `DJI F450`.

closed 2022-02-13 by bcoconni