JSBSim-Team/jsbsim #254: Building JSBSim on Visual Studio 2019

DebugBSD, 2020-02-09:
Hi, good morning

To build JSBSim from source code on Visual Studio2019 I had to add:

 ..\;..\simgear\xml;..\input_output;%(AdditionalIncludeDirectories)

To Release configuration on VS2019 because I couldn't build the Solution.

seanmcleod, 2020-02-09:
I just downloaded master and opened the included Visual Studio solution and project files in VS2019 and didn't have any issues compiling the x64 release version.

The included project file has the following set `Additional Include Directories` and which seems sufficient to compile JSBSim.

```
$(ProjectDir)src;$(ProjectDir)src\simgear\xml;%(AdditionalIncludeDirectories)
```

Having said that I did hit an issue in linking due to some recent changes in JSBSim in this commit - https://github.com/JSBSim-Team/jsbsim/commit/41bc6592c1dae1c33dc398a3c7e7e91a4b54e2c0

```
1>FGfdmSocket.obj : error LNK2001: unresolved external symbol __imp_freeaddrinfo
1>FGfdmSocket.obj : error LNK2001: unresolved external symbol __imp_getaddrinfo
1>C:\source\Temp\jsbsimbuild\Release\JSBSim.exe : fatal error LNK1120: 2 unresolved externals
1>Done building project "JSBSim.vcxproj" -- FAILED.
```

seanmcleod, 2020-02-09:
I'll add a commit to update the Visual Studio project file to fix the linker error.

bcoconni, 2020-02-10:
@DebugBSD Alternatively JSBSim uses CMake as its build tool and, at least since version 2017, MSVC is able to build a project from the CMake files.

@seanmcleod 
> Having said that I did hit an issue in linking due to some recent changes in JSBSim in this commit - 41bc659

Correct, JSBSim now needs to be compiled against the libraries `wsock32.lib` and `ws2_32.lib` which should added to the MSVC project files.
> I'll add a commit to update the Visual Studio project file to fix the linker error.

Thanks, that is much appreciated.

I also figured out that [CMake can automatically generate the project file](https://stackoverflow.com/questions/395169/using-cmake-to-generate-visual-studio-c-project-files). Maybe we could include that task in our CI/CD process and upload the files in our release section along with the executables and Python wheels ?

seanmcleod, 2020-02-10:
@bcoconni yep I was thinking that ideally whenever any of the cmake input files are changed and committed that we auto-generate an updated VS project file and commit it to the repo so that anyone cloning the repo/downloading a zip file of the repo always has an up to date project file.

bcoconni, 2020-02-10:
@seanmcleod Well it is just a matter of adding a couple of lines of code to AppVeyor CI/CD command files to get the project files auto generated and published to the release section. I can do that part but I can't test the result myself since I do not have MSVC installed on my PC (and do not plan to). 

However pushing automatically a new commit whenever the CMake files are changed is one *big* step further since we would need some kind of bot for that. Well, I guess this can be done with [GitHub Actions](https://help.github.com/en/actions): I've already [set up JSBSim CI on Actions](https://github.com/JSBSim-Team/jsbsim/actions) (CD remains to be done) but the bot would need some further investigations and quite some testing in order to make sure that the `master` branch will not be polluted by undue commits. Hopefully someone already coded a similar Action, that would save the burden of figuring out ourselves.

lookingforfeng, 2020-04-03:
pull requset not been acpet yet

bcoconni, 2020-04-04:
The problem should now be fixed after PR #267 has been merged.

closed 2020-04-07 by seanmcleod