JSBSim-Team/jsbsim #255: Properties in JSBSim

fvarra, 2020-02-14:
Hello everyone,

I'm still a beginner in JSBSim and I have some doubts regarding the Properties that the program uses. I know that it is possible to print all the properties that an specific aircraft has, but is there any way of knowing how they are defined? For example, I would need to use the propeller Thrust Coefficient (Ct), does it corresponds to "propulsion/engine/thrust-coefficient"? How can I know that?

Also, I would be interested in creating specific Properties to be used in my aircraft. Is it possible? If yes, in what part of the aircraft file must be created? Or it must be done in the script file?

Thank you very much in advance,

bcoconni, 2020-03-16:
Hi Fernando,

First of all, sorry for the very long time before answering your request. I have been quite busy on other topics for the last weeks and had not much time to dedicate to JSBSim.
> I know that it is possible to print all the properties that an specific aircraft has, but is there any way of knowing how they are defined?

Indeed properties are not always very well documented in JSBSim. There are several ways you can get access to them: 
1. If you are using the executable `JSBSim` (or `JSBSim.exe` on Windows), you can pass the parameter `--catalog` to the command and you should get the list of all the properties that are defined in your model. Note that this will just produce a plain list of properties with no details about what they are. But at least it shows you which properties are available for your model.
2. You can refer to JSBSim general documentation has is described in the [user documentation](https://github.com/JSBSim-Team/jsbsim#user-documentation) or alternatively to the C++ docs (https://jsbsim-team.github.io/jsbsim/) although the latter is certainly not the friendliest way to get the information you are looking for.
3. You can review the [aircraft examples that are provided with JSBSim](https://github.com/JSBSim-Team/jsbsim/tree/master/aircraft) and [their scripts](https://github.com/JSBSim-Team/jsbsim/tree/master/scripts). The big advantage of these examples is that you can see property usage in a context i.e. how a given property is created, accessed, processed and updated.
4. Finally since we are an open source project, the definitive and most reliable source of information is the code. This is indubitably the hardest means to get access to a property definition and should be used as a last resort. This is definitely for hard core users and I would rather suggest for the average user to post a request on GitHub rather than trying to get their way through the code.
> For example, I would need to use the propeller Thrust Coefficient (Ct), does it corresponds to "propulsion/engine/thrust-coefficient"? How can I know that?

The property `propulsion/engine/thrust-coefficient` is indeed the property that returns Ct but it also accounts for the `<ct_factor>` if you defined one in your propeller model.
> I would be interested in creating specific Properties to be used in my aircraft. Is it possible? If yes, in what part of the aircraft file must be created? Or it must be done in the script file?

New properties can be defined almost anywhere in the model, although the general practice is to define a property in the block where it is used. Otherwise you might tumble on a situation where a property is referenced before it is defined.

Definition of properties is generally a simple list of `<property>` tags as below:
https://github.com/JSBSim-Team/jsbsim/blob/2f2398cbafc0fc4f14dc6157d1923548ec8804ba/aircraft/c172x/c172ap.xml#L38-L43

fvarra, 2020-03-23:
Hello,

No worries at all, and thank you very much for your detailed explanation. I will have a look at the examples.

> This is definitely for hard core users and I would rather suggest for the average user to post a request on GitHub rather than trying to get their way through the code.

I have a doubt about the property Aerodynamic Reference Point (AERORP). I found here [https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGForce.html](url) that it is defined as the point of application of the aerodynamic forces, so I suppose it is the Centre of Pressure. However, by definition, it is not fixed as it depends on the angle of attack, so how can I define this point?

Regards and thanks again,

seanmcleod, 2020-03-23:
The Aerodynamic Reference Point (AeroRP) doesn't have to be aligned with the centre of pressure.

For example think of a model placed in a wind tunnel, with say the balance point attached to a point in the cockpit. So the forces and moments measured will be around the cockpit, nowhere near the centre of pressure. 

If that is the source of your aerodynamic data then you would set the AeroRP to be the position of the cockpit and JSBSim will take care of calculating the forces and moments around the cg etc.

fvarra, 2020-03-26:
Ok, I understand.

Thank you so much!

closed 2020-03-26 by fvarra

agodemar, 2020-03-28:
The aerodynamic reference point is necessary because you might have you aerodynamic moment (e.g. pitching moment) referring to that specific pole. The pole is the AeroRP. When this pole is not coincident with the center of gravity (CoG), at each instant in the simulation the resultant aerodynamic force is transported from AeroRP to the CoG.

fvarra, 2020-04-02:
Great, thank you very much for the clear explanation.

kvnjonathan70, 2020-05-26:
So in this case aeroRP and CG should be same ?

seanmcleod, 2020-05-26:
@kvnjonathan70 in which case? The AeroRP is typically based on the source of the aerodynamic data, so the provider of the aerodynamic data will/should state what the AeroRP is. Plus remember the cg changes based on how the aircraft is loaded, and changes during flight as fuel is burnt etc. so it isn't in a constant location.