JSBSim-Team/jsbsim #256: Segmentation fault in AeromatiC++ 3.3.10

NikolaiVChr, 2020-02-18:
These input will fail it:

```
$ ./aeromatic-3.3.10.exe

** AeromatiC++ version 3.3.10
Aeromatiic is a JSBSim configuration file generation utility.
Please enter aircraft data when prompted.

You can always enter 'h' to get verbose help

** General Information **

Output directory [] (\Users\Nikolai): .
Create a subdirectory? [] (yes): yes
Overwrite? [] (yes): yes
Aircraft name [] (my_aircraft): trainer
Use dedicates System files? [] (yes): yes
Select a system of measurement [] (no):
 0: English (feet, pounds)
 1: Metric (meters, kilograms)
1
Type of aircraft [] (0):
 0: Light General Aviation (Glider, Small Commuter)
 1: High Performance (WWII Fighter, Military Trainer, Aerobatic, Air Racer)
 2: Fighter Jet
 3: Jet Transport (Passenger Jet Airliner, Transonic Jet Transport)
 4: Propeller Transport (Propeller Airliner, Propeller Transport)
1

** Weight and Balance **

Stall speed VS1 (clean, no flaps) [km/h] (0): 185
Maximum takeoff weight [kg] (4535.15): 6150
Empty weight [kg] (0): 3430
Inertia Ixx [kg/m2] (0):
Inertia Iyy [kg/m2] (0):
Inertia Izz [kg/m2] (0):

** Geometry **

Length [m] (12.192): 10.97
Select a wing shape [] (0):
 0: Straight
 1: Elliptical
 2: Delta
0
Wing span [m] (12.192): 10.86
Wing area [m2] (0): 19.3
Wing aspect ratio [] (0): 5.26
Wing taper ratio [] (1): 0.598
Wing root chord [m] (0): 2.24
Wing incidence [] (0): 0.3
Wing dihedral [] (0): 3.44
Wing sweep (quarter chord) [] (0): 8.483
Htail area [m2] (0): 4.12
Htail arm [m] (0): 4.6
Vtail area [m2] (0): 2.34
Vtail arm [m] (0): 3.97

** Systems **

Segmentation fault
```

NikolaiVChr, 2020-02-18:
Hmm, I just tried Release_Candidate_v1.0.0 aeromatic.

That one still was called 3.3.10 but it did not fail.

Thats good. Maybe its version number should be bumped, so people know its better.

bcoconni, 2020-03-16:
@NikolaiVChr 
Aeromatic code has not been updated for quite a while except for minor maintenance so the problem you are reporting is unexpected. Where was the first executable (the one which failed) coming from ?
@ermarch
May be we should bump the version number indeed ?

ermarch, 2020-03-17:
I tried to replicate the problem with the latest GIT version (even with the same input) but did not see the problem. I have no problem with updating the version number.

seanmcleod, 2020-04-08:
@NikolaiVChr can this issue be closed? Or are you still able to reproduce the crash?

closed 2020-04-08 by NikolaiVChr