JSBSim-Team/jsbsim #259: Engine model with RPM as input

fvarra, 2020-03-08:
Hello everyone,

I am trying to install in my aircraft some electric engines. In order to be able to simulate real flight test which are in function of the RPM of the engines, I would need this variable as an input... instead of the thrust position as usual.

I wouldn't mind using another type of engines to do this, but I haven't found that this can be done with other engines.

I also have a model of the control of the aircraft which relates the thrust level with the RPM. I have been trying to implement a PID (with RPM as the output) but it didn`t work, and also a system that relates the thrust level with the RPM, but I find an error saying that this property is already tied… I guess it's because the engine has already given a value to this property.

Do you have any idea of how I could proceed?

Thank you in advance,

Fernando

bcoconni, 2020-03-16:
Hi Fernando,

>In order to be able to simulate real flight test which are in function of the RPM of the engines, I would need this variable as an input... instead of the thrust position as usual.

The engine rotation speed is driven by the propeller in JSBSim, not by the engine. This is a limitation that results from the way the interaction between the engine and its propeller is coded. And indeed it is a read only property.

I guess your best option is to replace the engine by an external reaction. The thrust computed by your PID could be sent to the external reaction so that the resulting force would be accounted for by your model.

The `ball` example uses an external reaction to model the drag of a parachute. I think it can be easily modified to model the thrust from your aircraft electric engines.
https://github.com/JSBSim-Team/jsbsim/blob/2f2398cbafc0fc4f14dc6157d1923548ec8804ba/aircraft/ball/ball.xml#L100-L132

fvarra, 2020-03-23:
Hello,

Perfect, I understand, I will try to do that with external reactions.

Thank you very much,

closed 2020-03-23 by fvarra