JSBSim-Team/jsbsim #262: Generic engine and pre-computed mass properties

madz-mit, 2020-03-20:
Hi JSBSim-Team,
(1) I have some engine data that I need to include for my aircraft without using JSBSim's internal model equations. 
It is a 3D table with Altitude, Speed and Throttle used to directly obtain thrust (and later, may be sfc too as a 3D table.) 
How do I include it with the native code? Or do I have to write a separate class for a new EngineType? A few tips will help.
(2) Also, I have pre-computed mass properties so that I have CoG positions and MoIs for the aircraft as 1D look up table functions of mass. So I want to start off with the all-up weight and use the sfc to compute the instantaneous mass. This mass value would be then used to obtain the instantaneous CoG and MoI.
 
Madhan

bcoconni, 2020-03-31:
Regarding the topic (1), as I suggested in issue #259, your better option is to use an external reaction to which you could supply the thrust as a force that is computed from the 3D table you have.

Regarding topic (2), I guess you can use `<pointmass>` elements which are basically lumped mass which you can modify either in position or in mass.

See for example the model `weatherballon` below which defines a point mass for the Radiosonde.
https://github.com/JSBSim-Team/jsbsim/blob/66d233c7b17d7150f6d8403e3c15be342d7d9c94/aircraft/weather-balloon/weather-balloon.xml#L73-L92
 Its properties can be modified with `inertia/pointmass-weight-lbs[0]`, `inertia/pointmass-location-X-inches[0]`, `inertia/pointmass-location-Y-inches[0]`, `inertia/pointmass-location-Z-inches[0]`. The MoIs and CoG would be driven by the point mass location. Of course, depending on the complexity of your model, you might need to have several point masses to drive the CoG and the MoIs independently.

bcoconni, 2020-05-27:
Hi @madz-mit,

Have your questions been answered ? If you need any further support, let us know. Otherwise, could you please confirm that this issue can be closed ?

Thanks.

madz-mit, 2020-05-28:
OK, I will try out the methods you suggested. Thanks for the response.

closed 2020-05-28 by madz-mit