JSBSim-Team/jsbsim #263: Virtical tailarm distance "0"

kvnjonathan70, 2020-03-31:
Dear reader,

Vtailarm should be the distance between CG and AC of vertical tail.
why most AC configuration files represent is with zero value. ?

bcoconni, 2020-03-31:
I guess this is mainly because JSBSim only uses Vtailarm to export the properties `metrics/lv-norm` and `metrics/vbarv-norm`. If a model has no use of these 2 properties then Vtailarm can be left to 0 even though the value is non zero for the real aircraft.

seanmcleod, 2020-03-31:
I thought a number of years ago when I was trying to understand some of the elements in the aircraft configuration file that the only real reference I could find to it (Vtailarm) being used internally in the JSBSim source code was in one of the turbulence models. But doing a quick search now I can't find any such references.

bcoconni, 2020-03-31:
Well, the only references I could find to Vtailarm are:
https://github.com/JSBSim-Team/jsbsim/blob/66d233c7b17d7150f6d8403e3c15be342d7d9c94/src/models/FGAircraft.cpp#L59
https://github.com/JSBSim-Team/jsbsim/blob/66d233c7b17d7150f6d8403e3c15be342d7d9c94/src/models/FGAircraft.cpp#L137-L138
https://github.com/JSBSim-Team/jsbsim/blob/66d233c7b17d7150f6d8403e3c15be342d7d9c94/src/models/FGAircraft.cpp#L155-L162
https://github.com/JSBSim-Team/jsbsim/blob/66d233c7b17d7150f6d8403e3c15be342d7d9c94/src/models/FGAircraft.cpp#L184
and the method `FGAircraft::GetVTailArm()` is not use anywhere in the code.

seanmcleod, 2020-03-31:
Just double-checked my old JSBSim code and I wasn't imagining things :wink:

From `void FGAtmosphere::Turbulence(void)`

````
    if (Aircraft->GetVTailArm() > 0)
      vTurbPQR(eR) = vBodyTurbGrad(eX)/Aircraft->GetVTailArm();
    else
      vTurbPQR(eR) = vBodyTurbGrad(eX)/10.0;
````

But that code has obviously been changed since then.

bcoconni, 2020-03-31:
It must be very old then :smile: , since the Turbulence code is in the class `FGWinds` since Jun 21, 2011 and commit 4518f4e9b7dd7ed75945171f6ea2b11a56286a92 and even then, the code was not referring to `GetVTailArm()`.

seanmcleod, 2020-03-31:
Looks like Aug 11, 2010.

bcoconni, 2020-03-31:
OK found it ! It has been removed May 18, 2011 with commit 19d09d6de3b6dbc4a96a09ce158e96e46a4ab643.

The lines of code you mentioned were part of @jonsberndt 's turbulence model which he tagged himself "very experimental and incomplete at the moment". Given the context of the commit, the Milspec turbulence model had just been added (thanks to a contribution by Andreas Gaeb) and became the default turbulence model (commit b3e652bb1d734adb5791d42b228deaaa7e022f2a).  Jon removed his own model (and the call to `FGAircraft::VTailArea()`) following that modification.

seanmcleod, 2020-03-31:
Pretty good memory, although not potentially the most useful thing to remember from 9-10 years ago :wink:

bcoconni, 2020-03-31:
I wish my memory was that good but I have just reconstructed the sequence from the commits history. If someone would need to know why it is important to write good commit messages, that was a pretty good illustration :wink:

kvnjonathan70, 2020-04-01:
thanks for the quick response, really appreciate.   
i saw this equation below in jsbsim website before posting my question n i got confused.
but now i am cleared that is not being used,  also its a good practice to put real value in case of turbulence model being introduced.
 
 vbarv = VTailArm*VTailArea / (WingSpan*WingArea);

bcoconni, 2020-04-01:
@kvnjonathan70 It looks like your question has been answered. Do you need further support on this topic or shall we close the issue ?

kvnjonathan70, 2020-04-01:
Done.. Thanks 😇

closed 2020-04-01 by kvnjonathan70