JSBSim-Team/jsbsim #272: introduce turbulence Dryden spectrum

kvnjonathan70, 2020-04-08:
Dear JSBSim-Team! 
i am very new to jsbsim, my questions today and upcoming might not be too interesting. But your response would help me to gain more knowledge is this software and subject and provide further development in this program. 

i am learning to introduce  wind turbulence to any model at some altitude at some time with some intensity. 

i read the script p51d tail wind. xml, FGwind, FGatmosphere and #142. 
i am confused as how to give a turbulence. 
is the property atmosphere /turb- type needs to b assigned in the event section of the script, or aircraft configuration file (like interface properties) or in the initialization file. 

if in event section then is the process similar to cosine gust? 
i havent seen turbulence properties used in a/c configuration file so far.. and in initialize file can i give wind velocities in rad/sec for continuous turbulence? 
i am from mechanical background and does not have programming knowledge like c++, so i am looking to work around using external xml code.

bcoconni, 2020-04-09:
There is a bit of [documentation for turbulence in the C++ docs](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGWinds.html). There is a matrix in this documentation and the parameters can be adjusted with the properties `atmosphere/turbulence/milspec/windspeed_at_20ft_AGL-fps` and `atmosphere/turbulence/milspec/severity`.

Be aware that turbulence is supposed to be added to a global wind so might want need to set up a wind direction and intensity first and then, only after, set up the turbulence.

kvnjonathan70, 2020-04-10:
Thanks for coming up.. i was expecting you 😊... just to understand it for correct application, do you remember any example file to refer?  i checked this latest rolling release but could not find any application to dryden than cosine gust in the script as event.

bcoconni, 2020-04-10:
> do you remember any example file to refer?

Unfortunately I don't. I would say you could try to set up the Milspec turbulence model with a script event similar to the one set in `ZLT-NT-moored-1.xml`
https://github.com/JSBSim-Team/jsbsim/blob/c4ea8194f110c1903364e7c55c30393b73c85456/scripts/ZLT-NT-moored-1.xml#L75-L89
Which could be something like:
```xml
  <event name="Strong wind from north">
   <description>Strong wind from north with turbulence</description>
   <condition>
    simulation/sim-time-sec GT 30.0
   </condition>
   <!-- Ramp up the wind. -->
   <set name="atmosphere/wind-east-fps" value="30.0" action="FG_RAMP" tc ="30.0"/>
   <set name="atmosphere/turb-type" value="4"/>
   <set name="atmosphere/turbulence/milspec/windspeed_at_20ft_AGL-fps" value="25.0" action="FG_RAMP" tc ="30.0"/>
   <set name="atmosphere/turbulence/milspec/severity" value="3.0" action="FG_RAMP" tc ="30.0"/>
  </event>
```
That's untested code but this could serve as a start.

Let us know your progress, once you get a basic example that works we could add a new script to JSBSim to serve as an example for others :wink:

kvnjonathan70, 2020-04-10:
Thanks for detailed structure of code.. i really appreciate .. cool.. i got it!  indeed vl test it out and let our people know about it to save time. i have taken a task to simulate canard with differential horizontal tail.  vl keep posting about it with all the doubt in different threads
 
now m closing the thread, i vl reopen when test is complete

closed 2020-04-10 by kvnjonathan70