JSBSim-Team/jsbsim #273: block representation application for FCS

kvnjonathan70, 2020-04-08:
Dear team, i found it a bit difficult in visualizing the fcs with many component . i also used jsbsim commander but it doest support on complicated a/c files like f16. i read thomas vogeltanz paper on fcs for jsbsim. they made a beautiful scilab application for visual representation. Is there anything to view fcs in much simplified way?

seanmcleod, 2020-04-08:
@kvnjonathan70 do you have a link for Thomas's paper that doesn't require a subscription?

kvnjonathan70, 2020-04-08:
its requires subscription. but i have the paper, i can share it with u, if u dont mind

seanmcleod, 2020-04-08:
I've emailed Thomas asking if he could email me a copy of his paper.

kvnjonathan70, 2020-04-08:
lets c what can be done. i hope if he shares his application gui then a lot of error due to visualizing fcs would be reduced.

seanmcleod, 2020-04-08:
I see he has a github repo for the paper but it's empty.

kvnjonathan70, 2020-04-08:
i requested him 5 days ago, but he doesn't seem to b active

seanmcleod, 2020-04-10:
@kvnjonathan70 I pinged you via your email address listed on LinkedIn.

kvnjonathan70, 2020-04-10:
No sir i havnt received any msg out there, please contact me at kvnjonathan70@gmail.com

seanmcleod, 2020-05-01:
@kvnjonathan70 take a look at the mention of reviving JSBSimCommander, although it is a long-term goal - https://github.com/JSBSim-Team/jsbsim/issues/279#issuecomment-619623384

kvnjonathan70, 2020-05-05:
Hi, its been a long time,  it seems that you have been working on phython built jsbsim, i also started to learn phython as i noticed that jsbsim is advancing in it. I think it could be possible to build GUI in it. I am planning to starting working on building GUI  after i finish my course, if you have previously worked on it then you might be having some code. Would you mind sharing ?

seanmcleod, 2020-05-05:
I haven't done any significant Python GUI work, just some real basic PyQT stuff. Most of my GUI experience is with WPF.

kvnjonathan70, 2020-05-06:
Btw while i was refering to FCS of simplified f16. I found that output of PID and kinematic is same for yaw control.  I found is bit confusing because the output of PID is summed along with rudder cmd and yaw trim cmd. 

<pid name="fcs/yaw-load-pid">
     <trigger>fcs/rudder-pid-trigger</trigger>
     <input>fcs/yaw-trim-error</input>
     <kp> 0.105500 </kp>
     <ki> 0.000010 </ki>
     <kd> 0.00005 </kd>
     <clipto>
    <min>-1</min>
    <max>1</max>
     </clipto>
     <output>fcs/rudder-pos-norm</output>
   </pid>

   <summer name="fcs/yaw-scheduler">
     <input>fcs/rudder-cmd-norm</input>
     <input>fcs/yaw-trim-cmd-norm</input>
     <input>fcs/yaw-load-pid</input>
     <clipto>
    <min>-1</min>
    <max>1</max>
     </clipto>
   </summer>

   <kinematic name="fcs/rudder-position">
    <input>fcs/yaw-scheduler</input>
    <traverse>
     <setting>
      <position>-1</position>
      <time>0.4</time>
     </setting>
     <setting>
      <position>1</position>
      <time>0.4</time>
     </setting>
    </traverse>
    <output>fcs/rudder-pos-norm</output>
   </kinematic>

kvnjonathan70, 2020-05-06:
If commander revives it would be a huge upgrade i guess, I will start with phython and see how far i could go for its gui development

bcoconni, 2020-05-06:
@kvnjonathan70 regarding your question about f16's PID and kinematic, is it related to the topic of the current issue ? Because if it is not, I would suggest that you open a new issue to discuss that topic. The goal being that an issue stay focused on one single discussion for triaging and progress monitoring (issues that cover multiple topics tend to never being closed which result in the frustration of all participants - those asking the questions as well as those trying to answer them).

Thanks.

kvnjonathan70, 2020-05-06:
sure 😌

seanmcleod, 2020-05-06:
@kvnjonathan70 you could definitely start from scratch in terms of a Python based subset of JSBSimCommander, or you could help out with getting JSBSimCommander updated.

kvnjonathan70, 2020-05-06:
@seanmcleod  i have planned to do it so, i will reopen the issue after some progress from my end

closed 2020-05-06 by kvnjonathan70