JSBSim-Team/jsbsim #274: JSBSim on Raspbian - GCC Version

Marthi0, 2020-04-09:
Hi JSBSim-Team,

I am trying to compile JSBSim on a Raspberry Pi under Raspbian Buster as a shared library. **Has anyone ever tried something like that ?**
I have many warnings during the compilation with CMake and some errors during linking with my application.
I have no problem to build and link on my Debian so I start looking difference between the 2 distros and I saw that I don't have the same compiler version : 6.3.0 on Debian (working) and 8.3.0 on Raspbian (not working). I do not see any recommandation on the compiler version to use. **Is there one ?**
Find below the first warning I get during JSBSim build with _make_ command :
```
In file included from /usr/include/c++/8/vector:69,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGScript.h:40,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGScript.cpp:46:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const double&}; _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<double>::iterator’ {aka ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
```
It seems that something changed in GCC 7.1. I will try to downgrade my GCC version to see if it change something.

Thank you for your help,

Marthi

bcoconni, 2020-04-09:
> I am trying to compile JSBSim on a Raspberry Pi under Raspbian Buster as a shared library. **Has anyone ever tried something like that ?**

Maybe the guys from [ArduPilot](https://ardupilot.org/) could be of some help since they have experience of embedded systems and are [sometimes using JSBSim](https://ardupilot.org/dev/docs/sitl-with-jsbsim.html). Not sure that anyone of them ever tried to compile JSBSim on a Raspberry Pi. Might be worth a try.
> I saw that I don't have the same compiler version : 6.3.0 on Debian (working) and 8.3.0 on Raspbian (not working). I do not see any recommandation on the compiler version to use. **Is there one ?**

There is no recommendation regarding compilers except that they must be fully C++11 compliant. Honestly we do not make special efforts to support a wide variety of compilers beyond using strict C++11 code and having no dependencies on external libraries. Occasionally someone report that JSBSim is not compiling on their favorite platform/compiler (just as you do) and we have generally managed to fix the issue. However since there are generally no follow up of these modifications, the compilation might break on such platforms from time to time.
> Find below the first warning I get during JSBSim build with `make` command :

That's quite weird since the compilation breaks when the standard library `<vector>` is included and it is the first statement in the file `FGScript.h` (line 40)
https://github.com/JSBSim-Team/jsbsim/blob/c4ea8194f110c1903364e7c55c30393b73c85456/src/input_output/FGScript.h#L33-L46
Have you checked that your C++ standard library `libstdc++` version is of the same than your compiler ? May be you can try to compile a very simple C++ file `vector.cpp` like below:
```c++
#include <vector>
#include <iostream>
int main(void) {
  std::vector<int> a ={-1};
  std::cout << a[0] << std::endl;
  return 0;
}
```
And compile it with the command
```bash
> g++ vector.cpp -o vector
```
If the compiler fails to generate an executable `vector` or if the executable does not return `-1` then there is definitely a problem with your C++ installation.

Marthi0, 2020-04-10:
Thank you for your quick answer.

I tried your example and it works perfectly so my installation might not be the problem.

I tried with GCC 6.5.0 and I got only one warning during JSBSim build but same errors during linking.

I succeed to compile JSBSim under Raspbian few years ago but this time I wanted to install it in a proper way as an external library.

Unless you have other ideas, I think you can close the issue.
Thank you for your help.

Marthi

Build warning for information :
`
In file included from /usr/include/c++/6/map:60:0,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGXMLElement.h:38,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGXMLElement.cpp:31:
/usr/include/c++/6/bits/stl_tree.h: In member function ‘std::pair<std::_Rb_tree_node_base*, std::_Rb_tree_node_base*> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_get_insert_hint_unique_pos(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, const key_type&) [with _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, double>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, double> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, double> >]’:
/usr/include/c++/6/bits/stl_tree.h:1928:5: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, double>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, double> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, double> > >::const_iterator {aka std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, double> >}’ will change in GCC 7.1
     _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
`

bcoconni, 2020-04-10:
> I tried your example and it works perfectly so my installation might not be the problem.

And what about trying the compilation while forcing C++ to the standard C++11 ? 
```bash
> g++ -std=c++11 vector.cpp -o vector
```
`g++` has full support of C++11 since version 4.8 so version 6.3 and 8.3 should not make any difference. I'm compiling JSBSim locally with version 9.3.1 and no issues are reported regarding a change in GCC that took place in version 7.1

> Build warning for information :
```
In file included from /usr/include/c++/6/map:60:0,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGXMLElement.h:38,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGXMLElement.cpp:31:
/usr/include/c++/6/bits/stl_tree.h: In member function ‘std::pair<std::_Rb_tree_node_base*, std::_Rb_tree_node_base*> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_get_insert_hint_unique_pos(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, const key_type&) [with _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, double>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, double> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, double> >]’:
/usr/include/c++/6/bits/stl_tree.h:1928:5: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, double>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, double> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, double> > >::const_iterator {aka std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, double> >}’ will change in GCC 7.1
     _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
```
I'm a bit puzzled because the compilation fails as soon as a standard library is included, at this stage the compiler has not yet parsed a single line of code specific to JSBSim.

BTW did you provide the complete error message so far ? We had such an issue with issue #225 where the original poster just copied an extract of the compiler output and it was misleading because vital information was missing from the extract. [It was only after the poster provided the complete compiler output that we could make progress in resolving the issue](https://github.com/JSBSim-Team/jsbsim/issues/225#issuecomment-560011852)

Marthi0, 2020-04-14:
Sorry for the late answer.

> And what about trying the compilation while forcing C++ to the standard C++11 ?
> 
>  `g++ -std=c++11 vector.cpp -o vector`
>
> g++ has full support of C++11 since version 4.8 so version 6.3 and 8.3 should not make any difference. I'm compiling JSBSim locally with version 9.3.1 and no issues are reported regarding a change in GCC that took place in version 7.1

There is still no problem with your example, even if I force C++11.

Please, find below the complete build log of JSBSim compilation (sorry for the long message but there are many errors)..

Cmake log :
```
pi@raspberrypi:~/VIABLE/jsbsim_github/build $ cmake -DBUILD_SHARED_LIBS=ON -DBUILD_PYTHON_MODULE=OFF -DCMAKE_CXX_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_C_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Could NOT find CxxTest (missing: CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE)
-- Found LSB_Release: /usr/bin/lsb_release
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/VIABLE/jsbsim_github/build
```

Make log :
```
pi@raspberrypi:~/VIABLE/jsbsim_github/build $ make
Scanning dependencies of target IOStreams
[  0%] Building CXX object src/simgear/io/iostreams/CMakeFiles/IOStreams.dir/sgstream.cxx.o
[  0%] Built target IOStreams
Scanning dependencies of target Init
[  1%] Building CXX object src/initialization/CMakeFiles/Init.dir/FGInitialCondition.cpp.o
[  2%] Building CXX object src/initialization/CMakeFiles/Init.dir/FGTrim.cpp.o
In file included from /usr/include/c++/8/vector:69,
                 from /usr/include/c++/8/bits/random.h:34,
                 from /usr/include/c++/8/random:49,
                 from /home/pi/VIABLE/jsbsim_github/src/FGFDMExec.h:45,
                 from /home/pi/VIABLE/jsbsim_github/src/initialization/FGTrim.h:53,
                 from /home/pi/VIABLE/jsbsim_github/src/initialization/FGTrim.cpp:45:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {JSBSim::FGTrimAxis}; _Tp = JSBSim::FGTrimAxis; _Alloc = std::allocator<JSBSim::FGTrimAxis>]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<JSBSim::FGTrimAxis>::iterator’ {aka ‘__gnu_cxx::__normal_iterator<JSBSim::FGTrimAxis*, std::vector<JSBSim::FGTrimAxis> >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {JSBSim::FGTrimAxis}; _Tp = JSBSim::FGTrimAxis; _Alloc = std::allocator<JSBSim::FGTrimAxis>]’:
/usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<JSBSim::FGTrimAxis*, std::vector<JSBSim::FGTrimAxis> >’ changed in GCC 7.1
    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
    ^~~~~~~~~~~~~~~~~
[  3%] Building CXX object src/initialization/CMakeFiles/Init.dir/FGTrimAxis.cpp.o
[  3%] Built target Init
Scanning dependencies of target InputOutput
[  4%] Building CXX object src/input_output/CMakeFiles/InputOutput.dir/FGGroundCallback.cpp.o
[  5%] Building CXX object src/input_output/CMakeFiles/InputOutput.dir/FGPropertyManager.cpp.o
[  5%] Building CXX object src/input_output/CMakeFiles/InputOutput.dir/FGScript.cpp.o
In file included from /usr/include/c++/8/vector:69,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGScript.h:40,
                 from /home/pi/VIABLE/jsbsim_github/src/input_output/FGScript.cpp:46:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const double&}; _Tp = double; _Alloc = std::allocator<double>]’:
[... 349 lines cut ...]
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign(char const*)@GLIBCXX_3.4.21'
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8'
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `vtable for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4.21'
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `vtable for std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4.21'
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::insert(unsigned int, char const*)@GLIBCXX_3.4.21'
[... cut ...]
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()@GLIBCXX_3.4.21'
~/raspi/sysroot/usr/local/lib/libJSBSim.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::replace(unsigned int, unsigned int, char const*, unsigned int)@GLIBCXX_3.4.21'
collect2: error: ld returned 1 exit status
```

Maybe there is a link between JSBSim build warnings and linking problems during cross compilation ...
Sorry again for heavy error logs ...

seanmcleod, 2020-04-14:
Take a look at https://stackoverflow.com/questions/48149323/what-does-the-gcc-warning-project-parameter-passing-for-x-changed-in-gcc-7-1-m

Marthi0, 2020-04-14:
> Take a look at https://stackoverflow.com/questions/48149323/what-does-the-gcc-warning-project-parameter-passing-for-x-changed-in-gcc-7-1-m

Thank you for this link. This can explain warnings during JSBSim build and it seems fine to ignore them. So my problem is more on cross compilation linking. I will try other libraries and compilers. If I find a solution, I will post it here.

Thank you for your help.

bcoconni, 2020-04-26:
@Marthi0 Should we consider this issue closed ?

Marthi0, 2020-05-07:
Yes, you can close it. This is definitely a cross compilation issue and not related to JSBSim.
By compiling and installing JSBSim natively on my RPI as a shared library and then cross compiling my executable, I had some compiler version issues.
Finally, I also cross compiled JSBSim and linked it statically and it is working.
Thank you for your help.

closed 2020-05-07 by Marthi0