JSBSim-Team/jsbsim #276: FlightGear segfault - jsbsim.cxx:1401

legoboyvdlp, 2020-04-16:
Every so often when changing LOD ranges in a JSBSim craft, I get a segfault. Analysis of the crash dump revealed (well, a rather useless error message)

The instruction at 0x%p referenced memory at 0x%p. The memory could not be %s
INVALID_POINTER_READ_c0000005_fgfs.exe!FGJSBsim::get_agl_ft

GroundReactions->SetStaticFFactor((*material).get_friction_factor());

I do not understand how this could happen, since you have a nil check on material. It definitely exists, or that code is not run.

Call stack:
[0xb]   fgfs!FGJSBsim::get_agl_ft + 0x239   
[0xc]   fgfs!FGFSGroundCallback::GetAGLevel + 0x79   
[0xd]   fgfs!JSBSim::FGGroundCallback::GetAGLevel + 0x2f   
[0xe]   fgfs!JSBSim::FGLocation::GetContactPoint + 0x42   
[0xf]   fgfs!JSBSim::FGLocation::GetAltitudeAGL + 0x95   
[0x10]   fgfs!FGJSBsim::copy_from_JSBsim + 0x2d0   

Just to clarify, that when you change LOD ranges, then a reload scenery operation occurs.

bcoconni, 2020-04-17:
> I do not understand how this could happen, since you have a nil check on material. It definitely exists, or that code is not run.

The fact that a pointer is not null does not mean that it is valid. And obviously in this case, the pointer `material` is invalid and the OS kills FlightGear as soon as it tries to invoke the method `get_friction_factor` on it.

The `material`pointer is obtained from the line of code below:
```c++
FGInterface::get_agl_ft(t, pt, alt_off, contact, normal, vel,
                          angularVel, material, id);
```
which, it should be noted, has been purposely modified to ignore the return value from the method `FGFunction::get_agl_ft`.

**Commit:** https://sourceforge.net/p/flightgear/flightgear/ci/86b3469fbff161a99db39131a810c06ba0ba2767/

**Mail discussion** (note that the link in the code is broken):
https://sourceforge.net/p/flightgear/mailman/message/32246380/
https://sourceforge.net/p/flightgear/mailman/message/32247050/

From what I could gather from the discussions above, the check on the return value of `FGInterface::get_agl_ft` has been discarded because the method always returns a sensible value for the altitude above ground level. It is not said however that the other parameters (including `material`) are populated with meaningful values. So it might be interesting to discard `material` when `FGInterface::get_agl_ft` returns `false`. Something like
```c++
if (!FGInterface::get_agl_ft(t, pt, alt_off, contact, normal, vel,
                          angularVel, material, id))
  material = nullptr; // Discard the material data when FGInterface reports an problem.
```

bcoconni, 2020-04-17:
And by the way, the pointer `material` is not set to null when declared (line 1381) which is **very, very evil™️** .
```c++
const simgear::BVHMaterial* material;
```

bcoconni, 2020-04-26:
@legoboyvdlp Is the crash still occurring every so often ? Or did the code modification improve the situation ?

legoboyvdlp, 2020-04-27:
Hi,
You wanted me to modify my local code to set material to null, just to confirm? I can do that and check sometime later this week. Or has this been pushed to -next? :)

bcoconni, 2020-04-27:
> You wanted me to modify my local code to set material to null, just to confirm?

Yes, that was basically the idea. Sorry for not having been clear about that.
> I can do that and check sometime later this week.

That would be much appreciated. Thanks.
> Or has this been pushed to -next? :)

No it hasn't. My local build of FlightGear is currently broken so I was waiting for your confirmation that the change actually fixes the bug before committing it to `next`.

legoboyvdlp, 2020-04-27:
Yes, I believe this fix works. I reset LOD about fifteen times and saw no crash. I will verify further with a longer flight later on. Thank you!

bcoconni, 2020-04-27:
Great ! Let me know when you think it's robust enough and I will push the change to FlightGear.

legoboyvdlp, 2020-04-27:
Hi Bertrand,
I'm afraid there may be deeper issues.
I have a crash with callstack:

```[0x0]   osg131_osg!osg::Group::traverse + 0x36   
[0x1]   fgfs!FGGroundCache::CacheFill::apply + 0x84   
[0x2]   osg131_osg!osg::Group::accept + 0x71   
[0x3]   osg131_osg!osg::Switch::traverse + 0x79   
[0x4]   fgfs!FGGroundCache::CacheFill::apply + 0x84   
[0x5]   osg131_osg!osg::Switch::accept + 0x6e   
[0x6]   fgfs!FGGroundCache::prepare_ground_cache + 0x84e   
[0x7]   fgfs!FGInterface::prepare_ground_cache_ft + 0xbe   
[0x8]   fgfs!FGJSBsim::update_ground_cache + 0x14a   
```

I can get this on every flight. My procedure is to setup a flight, and depart from KDCA runway 01. I turn left heading 290 after departure. 

Looking in the WinDbg application, for what its worth, I see that _material has a valid value. I am not experienced in interpreting output, but looking at this, I think there is an iterator that is referencing a pointer that is null or invalid:
[image]

I see groundcache.cxx line 108 - we refer to group multiple times, does it ever check it is valid?

legoboyvdlp, 2020-04-27:
Just to be absolutely clear, I believe this is unrelated to the debug > reload scenery operation - i.e. the first fix has worked, this has coincidentally appeared.

legoboyvdlp, 2020-04-27:
Ok - I think that applying the fix has caused it. After reverting, there appear not to be any problems.

So:

1. The fix works to solve the original problem, but causes other problems. Well, it might have been because I added a duplicated call to get_agl_ft as such:

`FGInterface::get_agl_ft(t, pt, alt_off, contact, normal, vel,
                          angularVel, material, id))

if (!FGInterface::get_agl_ft(t, pt, alt_off, contact, normal, vel,
                          angularVel, material, id))
  material = nullptr; // Discard the material data when FGInterface reports an problem.`

That is, I had two calls one after the other, the second one would check the return value. So quite possibly that would cause it :)  

In any case, your fix works -- only, possibly either due to my bad implementation of the fix or something else, it causes other problems.

2. It would be good to guard the groundcache::cache_fill code in any case against invalid pointers so it does not cause segfaults!

bcoconni, 2020-04-29:
> I see groundcache.cxx line 108 - we refer to group multiple times, does it ever check it is valid?

This code and the OSG stuff are in the realm of "pure" FlightGear I guess you'd better ask them.
> The fix works to solve the original problem, but causes other problems

The bug due to an invalid `material` pointer is possibly hiding some other bugs in the ground cache code.
> That is, I had two calls one after the other, the second one would check the return value. So quite possibly that would cause it :)

That's quite likely, `FGInterface::get_agl_ft` is supposed to be called once. I don't know what happens when calling it twice in a row especially if it returned an error the first time.
> In any case, your fix works -- only, possibly either due to my bad implementation of the fix or something else, it causes other problems.

I would prefer that you test the fix with only one call to `FGInterface::get_agl_ft`, if you don't mind. I would like to make sure that the fix does not worsen the situation before proceeding with committing the fix to FlightGear.

legoboyvdlp, 2020-04-29:
Ok, will do :+1:

legoboyvdlp, 2020-04-29:
Hi,

Yes, this works just fine!

```diff --git a/src/FDM/JSBSim/JSBSim.cxx b/src/FDM/JSBSim/JSBSim.cxx
index 8e1558955..559a8511c 100644
--- a/src/FDM/JSBSim/JSBSim.cxx
+++ b/src/FDM/JSBSim/JSBSim.cxx
@@ -1378,15 +1378,17 @@ FGJSBsim::get_agl_ft(double t, const FGColumnVector3& loc, double alt_off,
                      double contact[3], double normal[3], double vel[3],
                      double angularVel[3])
 {
-  const simgear::BVHMaterial* material;
+  const simgear::BVHMaterial* material = nullptr;
   simgear::BVHNode::Id id;
   double pt[3] {loc(1), loc(2), loc(3)};

   // don't check the return value and continue above scenery discontinuity
   // see http://osdir.com/ml/flightgear-sim/2014-04/msg00145.html
-  FGInterface::get_agl_ft(t, pt, alt_off, contact, normal, vel,
-                          angularVel, material, id);
-
+  if (!FGInterface::get_agl_ft(t, pt, alt_off, contact, normal, vel,
+                          angularVel, material, id)) {
+      material = nullptr; // Discard the material data when FGInterface reports an problem.
+  }
+
```

I corrected not only the call to get_agl_ft but also the initialization of the pointer to nullptr (as you said above). It is working well with no evidence of any of the earlier problems.

bcoconni, 2020-04-30:
The fix has just been pushed to FlightGear `next` branch.
I have suggested to push it to the branches `2018.3` and `2020.1` as well.

If you agree, I suggest that we now close this issue.

closed 2020-04-30 by legoboyvdlp

bcoconni, 2020-04-30:
Thanks for the bug report and the testing :+1: