JSBSim-Team/jsbsim #277: C.g. and empty weight of empty aircraft inconsistent with defined c.g. and empty weight

legoboyvdlp, 2020-04-17:
Hello,
I defined my C.G. as the empty aircraft center of gravity, coordinates measured relative to the Airbus datum. The aircraft already had C.G. at 25% MAC, so I moved it to 28.5% C.G.

`<location name="CG" unit="M">
		<x> 18.9966 </x> 
		<y> 0 </y>
		<z> 0.19249 </z>
	</location>`

At present, when I set all pointmasses to zero, and load the aircraft with zero fuel, my centre of gravity is at 18.659602 as seen in FlightGear's fuel and payload dialog. 

My empty weight also does not match the <emptywt> specified in the jsbsim configuration, which was  95627.7 pounds.

[image]

legoboyvdlp, 2020-04-17:
Difference in empty weight is 3722 lb.

bcoconni, 2020-04-17:
Could you please provide the properties values for weight & C.G. ? There is nothing that can be made of a screen capture from a FlightGear dialog here since we do not know how FlightGear processes JSBSim data to display them.

And please, could you provide a copy of your `<mass_balance>` settings directly in this issue report rather than asking us to find our way through a repository ?

That would be much appreciated. Thanks.

legoboyvdlp, 2020-04-17:
`<mass_balance>
	<ixx unit="SLUG*FT2">  611903 </ixx>
	<iyy unit="SLUG*FT2"> 1735389 </iyy>
	<izz unit="SLUG*FT2"> 1401293 </izz>

	<ixy unit="SLUG*FT2">       0 </ixy>
	<ixz unit="SLUG*FT2">       0 </ixz>
	<iyz unit="SLUG*FT2">       0 </iyz>
	<emptywt unit="LBS" >   95627.7 </emptywt> 
	
	<location name="CG" unit="M">
		<x> 18.9966 </x> 
		<y> 0 </y>
		<z> 0.19249 </z>
	</location>
	
	<pointmass name="Forward Cabin">
		<weight unit="LBS">3722</weight>
		<location name="POINTMASS" unit="M">
			<x> 10 </x>
			<y> 0 </y>
			<z> 0.5 </z>
		</location>
	</pointmass>
	
	<pointmass name="Center Cabin">
		<weight unit="LBS">11361</weight>
		<location name="POINTMASS" unit="M">
			<x> 16.5 </x>
			<y> 0 </y>
			<z> 0.5 </z>
		</location>
	</pointmass>
	
	<pointmass name="Aft Cabin">
		<weight unit="LBS">8422</weight>
		<location name="POINTMASS" unit="M">
			<x> 26 </x>
			<y> 0 </y>
			<z> 0.5 </z>
		</location>
	</pointmass>
	
	<pointmass name="Fwd Cargo">
		<weight unit="LBS">3000</weight>
		<location name="POINTMASS" unit="M"> 
			<x> 14.0 </x>
			<y> 0 </y>
			<z> -1.1 </z>
		</location>
	</pointmass>
	
	<pointmass name="Aft Cargo">
		<weight unit="LBS">3000</weight>
		<location name="POINTMASS" unit="M">
			<x> 24.0 </x>
			<y> 0 </y>
			<z> -1.1 </z>
		</location>
	</pointmass>
	
	<pointmass name="Aft Bulk Cargo">
		<weight unit="LBS">0</weight>
		<location name="POINTMASS" unit="M">
			<x> 29.0 </x>
			<y> 0 </y>
			<z> -1.1 </z>
		</location>
	</pointmass>
</mass_balance>`

legoboyvdlp, 2020-04-17:
As a confirmation that all pointmasses are zero:
[image]

and fuel is zero:
[image]

The center of gravity and empty weight;
[image]

legoboyvdlp, 2020-04-17:
Sorry for the noise - the indexes of aliased pointmass were incorrect in the fuel payload dialog. It is working correctly now.

closed 2020-04-17 by legoboyvdlp

bcoconni, 2020-04-17:
> Sorry for the noise - the indexes of aliased pointmass were incorrect in the fuel payload dialog. It is working correctly now.

No problem 😄 The most important is that your problem has been fixed.