JSBSim-Team/jsbsim #278: Rates not integrating correctly?

natewillis, 2020-04-17:
I've been attempting to build my own simple rocket/missile launch based on the J246 example that has been included and I can't seem to make sense of the data that I'm getting.  I'm modeling a single engine missile and attempting to keep the q/r rates at 0 via a nozzle gimbal.  During tuning, I turned my P gains way up and my M and N moments are huge (as expected) and that seems to propagate to huge q-dot and r-dot values, however the q and r seem to have no correlation to the q-dot and r-dot.  Its probably something stupid that I've done, but I've been pulling my hair out for hours trying to understand how huge q-dots and r-dots don't seem to have any influence on q and r (and by association my pitch and yaw).  Are there forces I'm missing or am I misunderstanding how these forces should integrate?  Thanks for any insight you can provide!

[image]

[aircraft_files.zip](https://github.com/JSBSim-Team/jsbsim/files/4495242/aircraft_files.zip)

bcoconni, 2020-04-19:
> however the q and r seem to have no correlation to the q-dot and r-dot

I struggled to find the explanation but I think I found it and it is quite simple. You are outputting the data at a rate of 10Hz i.e. every 12 time steps.
```xml
	<output name="nate_missile.csv" rate="10" type="CSV">
``` 
You are therefore filtering out the high frequency oscillations of **Q** and **R**. Now if you display **Q** and **R** at each time step, you'll see that they are responding to the high values of **Qdot** and **Rdot** as expected.

[image]

In order to catch high frequency oscillations you need to output the data at a rate of 120Hz
```xml
	<output name="nate_missile.csv" rate="120" type="CSV">
```

natewillis, 2020-04-20:
That's definitely it.  Thanks!  I'd been banging my head against the keyboard trying to figure that out for a while now.

closed 2020-04-20 by natewillis