JSBSim-Team/jsbsim #281: Making a new jsbsim model

VinhA2018, 2020-04-21:
Hello everyone, 
I'm currently working in a project that requires a lot of simulation of my Arduplane navigation code. This will be implemented on an "EasyStar II" glider. However, I would like to have a real jsbsim model of the glider in order to simulate the flights.
I've found a model of "EasyStar I" (older model without ailerons), but I need to add the ailerons. 
I've added `<aerosurface_scale name="Left Aileron Control">` and the others aerosurfaces needed, but still the model isn't stable and when it needs to use the aileron it simply crashes.
Can I have some help?
Thanks, Vinha

bcoconni, 2020-04-21:
Have you updated the aerodynamics of your model ? You need to implement the effect of your ailerons at the very least on the roll moment (below is the ailerons effect on the c172x roll moment):
https://github.com/JSBSim-Team/jsbsim/blob/bf0d3b8508b026e9958ad49720260b73b648ed2f/aircraft/c172x/c172x.xml#L1038-L1046
Note that for the EasyStar II, you might need to tune the coefficient `<value>0.23</value>`.

If you need higher fidelity, you might want to implement the ailerons effect on the other forces and moments (the c172x accounted for their effect on the yaw moment and on the side force):
https://github.com/JSBSim-Team/jsbsim/blob/bf0d3b8508b026e9958ad49720260b73b648ed2f/aircraft/c172x/c172x.xml#L950-L957
https://github.com/JSBSim-Team/jsbsim/blob/bf0d3b8508b026e9958ad49720260b73b648ed2f/aircraft/c172x/c172x.xml#L1180-L1188
There again you will most certainly need to tune the coefficients `<value>-0.05</value>` and `<value>0.0053</value>` for your particular need.

Generally speaking, the [c172x FDM](https://github.com/JSBSim-Team/jsbsim/blob/master/aircraft/c172x/c172x.xml) is a good place to start to see how ailerons are implemented in a model (search for 'aileron' in the the file).

VinhA2018, 2020-04-23:
Firstly, thanks for your support! 
I was already able to add the effects of the ailerons, however I don't really know how to really tune them. I'm pretty sure that it isn't stable because of these coefficients.
Should I tune one by one?

bcoconni, 2020-04-23:
> I was already able to add the effects of the ailerons

Given that you only mentioned `<aerosurface_scale name="Left Aileron Control">`, I was under the impression that you had only implemented the controls but not the aerodynamics.
> I'm pretty sure that it isn't stable because of these coefficients.

What do you mean by "it isn't stable" ?
> Should I tune one by one?

Most likely, yes. However, the coefficients from existing aircraft (and there are a number of them in JSBSim) can give you some insight into the value and sign of these coefficients. For example, it might be interesting to evaluate the effect of the size of the aircraft on the value of these coefficients. Then you can try to extrapolate to smaller sizes for your particular application.

VinhA2018, 2020-04-24:
> What do you mean by "it isn't stable" ? 

I mean, as soon as the the aircraft turns it starts to do random flight. I see the aircraft starting to do roll movements to the maximum. Sometimes it just turns the wrong direction demanded. Can this behavior be caused by the wrong coefficients, like the sign?

bcoconni, 2020-04-26:
> I mean, as soon as the the aircraft turns it starts to do random flight. I see the aircraft starting to do roll movements to the maximum. Sometimes it just turns the wrong direction demanded. Can this behavior be caused by the wrong coefficients, like the sign?

Indeed one possible explanation is that the coefficient magnitudes are too strong. As I mentioned earlier try to compare your model with other models and check if you are in the same ball park regarding the coefficients. A significant difference in magnitude (or in sign) between your model and others might be a hint of something going wrong.

Also be aware that the relative amplitude of the roll moments might vary with the flight configuration (velocity, turn rate, alpha, beta, etc.) and this might explain the apparent erratic behavior. For example the roll moment due to ailerons might be of higher amplitude initially then, as the yaw turn rate increases, another roll moment gets a higher magnitude and takes over your airplane roll movement giving the impression that the ailerons have lost their efficiency. There again, the solution to your problem might be tuning coefficients.

You can plot the roll moments `aero/coefficient/Clb`, `aero/coefficient/Clp`, `aero/coefficient/Clr`, etc. on a graph and observe how their relative amplitude vary and understand which one drives your aircraft trajectory at different step of the flight. Disabling all of them and re-enabling them one by one might also help understanding their relative importance in the aircraft response to the controls.
> Can the motor also be causing some trouble? When I start my simulation, only after 10 seconds that the aircraft starts to gain some velocity and starts the takeoff.

This sounds like the engine is providing too much power/thrust while idling. I guess your aircraft uses an electric engine so it might easy to adjust the idling power to a value slightly lower than friction.

rega0051, 2020-05-11:
@VinhA2018 

I built a model of a StratoSurfer that is fairly similar to the EasyStar. The aerodynamic model is based on a model built in OpenVSP. The prop model comes from the data published from APC, as noted. BTW, most of the small aircraft models I came across, before making my own, had hacked prop models: Ixx, CT, and CP were just scaled and copied from a Cub or some such, causes all kinds of strange gyroscopic and prop torque effects.

https://github.com/UASLab/OpenFlightSim/tree/master/Simulation/aircraft/StratoSurfer

bcoconni, 2020-05-27:
Hi @VinhA2018,

Have your questions been answered ? Hopefully @rega0051's model will have provided you some guidance. If you need any further support, let us know. Otherwise, could you please confirm that this issue can be closed ?

Thanks.

VinhA2018, 2020-06-15:
Hello again,
Yes, thanks for all the help!

closed 2020-06-15 by VinhA2018