JSBSim-Team/jsbsim #285: MinGW32 cannot build Python wheel packages

bcoconni, 2020-04-30:
When building JSBSim with `-DBUILD_PYTHON_MODULE=ON`, the library is built with the `gcc` compiler but, for some reason, the Python module is built with MSVC which triggers a complete newcompilation of the library. As a result, the library is compiled twice.

That's a shame because I would prefer to build the Python modules for Windows with MinGW32 rather than MSVC because it would avoid to ship the DLL `msvcp140.dll` with its questionable license with our Python Windows package.

seanmcleod, 2020-05-01:
So to refresh my memory I took a look at https://github.com/JSBSim-Team/jsbsim/issues/73 which is where the `msvcp140.dll` first cropped up.

I didn't see us discussing the option to link statically with the C/C++ library at the time as one option as opposed to distributing `msvcp140.dll`.

In terms of a questionable license I'm not sure what the potential issue is. This link from Microsoft https://docs.microsoft.com/en-us/visualstudio/productinfo/2015-redistribution-vs mentions:

> you may copy and distribute with your program any of the files within the followng folder and its subfolders except as noted below. You may not modify these files.

> C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\redist

I just double-checked and and `msvcp140.dll` does live in that directory tree.

Now while looking around for some info on Python modules and `msvcp140.dll` I came across this very detailed explanation from one of the Python developers.

> As I’m largely responsible for the compiler change (with the full support of the CPython developers, of course)

https://stevedower.id.au/blog/building-for-python-3-5

And the recommendation at the end of it is to link statically against the C/C++ runtimes.

However he mentions that `distutils` from python 3.5 onwards has been updated to build python modules using static linking. So does that mean we're using a pre-python 3.5 version of `distutils` when we build our python module?

Lastly right at the end of the page there is mention of `MinGW` and compatibility issues. Although the post was written in Aug 2015 so maybe they've been addressed since then?

bcoconni, 2020-05-03:
Thanks for the detailed answer @seanmcleod, that was very helpful.
> I didn't see us discussing the option to link statically with the C/C++ library at the time as one option as opposed to distributing `msvcp140.dll`.

Indeed, we haven't discussed such an option after we resolved the issue #73.

Since it's quite a looong answer that I have written below, here is the executive summary aka TL;DR:
* It seems there is no license issues about shipping `msvcp140.dll` with our Python packages because our license is the LGPL 2.1 and because our packages are distributed by GitHub who (hopefully) own a valid copy of MSVC. It would have been a different story if JSBSim was licensed under the GPL.
* Python build tools `distutils` have dropped support for MinGW32 (and cygwin) since Python 3.4 (which is now officially obsolete). As an obvious consequence, we cannot build our Python wheel packages with MinGW32.
* The only viable option seems to be dynamically linking JSBSim with `msvcp140.dll` and ship it with our Python packages.

Now here is the detailed answer. Brace yourselves:
### License
> In terms of a questionable license I'm not sure what the potential issue is. This link from Microsoft https://docs.microsoft.com/en-us/visualstudio/productinfo/2015-redistribution-vs mentions:

Yes, I am aware of that statement however your quotation is incomplete (bold-facing is mine):
> **Subject to the License Terms for the software,** you may copy and distribute with your program any of the files within the following folder and its subfolders except as noted below.

Which license are they referring to ? For instance the top paragraph of [Distributable Code Files for Visual Studio 2017](https://docs.microsoft.com/en-us/visualstudio/productinfo/2017-redistribution-vs#-distributable-code-files-for-visual-studio-2017) say (bold-facing is mine again):
> The following section is the "REDIST list" that is referenced in the "Distributable Code" section of the Microsoft Software License Terms for Visual Studio Enterprise 2017, Visual Studio Professional 2017, Visual Studio Community 2017 ("the software"). **If you have a validly licensed copy of such software**, you may copy and distribute with your program the unmodified form of the files listed below, subject to the License Terms for the software.

I infer from that statement that you need to own a license of Visual Studio to be granted the right to copy and distribute the aforementioned files. In the case of our CI workflow, GitHub own licenses of Visual Studio (at least, I hope they do :wink:) and our packages are distributed through their site so I can only guess that we are covered by their license. 

In addition, a lot of open source projects (including Python) are happily distributing some files from the `redist` folder with their packages so it seems the redistribution of `msvcp140.dll` and the likes is not frowned upon by Microsoft.

For a bit of context, I started being involved in open source in the early 2000 which was the time when Steve Ballmer was Microsoft CEO. Remember that guy who said that open source was a "communist cancer" ? He did all he could to shoot down open source projects with software patents, undisclosed APIs and purposely standards violations. Although his successor, Satya Nadella, is unquestionably having a friendly attitude towards open source (and this is why I did not flee from GitHub when Microsoft purchased them), it remains that, as far as I am concerned, Steve Ballmer's era left some scars. Hence my uneasiness.

In addition to that, I have come across this answer from the GNU foundation in a [FAQ about the GPL license](https://www.gnu.org/licenses/gpl-faq.en.html#WindowsRuntimeAndGPL)
> **I'm writing a Windows application with Microsoft Visual C++ (or Visual Basic) and I will be releasing it under the GPL. Is dynamically linking my program with the Visual C++ (or Visual Basic) runtime library permitted under the GPL?**
>
> [...] _You may not distribute these libraries in compiled DLL form with the program_. To prevent unscrupulous distributors from trying to use the System Library exception as a loophole, the GPL says that libraries can only qualify as System Libraries as long as they're not distributed with the program itself.

Fortunately, JSBSim is licensed under the LGPL 2.1 which allows distribution of JSBSim with `msvcp140.dll` under section 7 of the LGPL 2.1
> **7.** You may place library facilities that are a work based on the Library side-by-side in a single library together with other library facilities not covered by this License, and distribute such a combined library, provided that the separate distribution of the work based on the Library and of the other library facilities is otherwise permitted, and provided that you do these two things:
>
> * **a)** Accompany the combined library with a copy of the same work based on the Library, uncombined with any other library facilities. This must be distributed under the terms of the Sections above.
> * **b)** Give prominent notice with the combined library of the fact that part of it is a work based on the Library, and explaining where to find the accompanying uncombined form of the same work.

However, the particular case of the GPL license that I mentioned above shows that licenses are, at least to me, muddy waters.

### Compilation with MinGW32
[... cut ...] heels.sh`.

#### Mac OSX
I have yet no idea about how the dependencies to system libraries are managed on MacOSX. At the moment, our CI workflow manages to build and test successfully Python packages for MacOSX. Yet, I don't know how good they are when installed on the average Macbook.

The strategy here is to basically wait until someone posts an issue on GitHub reporting that the wheel package does not work on his/her Macbook.

### Feedback
What do you all think ? Any thought on these topics ?

seanmcleod, 2020-05-03:
Hmm, well your answer was 10x more detailed than my answer :wink 

I'd suggest maintaining our status quo of shipping `msvcp140.dll` as part of the JSBSim wheel.

bcoconni, 2020-05-06:
> Hmm, well your answer was 10x more detailed than my answer :wink:

:smile: I was not trying to compete with your answer but rather building on the foundations it has laid. When I said the links you have provided were enlightening, I really meant it :+1:. I spent a whole afternoon gathering additional information and making up my mind on that topic, so I felt it needed to be documented somewhere for future reference. Hence that long answer.
> I'd suggest maintaining our status quo of shipping `msvcp140.dll` as part of the JSBSim wheel.

Good, so I consider the issue closed.

closed 2020-05-06 by bcoconni