JSBSim-Team/jsbsim #286: Using the JSBSim Cpp API

Jaeyoung-Lim, 2020-05-17:
I am trying to use the jsbsim cpp api as explained in https://github.com/JSBSim-Team/jsbsim/blob/master/README.md#using-the-c-api and link it with my application through cmake. 

I have built jsbsim from source and installed it to my system on my default path.

However, I am not able to link my example application through cmake. Are there any examples / information documented on how I can link my application to jsbsim through cmake?

Thank you in advance,

bcoconni, 2020-05-18:
Hi @Jaeyoung-Lim,

It's not available on all platforms but we have recently included an interface to`pkg-config` (commit 6401af3) which you could use to detect the headers and libraries location.

Given that JSBSim has no dependencies, I would not anticipate any difficulty but I might wrong. Could you please provide the error(s) that CMake reports when trying to link to JSBSim ? And I would be interested to know which platform you are using as well.

bcoconni, 2020-05-27:
Hi @Jaeyoung-Lim,

Is your issue solved ? If you need further support, let us know. Otherwise, could you please confirm that this issue can be closed ?

Thanks.

bcoconni, 2020-07-02:
Contact lost with issuer. Issue closed.

closed 2020-07-02 by bcoconni