JSBSim-Team/jsbsim #287: Broken accelerations/Nx, mismatched sign of accelerations/Nz and accelerations/pilot-z-norm.

mike402, 2020-05-17:
1. Stand the aircraft on the ground, slightly tilt nose up and one wing down.
2. Compare `accelerations/N{x,y,z}` and `accelerations/pilot-{x,y,z}-norm`.
Expected:
- same values because the aircraft is not rotating, so accelerations should be the same no matter where;
- `Nx` not zero when aircraft is tilted, because of gravity.
Get: 
- sign of `Nz` and `pilot-z-norm`are the opposite;
- `Nx` zero unless accelerating, but `pilot-x-norm` works as expected.

bcoconni, 2020-05-18:
> [...] slightly tilt nose up and one wing down
> [...] Expected: same values because the aircraft is not rotating

My English might well be wrong but as far as I know "tilt" means (slightly) rotate. So I am confused about what you mean there and if it is my English that's wrong then no offense will be taken 😄 
Also could you specify if the aircraft was motionless when you made the two checks ?
> `Nx` not zero when aircraft is tilted, because of gravity.

That's quite strange because there is no `Nx`, just `Ny` and `Nz`:
https://github.com/JSBSim-Team/jsbsim/blob/d73d76c7e875141797d51eba0c45a0fd21fa9e5b/src/models/FGAuxiliary.cpp#L336-L339
> sign of `Nz` and `pilot-z-norm`are the opposite;

That's purposely, `Nz` is meant to be positive equal to 1g when sitting on ground (see the minus sign on line 208):
https://github.com/JSBSim-Team/jsbsim/blob/d73d76c7e875141797d51eba0c45a0fd21fa9e5b/src/models/FGAuxiliary.cpp#L206-L209
> same values because the aircraft is not rotating, so accelerations should be the same no matter where

Well, if the aircraft attitude has been modified (at least one angle among `phi`, `theta`, `psi`) then the orientation of gravity (1g) is modified in the aircraft body frame and so are the components (`x`, `y`, `z`) of the accelerations.

For information, below is the code at stake. All the properties you mentioned are based on accelerations measured in the body frame (variable `in.vBodyAccel`) so it's not unexpected that the components (`x`, `y`, `z`) are modified when the aircraft attitude is modified.
https://github.com/JSBSim-Team/jsbsim/blob/d73d76c7e875141797d51eba0c45a0fd21fa9e5b/src/models/FGAuxiliary.cpp#L205-L216

seanmcleod, 2020-05-18:
@mike402 I also noticed that `Nx` isn't exported via the property tree. So I added it to my local source and tested things out with the 737 model.

```c++
  // Nz is Acceleration in "g's", along normal axis (-Z body axis)
  Nz = -vNcg(eZ);
  Ny =  vNcg(eY);
  Nx =  vNcg(eX);
```

The nose gear and the main gear all have their z-coordinate at -84.

```xml
        <contact name="Nose Gear" type="BOGEY">
            <location unit="IN">
                <x> 158 </x>
                <y>   0 </y>
                <z> -84 </z>
            </location>
```

So using the B737 runway example script, removing the events and simply outputting the 6 accelerations I see the following:

```sh
Repeating Notify (Event 0) executed at time: 100.007933
    accelerations/n-pilot-x-norm = 0.005480
    accelerations/n-pilot-y-norm = 0.002421
    accelerations/n-pilot-z-norm = -0.996545
    accelerations/Nx = 0.005480
    accelerations/Ny = 0.002421
    accelerations/Nz = 0.996545
```

So the body accelerations and the pilot accelerations are identical (excluding the sign difference that @bcoconni explained).

Then I increased the nose gear height.

```xml
        <contact name="Nose Gear" type="BOGEY">
            <location unit="IN">
                <x> 158 </x>
                <y>   0 </y>
                <z> -200 </z>
            </location>
```

And then I see the following output:

```sh
Repeating Notify (Event 0) executed at time: 100.007933
    accelerations/n-pilot-x-norm = 0.235143
    accelerations/n-pilot-y-norm = 0.002429
    accelerations/n-pilot-z-norm = -0.968421
    accelerations/Nx = 0.235143
    accelerations/Ny = 0.002429
    accelerations/Nz = 0.968421
```

Which is what you were expecting right?

So the question is how were you getting `Nx`?

mike402, 2020-05-18:
By "tilt" I mean of course "stand it not norizontally", no movement.

Sorry, turns out I was calculating `Nx` myself, from `accelerations/udot-ft_sec2`, which is wrong. But there is no other way, because it is not exposed to the property tree -- so it would be nice if it was added.

bcoconni, 2020-05-18:
> But there is no other way, because it is not exposed to the property tree -- so it would be nice if it was added.

@seanmcleod since you have already coded (and tested) that feature, would you mind submitting a pull request ? I'll then push the commit to FlightGear.

Thanks.

bcoconni, 2020-05-18:
> By "tilt" I mean of course "stand it not horizontally", no movement.

Ok, and I just realized that I misread your post, since you said the aircraft is not **rotating** while I wrongly read that it was not **rotated** hence my confusion.
> Sorry, turns out I was calculating `Nx` myself, from `accelerations/udot-ft_sec2`, which is wrong.

Indeed as a first approximation, `accelerations/{u,v,w}dot-ft_sec2` is the sum of the gravity and of the acceleration `in.vBodyAccel` which we discussed above. As a consequence it is zero in all three directions `x`, `y` and `z` for a motionless aircraft (the gravity being counteracted by the ground reactions).

seanmcleod, 2020-05-18:
@bcoconni I've submitted a pull request which adds the export of Nx.

bcoconni, 2020-05-18:
> I've submitted a pull request which adds the export of Nx.

Just merged it. Thanks.

bcoconni, 2020-05-18:
And I have just pushed the change to FlightGear as well.

@mike402 Has all your questions/concerns been answered ? Or do you need further support ?

mike402, 2020-05-23:
Thanks, everything works now!

closed 2020-05-24 by seanmcleod