JSBSim-Team/jsbsim #289: Synchronization of channel properties

kvnjonathan70, 2020-05-18:
hi all,
I am having some doubt regarding synchronization of channel properties. In a/c config file i have added a system file which has the following code,
<system>
<property>x</property>
<property>y</property>
<channel>
    <pure_gain name="y">
      <input>x</input>
      <gain>2 </gain>
    </pure_gain>
</channel>  
</system>

In my script file i have given following command,

<use aircraft="mb339" initialize="reset00"/>
  <run start="0.0" end="10" dt="0.02">    
    <event name="calculation">  
       <condition> simulation/sim-time-sec ge 0 </condition>  
  <set name="x" value="10"/>
      <notify>
       <property>y</property>
      </notify>
    </event>

so when i run the script i get zero value which to my knowledge should be 20,
calculation (Event 0) executed at time: 0.020000
    y = 0.000000

Also when i tried removing  <property>y</property> to check if <pure_gain name="y"> creates a property and produce desired value but i get error that property y does not exist and only property X can been seen in --catalog

Thanks in advance.

bcoconni, 2020-05-18:
> Also when i tried removing `<property>y</property>` to check if `<pure_gain name="y">` creates a property and produce desired value but i get error that property y does not exist and only property X can been seen in `--catalog`

You have hit one of the obscure "feature" of JSBSim: in flight component systems (i.e. `<pure_gain>` and the likes), output variables can silently be prepended with `fcs/`. In your particular case, the variable created by `<pure_gain>` is `fcs/y` not just `y`.

The comments in code lines 261-266 below explain that this has been done for backward compatibility with earlier JSBSim designs. If you want JSBSim to leave your properties names untouched then specify them with a "path" i.e. with a least one slash `/` such as `mb339/y`. Doing so will let you complete control over your properties names.

https://github.com/JSBSim-Team/jsbsim/blob/d73d76c7e875141797d51eba0c45a0fd21fa9e5b/src/models/flight_control/FGFCSComponent.cpp#L261-L274

seanmcleod, 2020-05-18:
@kvnjonathan70 see why I mentioned you should rather post the query here rather than emailing me direct.

I said you would benefit from @bcoconni's knowledge 😉 

Plus as I mentioned it will potentially help other users in future if they do a search when having a similar issue.

bcoconni, 2020-05-18:
> I said you would benefit from @bcoconni's knowledge :wink:

If I can be of any help :smiley: 
> Plus as I mentioned it will potentially help other users in future if they do a search when having a similar issue.

Indeed and I am under the impression @kvnjonathan70 is not the first to stumble upon that "subtlety" (even though I cannot grab any other occurrences yet). This topic might be a good candidate for our FAQ ?

kvnjonathan70, 2020-05-19:
@bcoconni , i understood the fact you said and tried, i do get property as fcs/y for y, and for custom i do get the name i had mentioned as "gain/y.

But the core problem does not seem to be solved, here is my system file code below,
```
<system>
<property>x</property>
<channel>
    <pure_gain name="gain/y">
      <input>x</input>
      <gain>2 </gain>
    </pure_gain>
</channel>
</system>
```
--catalog 
```
  x (RW)
    gain/y (RW)
```

script value
```
  <use aircraft="mb339" initialize="reset00"/>
  <run start="0.0" end="10" dt="0.02">
 <event name="calculation">
    <condition> simulation/sim-time-sec ge 0
     </condition>
	    <set name="x" value="10"/>
      <notify>
     <property>gain/y</property>
      </notify>
    </event>			   
  </run>
</runscript>

```

result 

```
calculation (Event 0) executed at time: 0.020000
    gain/y = 0.000000
```
Its still zero 

? 

> Plus as I mentioned it will potentially help other users in future if they do a search when having a similar issue. 
@seanmcleod  yes i agree with you :)

seanmcleod, 2020-05-19:
@kvnjonathan70 what happens if you have a separate event triggered at say `sim-tim-sec ge 0.03` and use `notify` to display the property `gain/y'?

I haven't checked the source code, or tried your example myself, just a hunch.

kvnjonathan70, 2020-05-19:
@seanmcleod  still no change, i am attaching the file below for testing 

[MB339.zip](https://github.com/JSBSim-Team/jsbsim/files/4648623/MB339.zip)

seanmcleod, 2020-05-19:
@kvnjonathan70 you didn't include the engine model in the zip file, also your script referred to a non-existent `reset00.xml` file.

After sorting those out and modifying the script to be:

```xml
<?xml version="1.0" encoding="utf-8"?>
<?xml-stylesheet type="text/xsl" href="http://jsbsim.sf.net/JSBSimScript.xsl"?>
<runscript xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xsi:noNamespaceSchemaLocation="http://jsbsim.sf.net/JSBSimScript.xsd"
    name="Test gain in mb339.">

  <use aircraft="mb339" initialize="reset01"/> 

  <run start="0" end="5" dt="0.008333">

    <property value="0"> simulation/notify-time-trigger </property>

    <event name="calculation">
       <condition> simulation/sim-time-sec ge 0 </condition>
       <set name="x" value="10"/>
      <notify>
        <property>gain/y</property>
        <property>x</property>
      </notify>
    </event>

    <event name="calculation-check">
      <condition> simulation/sim-time-sec ge 0.03 </condition>
      <set name="x" value="10"/>
      <notify>
        <property>gain/y</property>
        <property>x</property>
      </notify>
    </event>

    <event name="Repeating Notify" persistent="true">
      <description>Output message at 1 second intervals</description>
      <notify>
        <property>gain/y</property>
        <property>x</property>
      </notify>
      <condition> simulation/sim-time-sec >= simulation/notify-time-trigger </condition>
      <set name="simulation/notify-time-trigger" value="1" type="FG_DELTA"/>
    </event>

  </run>

</runscript>
```

I get the following output:

```sh
calculation (Event 0) executed at time: 0.008333
    gain/y = 0.000000
    x = 10.000000

Repeating Notify (Event 2) executed at time: 0.008333
    gain/y = 0.000000
    x = 10.000000

Start: Tuesday May 19 2020 11:15:35 (HH:MM:SS)

calculation-check (Event 1) executed at time: 0.033332
    gain/y = 20.000000
    x = 10.000000

Repeating Notify (Event 2) executed at time: 1.008293
    gain/y = 20.000000
    x = 10.000000

Repeating Notify (Event 2) executed at time: 2.008253
    gain/y = 20.000000
    x = 10.000000

Repeating Notify (Event 2) executed at time: 3.008213
    gain/y = 20.000000
    x = 10.000000

Repeating Notify (Event 2) executed at time: 4.008173
    gain/y = 20.000000
    x = 10.000000

Repeating Notify (Event 2) executed at time: 5.008133
    gain/y = 20.000000
    x = 10.000000
```

bcoconni, 2020-05-19:
The thing is that script events are evaluated at the end of a time step. So the event `calculation` is evaluated at the end of the time step for which the condition `simulation/sim-time-sec ge 0` is true i.e. after the first time step is completed.

If you want to set property values before the first time step is executed (aka initial conditions), then you should set these values outside of events, just after the`<run>` tag. Below is the "diff" for the modifications you should make to get the value you are expecting for the first time step.
```diff
@@ -1,9 +1,9 @@
   <use aircraft="mb339" initialize="reset00"/>
   <run start="0.0" end="10" dt="0.02">
+    <property value="10"> x </property>
  <event name="calculation">
     <condition> simulation/sim-time-sec ge 0
      </condition>
-	    <set name="x" value="10"/>
       <notify>
      <property>gain/y</property>
      <input>x</input>
```

kvnjonathan70, 2020-05-19:
@seanmcleod 
@bcoconni 
I am soooo much thankful to you, :)

@seanmcleod I will keep in mind that i provide complete files for testing, please accept my apology. 

The problem was that i was in an impression that "ge 0" would mean that event will execute  right from the beginning which is  start="0.0" and would include 0.03 sec as its greater then zero, so i would not need to provide  `<property value="10"> x </property>` as the initial condition. but after checking Repeating Notify things  things got clear.

> The thing is that script events are evaluated at the end of a time step. So the event calculation is evaluated at the end of the time step for which the condition simulation/sim-time-sec ge 0 is true i.e. after the first time step is completed.

I understood.

closed 2020-05-19 by kvnjonathan70