JSBSim-Team/jsbsim #291: Actuator <lag> expects a number only

Octal450, 2020-05-23:
Hi guys,
I am needing dynamic lag on an actuator in my control system, seems that JSBsim only accepts a number there, and not a property like lag_filter does.

My syntax is as follows:

`		<actuator name="fcs-actual/rudder/final-actuator">
			<input>fcs-actual/rudder/cmd-deg</input>
			<rate_limit>fcs-actual/rudder/rate</rate_limit>
			<lag>fcs-actual/rudder/lag</lag>
			<output>fcs-actual/rudder/final-deg</output>
		</actuator>`

Can this be added please, that I can put a property (or table) there?

Kind Regards,
Josh

seanmcleod, 2020-05-24:
@Octal450 about 10 years ago I needed dynamic property support for `rate_limit, lag, delay` in the actuator for a project I was working on, so I added support for properties but never got round to submitting it to JSBSim.

In the meantime I see `rate_limit` does support being specified as a property as well as a numeric.

I'm busy adding support for `lag` to also support being specified as a property or numeric, I'll submit a pull request for review.

Octal450, 2020-05-24:
Thank you much @seanmcleod 
Kind Regards,
Josh

bcoconni, 2020-05-26:
Hi all, while I was reviewing @seanmcleod PR #292, I realized that the exact purpose of this change is not clear to me: is this feature intended to change the lag parameter τ a couple of times during a simulation or is it intended to allow a continuous modification of τ ?

I am asking the question because currently the numerical integration of the lag filter is made assuming that the lag parameter τ is constant. So if τ is continuously varying then the numerical scheme will introduce a non negligible error that will accumulate as the simulation progresses. On the hand, if the lag parameter τ will seldomly be changed during a simulation then I guess the error can be deemed acceptable.

The use case is of primary importance because it will tell whether or not we need to modify the numerical algorithm. If we choose not to modify the numerical integration scheme, then I think we should document the limitation.

seanmcleod, 2020-05-26:
My personal use case and the way I've used it over the years is for the lag parameter (and the rate limit and delay parameters) to be set typically once for a lesson/task, e.g. offset landing task, but in particular if the task is an up and away task the parameter can be changed during the lesson by the instructor via an instructor station.

So these are discrete changes, typically with the aircraft trimmed in-between a change as opposed to continuous changes based on say a table lookup based on for example air-density or something.

Having property support allows my instructor station to remotely change the value by updating the property, rather than having to edit the aircraft file in-between and then restart the lesson. 

```xml
    <flight_control name="FCS: 737">

      <property value="36"> actuators/pitch-fcs-delay </property>
      <property value="0"> actuators/elevator-lag </property>
      <property value="0"> actuators/elevator-rate-limit </property>

      <property value="0"> actuators/roll-fcs-delay </property>
      <property value="0"> actuators/aileron-lag </property>
      <property value="0"> actuators/aileron-rate-limit </property>

      <property value="0"> actuators/yaw-fcs-delay </property>
      <property value="0"> actuators/rudder-lag </property>
      <property value="0"> actuators/rudder-rate-limit </property>
```

So personally for my uses cases I don't need an update to the numerical integration algorithm and would be happy to add some documentation regarding the limitation.

bcoconni, 2020-05-27:
@Octal450, I have just pushed the patch to FlightGear `next` branch. Could you please test and let us know if that resolves your issue ?

Thanks.

bcoconni, 2020-07-02:
@Octal450 Have you had any chance in testing this feature in FlightGear ?
Thanks.

Octal450, 2020-07-02:
@bcoconni I apologize, I had a condensed math class so it took all my time away from FG. I will test this shortly and report back.

Thank you
josh

bcoconni, 2020-09-19:
@Octal450 This issue is now closed. Would you find an issue, please re-open it.

closed 2020-09-19 by bcoconni