JSBSim-Team/jsbsim #293: Actuator delay causes trim issues

seanmcleod, 2020-05-24:
@rega0051 I noticed your commit message for your F450 quadcopter that you mentioned in a separate JSBSim issue - https://github.com/UASLab/OpenFlightSim/commit/c33769ada4845c3ac7a083c9cca2115e596d6c55

In which you mentioned that having an actuator delay broke the trim routines. I've recently taken a look at the actuator code while adding the ability to use a property in addition to a numeric for the lag property and noticed that there is explicit code in the actuator implementation to handle the trim status.

https://github.com/JSBSim-Team/jsbsim/blob/2f2398cbafc0fc4f14dc6157d1923548ec8804ba/src/models/flight_control/FGActuator.cpp#L136-L167

So `initialized` is set to 0 if the trim routines are active. Then the individual methods processing properties like lag, rate limit etc. then use the state of `initialized` to decide whether to model their specific effect. Here is an example from the lag routine.

https://github.com/JSBSim-Team/jsbsim/blob/2f2398cbafc0fc4f14dc6157d1923548ec8804ba/src/models/flight_control/FGActuator.cpp#L196-L207

However, the delay for modeling transport delay isn't modeled by `FGActuator`, rather it's modeled by it's base class `FGFCSComponent::Delay()` which doesn't make use of `initialized`.

https://github.com/JSBSim-Team/jsbsim/blob/cff1949a2ce177323afa666fdcf6866ca7b93604/src/models/flight_control/FGFCSComponent.cpp#L223-L229

Which means that the delay is modeled while the trim routines are running.

Having a closer look at `FGActuator` the following do take the trim status into account:

- Lag
- Hysteresis
- Rate Limit

The following don't:

- Delay
- Bias
- Deadband

I guess it makes sense that bias and deadband don't worry about the trim routines being active, but we should make delay aware.

rega0051, 2020-05-25:
This is great. I hadn't dug into the source of the initialized flag, should have done so. This actually plays into my concerns about nonlinearities during linearizing as well, #227. 

I think the Deadband should be flagged as well.

seanmcleod, 2020-05-25:
I can look into updating delay and deadband to take trim status into account.

Interestingly, or maybe by pure luck, I've been using a delay of 300ms for some of the test pilot lessons using the 737 model and haven't had an issue with having the aircraft trim at startup of the lesson.

bcoconni, 2020-05-26:
@seanmcleod 
Good catch ! Just make sure that all elements of `output_array` are reset to the trimmed output values if the trim has succeeded and nothing is done when the trim has failed. Something like
```c++
if (trim_success) {
  for (unsigned int i=0; i<delay;++i) output_array[i] = Output;
}
```

@rega0051 
> I think the Deadband should be flagged as well.

I guess extra care should be taken so that trim algorithms are forced to set a value outside of the dead band but I would advise against disabling it during trim as it would result in a solution to a different configuration (no dead band) than the targeted one (with dead band). As a result, the trimmed position might not result in a steady state solution once the simulation continues.

seanmcleod, 2020-05-26:
@bcoconni yep I hadn't thought of populating the `output_array` if the trim succeeds, but it makes sense.

Having a quick look at the code it isn't immediately obvious that there is a mechanism to get the result of the trim operation.

`GetTrimStatus()` returns a `bool` indicating whether a trim operation is in progress or not. I originally thought it might offer more status info e.g. `trim_busy`, `trim_succeeded`, `trim_failed` etc.

So looking at the code that invokes the trim routines, there are 2 instances.

https://github.com/JSBSim-Team/jsbsim/blob/2661d7c80607aa2e86676b96a24eaf7dad87c1a9/src/FGFDMExec.cpp#L1136-L1153

In this case the success or failure is returned as a `bool` from `DoTrim()` but it's simply a local variable that isn't stored. If the trim fails then an exception is thrown, which I'm pretty sure currently results in the app exiting.

The other instance of performing a trim is if the trim is specified as part of the initialization file.

https://github.com/JSBSim-Team/jsbsim/blob/3cefc8678c0c60c4e26350b065a202d6387b6cc2/src/JSBSim.cpp#L466-L481

In this case the success or failure of `DoTrim()` isn't captured at all currently.

So something along these lines?

```c++
void FGFCSComponent::Delay(void)
{
  if (!fcs->GetTrimStatus()) {
    if (trim_was_active) {
      trim_was_active = false;
      if (fcs->TrimSucceeded())
        for (unsigned int i = 0; i < delay; ++i) output_array[i] = Output;
    }
    output_array[index] = Output;
    if ((unsigned int)index == delay - 1) index = 0;
    else index++;
    Output = output_array[index];
  } else {
    trim_was_active = true;
  }
}
```

With the implementation of `TrimSuceeded()` to return the success/failure of the last trim operation.

bcoconni, 2020-05-26:
Indeed the situation is more convoluted than I expected. So I suggest to favor legibility over performance, in which case the whole vector `output_array` would be updated (i.e. filled) as long as `GetTrimStatus()` returns `true`. C++11 has a standard function for that very purpose: `std::fill`.
```c++
void FGFCSComponent::Delay(void)
{
  if (fcs->GetTrimStatus()) {
    // Update the whole history since the aircraft is trimmed in a steady state position.
    std::fill(output_array.begin(), output_array.end(), Output);
  }
  else {
    output_array[index] = Output;
    if ((unsigned int)index == delay-1) index = 0;
    else index++;
    Output = output_array[index];
  }
}
```
If your compiler complains about not finding `std::fill` then you might need to add `#include <algorithm>` to the file.

Of course, that would update `output_array` even if the trim failed but I realized that we are already doing something similar with the time steps history of the Adams-Bashforth schemes: all the time steps are reset unconditionally upon trim completion, whether it succeeded or failed.
https://github.com/JSBSim-Team/jsbsim/blob/8e1cb46c6eac0d15ce1fc6f55026de139c605e24/src/initialization/FGTrim.cpp#L338

seanmcleod, 2020-05-26:
Busy testing with VS2019, no need to include `<algorithm>`. Will submit a pull request.

seanmcleod, 2020-05-26:
For testing I added a delay to the elevator actuator of the C172x aircraft and ran the C1722 script which performs a trim.

Before the code changes to `Delay()` the trim fails.

```sh
  Full Trim
  Sorry, qdot doesn't appear to be trimmable

  Trim failed
  Trim Results:
       Angle of Attack:   1.27  wdot:  1.50e+01 Tolerance: 1.000000e-03  Failed
              Throttle:   0.69  udot:  4.39e-01 Tolerance: 1.000000e-03  Failed
            Pitch Trim:   0.00  qdot:  2.12e+00 Tolerance: 1.000000e-04  Failed
            Roll Angle:  -0.11  vdot: -1.93e-02 Tolerance: 1.000000e-03  Failed
              Ailerons:  -0.04  pdot:  3.26e-01 Tolerance: 1.000000e-04  Failed
                Rudder:  -0.00  rdot:  2.13e-04 Tolerance: 1.000000e-04  Failed
FATAL ERROR: JSBSim terminated with an exception.
The message was: Trim Failed
```

After the code changes the trim succeeds.

```sh
  Full Trim

  Trim successful
  Trim Results:
       Angle of Attack:   1.11  wdot: -1.35e-04 Tolerance: 1.000000e-03  Passed
              Throttle:   0.71  udot: -2.86e-05 Tolerance: 1.000000e-03  Passed
            Pitch Trim:   0.20  qdot: -3.94e-08 Tolerance: 1.000000e-04  Passed
            Roll Angle:  -0.11  vdot:  1.12e-05 Tolerance: 1.000000e-03  Passed
              Ailerons:  -0.04  pdot:  2.63e-06 Tolerance: 1.000000e-04  Passed
                Rudder:  -0.00  rdot: -2.89e-18 Tolerance: 1.000000e-04  Passed
```

@rega0051 once the pull request is merged please test it out with your F450 model

bcoconni, 2020-05-26:
@seanmcleod the PR #297 has been pushed to the `master` branch.
Thanks !

seanmcleod, 2020-06-11:
@rega0051 have you had a chance to test this change with your F450 model? Would be a useful confirmation and if so I'll close the issue.

rega0051, 2020-06-22:
Sorry for the delay :)  Yes, that did fix the issue!

seanmcleod, 2020-06-22:
@rega0051 thanks for the confirmation.

closed 2020-06-22 by seanmcleod