JSBSim-Team/jsbsim #295: Actuator - request: set initial positoon

legoboyvdlp, 2020-05-25:
Hi,
I think ability to specific initial position of an actuator would be good :) I didn't see that in the documents, sorry if it already exists, but it would be helpful, rather than having to use /sim/elapsed-sec to move valves to the initial position I want during startup of the simulator.

bcoconni, 2020-05-26:
Hi @legoboyvdlp,

Well there is a tag `<init>` that is currently read by JSBSim - including for the actuator - but nothing is done from the values after having been read.
https://github.com/JSBSim-Team/jsbsim/blob/8e1cb46c6eac0d15ce1fc6f55026de139c605e24/src/models/flight_control/FGFCSComponent.cpp#L113-L118
They were introduced by the commit f27e2f38c51048ef3c0191d9db814835928daad7 which did not anything from them. I guess we can use this feature to set the actuator (and also some other flight control components) initial position.

bcoconni, 2020-05-26:
I'd like to clarify what exactly is the problem: what happens when you initialize the `<input>` parameter of the actuator to the initial position value ? Are you obtaining the initial position you are looking for ? If not, what are you obtaining ?
In other words, what are you doing ? What are you getting ? And what were you expecting to get ?

bcoconni, 2020-07-02:
@legoboyvdlp, could you please answer the questions above ? Thanks.

legoboyvdlp, 2020-07-03:
My apologies. Please feel free to remind me sooner :) 

I just made a test actuator. 
If I init the <input> to 1 in set file it starts at 1. This is good.

If I output from another fcs component, such as a <switch>, which should start with value 1,  I have set rate_limit to zero (to simulate un-powered valve) and yet it starts with 1.  It also works.

This indicates my problem in the original post is in my system, not your code.

closed 2020-07-03 by legoboyvdlp