JSBSim-Team/jsbsim #298: Is there a way initialise the simulation with the aircraft in particular state (i.e.: aircraft is airborne at a particular state, with particular speed etc)?

DheerendraTomar, 2020-05-27:
Hi everyone, 

I am new to JSBSim and would like to know If it is possible to initialise a320 in JSBSim in a particular state? Also, Can we use joystick's input to control the aircraft directly? I came across this [JSBSim newsletter](http://jsbsim.sourceforge.net/Newsletter_2_2.pdf) in which they use sockets for changing the properties? 

Please let me know , if you need more clarity of what I am trying to do?

kvnjonathan70, 2020-05-27:
well you can start from this link below
https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGOutput.html

seanmcleod, 2020-05-27:
Take a look at the initialization file support. 

For example here is `aircraft\737\cruise_init.xml`

```xml
<?xml version="1.0" encoding="utf-8"?>
<initialize name="cruise">
  <!--
    This file sets up the aircraft to start off
    at cruise conditions.
  -->
  <altitude unit="FT"> 30000.0 </altitude>
  <vt unit="FT/SEC">     750.0 </vt>
  <gamma unit="DEG">       0.0 </gamma>
  <latitude unit="DEG">   47.0 </latitude>
  <longitude unit="DEG"> 122.0 </longitude>
  <phi unit="DEG">         0.0 </phi>
  <psi unit="DEG">       225.0 </psi>
  <beta unit="DEG">        0.0 </beta>
</initialize>
```

Typically you'll want to trim the aircraft at these starting conditions. You can either initiate the trim from within a script, for example take a look at `scripts\737_cruise.xml`, here are a couple of relevant snippets from it.

```xml
  <use aircraft="737" initialize="cruise_init"/> 

    <!--
      For "do_simple_trim" (Classic trim):
      0: Longitudinal
      1: Full
      2: Ground
      3: Pullup
      4: Custom
      5: Turn
      6: None
    -->
    
    <event name="Trim">
      <condition> simulation/sim-time-sec gt 0.1 </condition>
      <set name="simulation/do_simple_trim" value="1"/>
      <delay>5.0</delay>
      <notify>
        <property>propulsion/engine[0]/n2</property>
        <property>propulsion/engine[1]/n2</property>
        <property>propulsion/engine[0]/thrust-lbs</property>
        <property>propulsion/engine[1]/thrust-lbs</property>
        <property>velocities/vc-kts</property>
        <property>velocities/vc-fps</property>
        <property>velocities/vt-fps</property>
        <property>attitude/phi-rad</property>
        <property>attitude/theta-rad</property>
        <property>attitude/psi-rad</property>
      </notify>
    </event>
```

Or you can also have the trim initiated when the initialisation file is read in by adding a `<trim>` element. If you do go with this approach and you're trimming an aircraft in the air as opposed to on the ground then you'll want to also add a `<running>` element to ensure that the engines are running, e.g.

```xml
  running> -1 </running>
 <trim> 1 </trim>
```

The -1 means all engines.

Not sure what you're planning to do with JSBSim, but I presume you've looked at FlightGear?

DheerendraTomar, 2020-05-27:
> well you can start from this link below
> https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGOutput.html

@kvnjonathan70 Thank you

DheerendraTomar, 2020-05-27:
> Take a look at the initialization file support.
> 
> For example here is `aircraft\737\cruise_init.xml`
> 
> ```
> <?xml version="1.0" encoding="utf-8"?>
> <initialize name="cruise">
>   
>   <altitude unit="FT"> 30000.0 </altitude>
>   <vt unit="FT/SEC">     750.0 </vt>
>   <gamma unit="DEG">       0.0 </gamma>
>   <latitude unit="DEG">   47.0 </latitude>
>   <longitude unit="DEG"> 122.0 </longitude>
>   <phi unit="DEG">         0.0 </phi>
>   <psi unit="DEG">       225.0 </psi>
>   <beta unit="DEG">        0.0 </beta>
> </initialize>
> ```
> 
> Typically you'll want to trim the aircraft at these starting conditions. You can either initiate the trim from within a script, for example take a look at `scripts\737_cruise.xml`, here are a couple of relevant snippets from it.
> 
> ```
>   <use aircraft="737" initialize="cruise_init"/> 
> 
>     
>     
>     <event name="Trim">
>       <condition> simulation/sim-time-sec gt 0.1 </condition>
>       <set name="simulation/do_simple_trim" value="1"/>
>       <delay>5.0</delay>
>       <notify>
>         <property>propulsion/engine[0]/n2</property>
>         <property>propulsion/engine[1]/n2</property>
>         <property>propulsion/engine[0]/thrust-lbs</property>
>         <property>propulsion/engine[1]/thrust-lbs</property>
>         <property>velocities/vc-kts</property>
>         <property>velocities/vc-fps</property>
>         <property>velocities/vt-fps</property>
>         <property>attitude/phi-rad</property>
>         <property>attitude/theta-rad</property>
>         <property>attitude/psi-rad</property>
>       </notify>
>     </event>
> ```
> 
> Or you can also have the trim initiated when the initialisation file is read in by adding a `<trim>` element. If you do go with this approach and you're trimming an aircraft in the air as opposed to on the ground then you'll want to also add a `<running>` element to ensure that the engines are running, e.g.
> 
> ```
>  <running> -1 </running>
>  <trim> 1 </trim>
> ```
> 
> The -1 means all engines.
> 
> Not sure what you're planning to do with JSBSim, but I presume you've looked at FlightGear?

@seanmcleod Thank you so much. This script you provided helps a lot. Well, I want a RL environment. I built one using X-Plane before as I was very new to this field and didn't know much about simulator, FDMs etc. The main problem I face with my X-Plane RL environment is that every time when the episode terminates and things reset from start, X-Plane is highly unstable and upsets the aircraft. I don't need GUI for my work and I didn't find any way to use X-Plane's FDM. That's how I came across JSBSim and I have realised that it is the best thing for my project. I have also came across [this RL environment](https://github.com/Gor-Ren/gym-jsbsim) and will be exploring it. 

In the end, I am very thankful for the open source community for support.

seanmcleod, 2020-05-27:
@DheerendraTomar okay so RL = Reinforcement Learning I assume? Based on the  RL environment link you mentioned.

DheerendraTomar, 2020-05-27:
> @DheerendraTomar okay so RL = Reinforcement Learning I assume? Based on the RL environment link you mentioned.

@seanmcleod Yes. RL = Reinforcement Learning

seanmcleod, 2020-05-27:
I just glanced at Gor-Ren's MSc thesis and it looks like it provides a pretty useful basis for RL using JSBSim.

DheerendraTomar, 2020-05-27:
> I just glanced at Gor-Ren's MSc thesis and it looks like it provides a pretty useful basis for RL using JSBSim.

Thank you. I am reading the same.

DheerendraTomar, 2020-06-18:
Hi everyone,

I am still new to JSBSim. I am using JSBSim with Python. The Python code goes like this..

`

    import jsbsim
    import time

    fdm = jsbsim.FGFDMExec('C:\JSBSim', None)
    fdm.load_model('A320')
    fdm.load_ic('basic_ic.xml', True)
    fdm.run_ic()
    fdm.set_output_directive('data_output/flightgear.xml' )
    #fdm.load_script('scripts/737_cruise.xml')

    fdm.print_simulation_configuration()
    frame_duration = fdm.get_delta_t()
    sleep_nseconds = (frame_duration if True else sleep_period) * 1E9
    current_seconds = initial_seconds = time.time()
    result = fdm.run()

    print(result)

    while result and fdm.get_sim_time() <= 100.0:
        current_seconds = time.time()
        actual_elapsed_time = current_seconds - initial_seconds
        sim_lag_time = actual_elapsed_time - fdm.get_sim_time()

        for _ in range(int(sim_lag_time / frame_duration)):
            result = fdm.run()
            current_seconds = time.time()
`

The initialization file for the aircraft is at pastebin [basic_ic](https://pastebin.com/afGB2eFH)

The set ouput directive script(flightgear.xml) goes like this...

`    <?xml version="1.0" encoding="utf-8"?>
    

    <output name="localhost" type="FLIGHTGEAR" protocol="UDP" port="5550" rate="60">
    </output>

`

Flight Gear is started as mentioned in flightgear.xml file.

The console outputs are show in the the snapshots in this [drive link](https://drive.google.com/drive/folders/12hk3iYnYLq7xvSa85OKX3zPAoLJ1sk4b?usp=sharing)

I need a little guidance to set this up and How JSBSim runs the simulation when I run fdm.run(). Does it run whole simulation at once or it runs one time step and wait for next instruction. 

Note: I don't have any script file. I want the stick controls and throttle to provided by python and then run a single time step in the simulation and observe the changes in Flight Gear (Once everything works fine, will deactivate flightgear part). I will be later taking outputs like aircraft state from JSBSim as well.

seanmcleod, 2020-06-18:
`fdm.run()` executes a single time-step.

Here is a small snippet from one of the python tests, found in `tests\TestActuator.py`.

So it processes the initial conditions and then sits in a loop executing a single time-step until a specified amount of time has elapsed.

Within the loop it checks the position of an aileron. In your case during each time-step you could set the inceptor (stick, rudder, throttle etc.) command.

```python
        fdm.run_ic()

        while fdm.run() and fdm.get_sim_time() < time_limit:
            aileron_pos = fdm['fcs/left-aileron-pos-rad']
            self.assertEqual(aileron_pos, 0.0,
                            msg="Failed running the script %s at time step %f\nProperty fcs/left-aileron-pos-rad is non-zero (%f)" % (self.script_path, fdm.get_sim_time(), aileron_pos))

```

DheerendraTomar, 2020-06-18:
@seanmcleod Thank you so much for the explanation. The test files explains a lot, will explore them and update.

DheerendraTomar, 2020-06-19:
I am using A320. The aircraft is airborne at 5000 feet. But, I am facing this strange problem. Thust from one of engine is decreasing, I can't figure it out. 

The basic code for this is:

`
import os
import jsbsim

fdm = jsbsim.FGFDMExec('C:\JSBSim', None)
fdm.load_model('A320')
fdm.load_ic('basic_ic.xml', True)
fdm['propulsion/set-running'] = -1
fdm['ic/alpha-deg'] = 3.2758
fdm['ic/theta-deg'] = 3.2787
fdm['ic/psi-true-deg'] = 307
fdm['gear/gear-cmd-norm'] = 0
fdm['ic/q-rad_sec'] = 0.0
fdm['ic/p-rad_sec'] = 0.0
fdm['ic/r-rad_sec'] = 0.0
fdm['ic/roc-fps'] = 0.00001
fdm['fcs/pitch-trim-cmd-norm'] = -0.1122
fdm['fcs/roll-trim-cmd-norm'] = 0.0
fdm['fcs/yaw-trim-cmd-norm'] = 0.0
fdm['fcs/flap-cmd-norm'] = 0
fdm['fcs/elevator-cmd-norm'] = 0.0
fdm['fcs/roll-cmd-norm'] = 0.0
fdm['fcs/aileron-cmd-norm'] = 0.0
fdm['fcs/throttle-cmd-norm'] = 0.8
fdm['fcs/mixture-cmd-norm'] = 0.8
fdm['simulation/dt'] = 0.008333
#fdm['simulation/sim-time-sec'] = 100.0

fdm.run_ic()

print('*'*8 + 'Initial Conditions' + '*'*8 )
print(fdm['ic/vc-kts'])
print(fdm['ic/h-sl-ft'])
print(fdm['ic/u-fps'])
print(fdm['ic/p-rad_sec'])
print(fdm['ic/q-rad_sec'])
print(fdm['ic/r-rad_sec'])
print(fdm['ic/roc-fps'])
print(fdm['ic/alpha-deg'])
print(fdm['ic/theta-deg'])

fdm.print_simulation_configuration()
for i in range(100):
    fdm['fcs/elevator-cmd-norm'] = -0.2
    fdm['fcs/roll-cmd-norm'] = 0.0
    fdm['fcs/aileron-cmd-norm'] = 0.0
    fdm['fcs/throttle-cmd-norm'] = 0.8
    fdm['fcs/mixture-cmd-norm'] = 0.8
    fdm.run()
    print('*'*8 + 'Current Conditions' + '*'*8 )
    print('Current Altitude: %f'%(fdm['position/h-sl-ft']))
    print('Pitch degree: %f'%(fdm['attitude/pitch-rad']*57.2958))
    print('Roll degree: %f'%(fdm['attitude/roll-rad']*57.2958))
    print('Heading degree: %f'%(fdm['attitude/psi-rad']*57.2958))
    print('Engine 1 N2: %f'%(fdm['propulsion/engine[0]/n2']))
    print('Engine 2 N2: %f'%(fdm['propulsion/engine[1]/n2']))
    print('Engine 1 thrust-lbs: %f'%(fdm['propulsion/engine[0]/thrust-lbs']))
    print('Engine 2 thrust-lbs: %f'%(fdm['propulsion/engine[1]/thrust-lbs']))

`

The console output for the above code is:  [engine_thrust_problem](https://drive.google.com/file/d/1vL8o0YPuF_rM4BA8gTd2CTWG2W8FPp1h/view?usp=sharing)

seanmcleod, 2020-06-19:
Instead of 

```python
fdm['fcs/throttle-cmd-norm'] = 0.8
```
try

```python
fdm['fcs/throttle-cmd-norm[-1]'] = 0.8
```

Take a look at the implementation of `SetThrottleCmd`, if you pass -1 then the command will be applied to all engines. Without specifying an index in `fdm['fcs/throttle-cmd-norm']` it defaults to index 0.

https://github.com/JSBSim-Team/jsbsim/blob/15fe952905f9b153d05f5879634717bcac57c063/src/models/FGFCS.cpp#L315-L329

DheerendraTomar, 2020-06-19:
Alright. Thank you.

DheerendraTomar, 2020-06-19:
@seanmcleod It works like charm, like this

`fdm['fcs/throttle-cmd-norm[-1]'] = 0.8`

seanmcleod, 2020-06-27:
@DheerendraTomar do you have enough information now to initialise the simulation with a particular starting state for us to close this issue?

DheerendraTomar, 2020-06-27:
@seanmcleod Yes. I moved ahead with my simulation. Thank you for your help. I will close this issue now and would reopen when I need to ask something else. Apologies for the delay in closing the issue. Thanks

closed 2020-06-27 by DheerendraTomar

bcoconni, 2020-06-27:
@DheerendraTomar 
Thanks for your feedback.

Would you need some further support, please open a new issue rather than reopening this one. There are no limits to the number of issues that can be opened in a GitHub project so let's take advantage of it. It is easier afterwards to reference an issue when it addresses only one topic (see issue #xyz) rather than to specify which part of an issue is relevant for reference (see message abc in issue #xyz).

Thanks in advance.