JSBSim-Team/jsbsim #299: Htailarm for canard before CG and ventral fins at rear

kvnjonathan70, 2020-05-27:
Hello everyone, 
i was wondering how  would Htailarm be specified in aeromatics if we have canard and ventral section. should there be 2 htailarm in `<matrics>` where one is before CG and another after CG. 
also for the canard, should i specify negative length since is before CG.

bcoconni, 2020-05-27:
Hi @kvnjonathan70 ,
> i was wondering how would Htailarm be specified in aeromatics if we have canard and ventral section.

Unfortunately aeromatic is very limited and canard configurations are not managed. It really is a very basic tool to help create a stub model that the user will then enhance.
> should there be 2 htailarm in <matrics> where one is before CG and another after CG.
also for the canard, should i specify negative length since is before CG.

JSBSim does not use the horizontal tail arm beyond providing a few derived quantities:
https://github.com/JSBSim-Team/jsbsim/blob/8c0eb3789f3b1f00d5ca6e3b39de8266b4aaeb09/src/models/FGAircraft.cpp#L154-L162
So whether you need to separate the canard tail arm from the elevator tail arm is really you to decide depending on how you will implement the aerodynamics of your model.

kvnjonathan70, 2020-05-27:
thats  sweet :) i was confused that it might be used for pitch damping calculation
thanks @bcoconni  really appreciate your quick reply to my silly qus with being annoyed :)

by the way i have seen many models specially canard config like saab 37 jsbsim fdm, well its a delta wing with canard. so the modeller has considered entire wing area as tail area but not the canard area,  because the modeller wasnt able to achieve required values. 

in my case i have a ventral fin at rear and canard at front.. so i m confused at which area should i give as htailarea. should i give area of just canard or should i sum all the area of canard and ventral fin ? 

ps : i am able to achieve aerodynamic data of 3 force and 3 moment from cfd.. so i will be putting at values for simulation testing.

bcoconni, 2020-05-27:
Well, normally the tail arm is needed when you have the lift coefficient CLt ('t' for tail) of your elevator while you don't have the pitch moment generated by that very elevator. In that case, you can infer the pitch moment coefficient Cmt ('t' for tail again) from CLt (eqn 6.3,12 from B. Etkin - Dynamics of Atmospheric Flight):
<a href="https://www.codecogs.com/eqnedit.php?latex=C_{m_t}=-\bar{V}_HC_{L_t}&plus;C_{L_t}\frac{S_t}{S}(h-h_{n_{wb}})" target="_blank"><img src="https://latex.codecogs.com/gif.latex?C_{m_t}=-\bar{V}_HC_{L_t}&plus;C_{L_t}\frac{S_t}{S}(h-h_{n_{wb}})" title="C_{m_t}=-\bar{V}_HC_{L_t}+C_{L_t}\frac{S_t}{S}(h-h_{n_{wb}})" /></a>
where `VH` is the tail volume (eqn 6.3,9 from B. Etkin - Dynamics of Atmospheric Flight)
<a href="https://www.codecogs.com/eqnedit.php?latex=\bar{V}_H=\frac{\bar{l}_tS_t}{\bar{c}S}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\bar{V}_H=\frac{\bar{l}_tS_t}{\bar{c}S}" title="\bar{V}_H=\frac{\bar{l}_tS_t}{\bar{c}S}" /></a>
`S` is the main wing area, `St` the tail area, `c` the wing chord and `lt` the horizontal tail arm measured from the wing aerodynamic center to the tail aerodynamic center.
`h` is the distance from the wing leading edge to the CG normalized by the wing chord and `hnwb` the distance from the wing leading edge to wing aerodynamic center.

So I would say that if you get the forces and moments from CFD, you do not need the horizontal tail arm, unless you want to reconstruct the adimensionalized moement coefficients.

kvnjonathan70, 2020-05-27:
CLt and Cmt  is a really good example as it uses vbarh, thanks i understood the concept as where it is used.

> 
So I would say that if you get the forces and moments from CFD, you do not need the horizontal tail arm, unless you want to reconstruct the adimensionalized moement coefficients.

yes you are right, as i am not using vbarh in any function. i would not be needing htailarea and htailarm as i am getting static derivatives from cfd

closed 2020-05-28 by kvnjonathan70