JSBSim-Team/jsbsim #300: moment of inertia calcualtion

kvnjonathan70, 2020-06-04:
Deal all,

My aircraft design is at very initial stage, i have managed to obtained CFD aerodynamics coefficient, i do have structural design in stp format, the weights have not been assigned to the model.  I am filing the mass balance file of jsbsim, and i do not have exact MOI, i checked mass balance header and C++ file, i could find that it is used for structural to body transformation (else i could not understand  much)
My question is do i really need to put values by calculating using Jan Roskam method (Rx, Ry, Rz still couldnt be exact ) or i should leave it blank or zero at this point and obtain its value from output CSV file ?

bcoconni, 2020-06-04:
Hi @kvnjonathan70,

Mass inertia is used for the computation of the rotational accelerations. Basically, if you leave these parameters blank and just specify the entire aircraft weight at the CG, this will result in a 3DoF simulation of your vehicle (X, Y and Z translations only) and JSBSim will not be able to calculate the time evolution of pitch, roll and yaw angles. Even worse, if you apply moments to such an inertia-less vehicle, then the simulation will produce NaNs (because of infinite accelerations) and very likely end in an ugly crash.

Jan Roskam approximations are indeed a good start until you can get final values from the design office. An alternate option is if you have a space distribution of lumped weights. In that case, you can specify the weights and their relative placement to JSBSim which will infer the resulting inertia matrix. Maximum fidelity requires the inertia matrices for all parts but Jan Roskam's methods or the weights distribution can give an acceptable first order approximation.

kvnjonathan70, 2020-06-05:
@bcoconni thank you 😌 you have well explained the physics and its computation in jsbsim. i will go with approximation method as of now, as its an important parameter for 6 dof calculation.

closed 2020-06-05 by kvnjonathan70