JSBSim-Team/jsbsim #302: unwanted spinning of aircraft about CG

kvnjonathan70, 2020-06-11:
hi, i am facing an extreme spin issue, which is getting me sick !
 I have made the aerodynamic file from CFD produced coefficients, CFD output is showing a stable aircraft (negative pitch moment slope). I have taken the same reference point (nose tip) as CFD report and same reference area and length (not wing span and chord length )  for forces and moment calculation (cm is about nose tip).

 when i first tried 'do simple trim' its failed, eventually  i initialized an aircraft it tends to oscillate (pitch up n down) at any velocity n starts spinning. 

real model is ready, i do not wish to change it CG location and even if i do change the CG locations it still oscillates from small amplitude to large and then starts spinning. i have check the sign convention, where normal force is positive and pitch coefficients are negative. from where is the couple generating ? 

From the visualization the force magnitude at the tail look too large, may be that is causing spinning ? 

how can i debug this issue  ? the aircraft is stable !

I also tried to understand working of jsbsim, if i give XCG location from nose tip where nose tip is at 0 (structural frame) then how is jsbsim understanding that my nose tip is at 0,0,0. 

I also tried twice  XCG location (outside the aircraft) ,  now the rotation should be long dueto this new CG point, but still aircraft is spinning as if the CG is inside the aircraft !

seanmcleod, 2020-06-11:
Have you tried plotting all the individual forces and moments to see if there are maybe 1 or 2 that stand out?

If I understand you, you're saying that your AeroRP is the nose tip? JSBSim needs to know the AeroRP and the cg.

kvnjonathan70, 2020-06-12:
Yes plotting all coefficients with respect AOA has been done. There is a positive slope for normal force coefficients    and negative  for pitching moment. Non of them stands out.

Yes AeroRP is kept nose tip which i gave 0,0,0. but how is jsbsim understanding  that its starting point ?
And CG is the actual value. 
In mass balance file i have given mass moments  of inertia  ixx iyy and ixx, then weight(complete) of aircraft and CG location. No pointmasses.

seanmcleod, 2020-06-12:
I meant plotting the individual forces and moments during a run in JSBSim to look at them during the oscillations.

JSBSim just needs the relative offset between AeroRP and cg so that it can transform and apply the forces and moments around the cg.

Take a look at this issue for an example of how we went about debugging the issue in terms of analysing the moments etc. - https://github.com/JSBSim-Team/jsbsim/issues/89

kvnjonathan70, 2020-06-13:
I will check the forces and moments value in a graph n will update you its trends. unfortunately due to my company's policy, i wont be able to show you the plot, also there is no internet out there and i am not allowed to take the file home, thus my replies are quite slow, my apology.

Apart from many **mistakes** in my aerodynamic section (which i am unaware at the moment), **one is the sign convention**, the report which i was referring has body X and Z in opposite direction of what body axis is defined in JSBSim manual. **Now the oscillation has stopped but the `theta` continues to increase and then starts spinning**. 

> JSBSim just needs the relative offset between AeroRP and cg so that it can transform and apply the forces and moments around the cg.

Yes there is offset between AeroRP and CG. 
AeroRP is  at the nose tip and CG is where it has to be, there is a good amount of distance.

 

> Take a look at this issue for an example of how we went about debugging the issue in terms of analysing the moments etc. - #89 

I haven't completely understood the complete debugging (i will go through it again and again) but i would like to inform you that my glider do not have and gear or engine, also i do not have any rate coefficients such as `cmalphadot ` , thus assuming it to be zero at this moment be correct i guess.

seanmcleod, 2020-06-13:
Without being able to share more details in terms of graphs and/or the model itself will make it more difficult for others to help.

My point with the example of issue 89 wasn't that you may be having the exact same issue in terms of the gear etc., rather that with some fairly detailed graphs and analysis of the graphs etc. especially by multiple people that these sorts of issues can be understodd.

> also i do not have any rate coefficients 

You mention `cm-alpha-dot` as an example of one you don't have, do you also not have `cm-q` (pitch rate damping)?

If `theta` just continues to increase then that implies that `cm-de` (elevator power) and/or any pitch disturbance isn't being arrested by `cm-alpha`, `cm-q`, `cm-alpha-dot`.

kvnjonathan70, 2020-06-17:
Sure i will request once again to let me share the graph.

From CFD data right now i have only CX, CY, CZ, Cl, Cm, Cn.

No damping coefficients.

seanmcleod, 2020-06-17:
Which Cm coefficients do you have? Just `Cm-alpha` and `Cm-elevator-deflection`?

kvnjonathan70, 2020-06-24:
Sorry for extreme delay, i was continuously trying to fix it on my own but still failed.

I want to add few things which i found.

**1. All moments coefficients produced from CFD (Cl, Cm, Cn) are about nose  tip
2. All force Coefficients (CN, CA, CS) are at XCP.**

Things i have tried.
All forces and moments uses below property values as CFD coefficients uses the same reference value. 
```
  <property>CFD/ref-area-sqft</property> 
  <property>CFD/ref-length-ft</property>

```
**1. AeroRP = xcp** and Shifted Cm from nose to Xcp by using,
[image]

 I dont know the formula to shift other moments coefficient, Cl and Cn, so i kept both of them same (at nose)

a] In matrics :
AeroRP is kept same as CG in order to coincide CG and RP in body axis .

b] In Aerodynamics 

```
<aero_ref_pt_shift_x>
        <function name="metrics/xcp">
		<quotient>
		<product>
		<table>
             <independentVar lookup="row">aero/alpha-deg</independentVar>
             <independentVar lookup="column">velocities/mach</independentVar>
			 <independentVar lookup="table">aero/beta-deg</independentVar>
<tableData breakPoint="0">
*
**table contains (XCP - XCG)**
*
*
</tableData>
            </table>
	         <v>3.280840</v>
			</product>
			<property>metrics/cbarw-ft</property> 
			</quotient>
        </function>
    </aero_ref_pt_shift_x> 
```

[image]

Result : Brutal spinning

**2] shifted only Cm values from nose to cg** and kept AERORP same as CG
I dont know the formula to shift other moments coefficient, Cl and Cn, so i kept both of them same (at nose).

Result:brutal spinning
[image]

**3] AeroRP = nose tip, ie (0,0,0)**
moment coefficients are kept as it is without any shift.

Result: Brutal spinning 
[image]

Doubt
1] Moments are calculated considering AERORP
 ```
 if (AeroRPShift) vDeltaRP(eX) = AeroRPShift->GetValue()*in.Wingchord;
  vDXYZcg(eX) = in.RPBody(eX) - vDeltaRP(eX); (internally shihted to VdxyzCG)
```
 so does the forces also considers AERORP or it considers CG values defined in matrics ?

2] If aeroRP is used for both forces and moments, where forces are normal, axial and side (coefficient values available at XCP and moments coefficient at nose top ) then how should i correctly provide aeroRP location.

> Which Cm coefficients do you have? Just Cm-alpha and Cm-elevator-deflection?

yes i have the following,
 Cm-alpha and Cm-elevator-deflection 
 CN(normal)-alpha and CN-elevator-deflection 
 CA(axial)-alpha and CA-elevator-deflection, CA-aileron deflection

seanmcleod, 2020-06-24:
Is it not possible to have your CFD use the same AeroRP for both the forces and moments?

Will be tricky to control pitch with only `Cm-alpha` and `Cm-elevator-deflection`.

kvnjonathan70, 2020-06-24:
> Is it not possible to have your CFD use the same AeroRP for both the forces and moments?

Not at this moment, i do not have the CFD case file, i only have the results, so trying to get results   from this resource itself .

> Is it not possible to have your CFD use the same AeroRP for both the forces and moments?
True, but if i add those values by estimation then i wont be able to find out true behavior.

I ran MB339pan without  and damping coefficients, it worked well as desired. 

I found out that in CFD forces are always acting on CP, so the coefficients obtained are always on at Xcp

kvnjonathan70, 2020-06-25:
books i have referred so far provides formula to shift Cm from one reference to another, Is there any formula to shift rolling and yaw moment coefficient from one reference to another ?

agodemar, 2020-06-26:
@kvnjonathan70 
Consider Body axes (B) and Stability axes (S). The S frame is rotated with respect to the B frame by the angle of attack (e.g. alpha_B) around the common pitching axis (y). Hence, you have a transformation matrix T_SB ('from Body to Stability'):

[image]

This means that all vectors represented in the Body frame can be transformed into their representation in the Stability frame. For instance:
[image]

In the above formulas the subscript of each column vector means 'in reference B' or 'in reference S' and the quantities
[image]
are roll, pitch, and yaw components of the resultant external baricentric moment.

kvnjonathan70, 2020-06-26:
@agodemar 
Thanks for the reply, I understood this transformation, but my real doubt still remains unclear :( 
If AeroRP is the point were forces and moments are applied then what AeroRP should i use if available force coefficients are from XCP (different location varying with alpha) and Cm values available at nose

seanmcleod, 2020-06-26:
> When i provide force coefficients in body axis, where are these forces applied in JSBSim?

There are 2 components to the forces you specify. Their location `AeroRP` and the axes you use, one of `wind`, `stability`, `body`.

Take a look at `FGAerodynamics::Run()` - https://github.com/JSBSim-Team/jsbsim/blob/a7574e8b6248bc81fe3e3894902a6b75c4641645/src/models/FGAerodynamics.cpp#L132

First of all the forces are converted to the `body` axis if they aren't already specified in the `body` axis and also into the `wind` axis, see:

https://github.com/JSBSim-Team/jsbsim/blob/a7574e8b6248bc81fe3e3894902a6b75c4641645/src/models/FGAerodynamics.cpp#L205-L230

Then using the forces in the body axis, if there is a difference between the `AeroRP` and the current cg, which is `vDXYZcg` then an additional moment is calculated based on the this arm and the forces.

https://github.com/JSBSim-Team/jsbsim/blob/a7574e8b6248bc81fe3e3894902a6b75c4641645/src/models/FGAerodynamics.cpp#L280

> What is vFnative in aerodynamics.cpp ? does it represents location ion structural axis (aeroRP) ?

`vFnative` are the forces that you specify, again with 2 components, a location which is `AeroRP` and an axis.

seanmcleod, 2020-06-26:
> If AeroRP is the point were forces and moments are applied 

No, `AeroRP` is the reference location for your force and moment data. The forces and moments are applied by JSBSim around the current cg.

So `vDXYZcg` captures the difference between the `AeroRP` and the current cg which is changing all the time with fuel burn etc.

kvnjonathan70, 2020-06-26:
Sir From CFD data we have Resultant normal force coefficient at xcp(calculated from nose tip) for different AOA, for this case, how should i provide aeroRP ? Also there is no fuel system or anything which is responsible to move CG.

seanmcleod, 2020-06-26:
Doesn't the burning of the rocket fuel change the cg? But my main point is that generally the cg and the AeroRP aren't coincident.

agodemar, 2020-06-26:
@kvnjonathan70 Center of pressure is a well understood and elegant concept in aerodynamics, but is not convenient as long as you have to model external aerodynamic forces in flight simulation. That's why you should 'reduce' your aerodynamic resultant to a point which is fixed with the aircraft. This means that, being the reduction point not a center of pressure, you would end up with a resultant force *and* a resultant moment with respect to the reduction point.

If you have several coinditions at several angles of attack, for which you have the centers of pressure and the aerodynamic resultant forces, reducing them to a conventional fixed point is quite easy: You apply the resultant force to the reduction point *and* you associate to that new applied vector the 'transport moment'. The latter is the vector product of the resultant force times the position vector from the generic center of pressure to the reduction point.

kvnjonathan70, 2020-06-26:
@agodemar Thank you for the information, i will try this and inform you the same,
@seanmcleod you i got your point, I will try this and see its behavior

seanmcleod, 2020-06-26:
@kvnjonathan70 just to confirm you have the following data from the CFD right?

|Alpha|Moment about nose MRC|Force at cp|Distance from nose to cp|
|---|---|---|---|
|a1|m1|n1|x1|
|a2|m2|n2|x2|

kvnjonathan70, 2020-06-26:
Yes yes !

agodemar, 2020-06-26:
> @agodemar Sir i understood that i need to bring these force coefficients to one point, but i did not understand this sentence below
>    You apply the resultant force to the reduction point and you associate to that new applied vector the 'transport moment' vector as well. The latter is the vector product of the resultant force times the position vector from the generic center of pressure to the reduction point.

@kvnjonathan70 By definition, the center of pressure is a point where you can concentrate the resultant aerodynamic force and this applied vector *replaces* all the distributed elementary external forces due to superficial pressures and tangential stresses. Onece you apply the resultant force in the center of pressure, by definition, you are done. No moments should be put into the picture.
If I understand what you say, your CFD tool gives you the resultant force, the center of pressure *and* moment components around three axes with origin at nose. 

This should mean that the moments are given by the resultant force times the distance from the center of pressure to the nose. In other words, if you *transport* the resultant from the center of pressure to the nose you must account for those moment components. This means that you can assume your resultant as a force vector applied at the nose. It is what you are looking for: the nose is the Aero-RP that you should define in JSBSim.

seanmcleod, 2020-06-26:
@kvnjonathan70 take a look at the following diagram demonstrating the relationship between the centre of pressure (CP) and the aerodynamic centre (AC).

What's special about the AC is that the moment doesn't change as the AoA changes, it's constant, call it M, and that's because:

L1 * d1 = L2 * d2 = M

[image]

Now assume the point AC is moved to the nose then you will end up with two different moments, M1 and M2 for alpha1 and alpha2, which is what your CFD is outputting in terms of different moments for different alpha values.

seanmcleod, 2020-06-26:
So as @agodemar mentioned set `AeroRP` to be the nose. JSBSim will then use your supplied moment data, e.g. m2 assuming alpha is a2 and this will become `vMomentsMRCBodyXYZ` and to calculate the current moment around the current cg it will then calculate the moment due to the force n2 times the arm from the nose to the current cg `vDXYZcg*vForces` and add them together to get the moment around the current cg.

```c++
vMoments = vMomentsMRCBodyXYZ + vDXYZcg*vForces; // M = r X F 
```

kvnjonathan70, 2020-06-27:
I guess i have understood something wrong, would you mind making me clear about it ?

Below is the example table,
image](https://user-images.githubusercontent.com/59698147/85916873-621be800-b872-11ea-8317-a06a54a9c4db.png)
Normal axial  sign convention used,

[image]

`vMoments = vMomentsMRCBodyXYZ + vDXYZcg*vForces;`
 
I am confused how aeroRP works in this case ?

kvnjonathan70, 2020-06-27:
I have given aeroRP and CG in structural frame as aeroRP = (0,0,0){nose tip}  and CG=(1.9,0,0) respectively. now `in.RPBody(eX)` = 1.9 ? am i correct ?`
`vDXYZcg(eX) = in.RPBody(eX)`

From above table we have force from different Xcp with due to AOA, so how should i give in.RPBody(eX)` for different Xcp ?

kvnjonathan70, 2020-06-27:
I have given aeroRP and CG in structural frame as aeroRP = (0,0,0){nose tip}  and CG=(1.9,0,0) respectively. now `in.RPBody(eX)` = 1.9 ? am i correct ?`
`vDXYZcg(eX) = in.RPBody(eX)`

From above table we have force from different Xcp with due to AOA, so how should i give `in.RPBody(eX)` for different Xcp ?

kvnjonathan70, 2020-06-27:
I have given aeroRP and CG in structural frame as aeroRP = (0,0,0){nose tip}  and CG=(1.9,0,0) respectively. now `in.RPBody(eX)` = 1.9 ? am i correct ?`
`vDXYZcg(eX) = in.RPBody(eX)`

From above table we have force from different Xcp  due to AOA, so how should i give `in.RPBody(eX)` for different Xcp ?

seanmcleod, 2020-06-27:
You don't need to do anything specific regarding Xcp, JSBSim has all the required data, e.g. assume alpha = a2.

```
AeroRP = 0m
cg = 1.9m
```
|Alpha|Moment at nose MRC|Force|
|---|---|---|
|a2|-10290|4900|

Now JSBSim needs to calculate the moment about the cg.

```c++
vMoments = vMomentsMRCBodyXYZ + vDXYZcg*vForces;

vDXYZcg = cg - AeroRP = (1.9 - 0) = 1.9
vMoments = -10290 + 1.9 * 4900 = -980
```

So the moment about the cg is `-980Nm` which JSBSim uses with the moment of inertia to calculate the angular acceleration around the cg. The translational acceleration of the cg is simply the force `4900N` divided by the mass.

In effect the information regarding Xcp and it's movement is encoded in the moment data and force data that the CFD has output. There is no need to give JSBSim any details on Xcp.

kvnjonathan70, 2020-06-27:
@seanmcleod Thankyou so much so much for helping me understand the correct concept. I guess the spinning issue might be due to MOI. I will check it and might reopen the current issue if need 👍 :)
@agodemar I have checked the check the sign convention earlier, it correctly matches with the sign used in CFD report.
Thank you for the knowledge you shared. 👍

seanmcleod, 2020-06-27:
You mentioned only having two pitch moments, `Cm-alpha` and `Cm-elevator-deflection`. So if you log those and the force, alpha and elevator deflection from just before the start of the spin and into the spin you should be able to see what causes the spin.

kvnjonathan70, 2020-07-01:
there  seems to be a different issue, I am not able to identify right now, I will reopen the issue with a solution

closed 2020-07-01 by kvnjonathan70