JSBSim-Team/jsbsim #303: Testing fails on macOS

JetForMe, 2020-06-14:
Hi there. I cloned the repo today and tried to build. It seemed to build okay, but all the tests failed. I notice the tests are using Python 2.7, but when I install cython via Machomebrew, it installed Python 3.8. Not sure if Python 3 is needed for the tests to run.

The way Machomebrew installs some things keeps them hidden from `/usr/local` to avoid interfering with macOS-provided versions of the same packages. Is there a way to tell JSBSim to use a specific Python?

In any case, I've attached the `LastTest.log` file. It also complains about numpy, I'm trying to update that now, but it wants to build gcc from scratch so it'll take a while.

[LastTest.log](https://github.com/JSBSim-Team/jsbsim/files/4776079/LastTest.log)

bcoconni, 2020-06-14:
Hi @JetForMe,

The good news is that JSBSim is using continuous integration on Mac OSX (see for instance https://github.com/JSBSim-Team/jsbsim/runs/710967579?check_suite_focus=true). This means that it is feasible to build and run JSBSim tests on MacOSX.

I am not aware of the particular settings that GitHub is using for their MacOSX platforms but FWIW you can check the way we are building and testing JSBSim on MacOSX from our GitHub scripts (just focus on the `run:` lines and ignore the other lines for a start)
https://github.com/JSBSim-Team/jsbsim/blob/8c0eb3789f3b1f00d5ca6e3b39de8266b4aaeb09/.github/workflows/cpp-python-build.yml#L256-L296

Also from the log you attached, I have seen that you did not install `pandas` which is used by JSBSim tests to check that CSV files are equal to a reference file.

The complete list of Python modules needed for tests is:
* `numpy`
* `pandas`
* `scipy`

Note that most of these modules are only needed for tests. You can use JSBSim without `pandas` and `scipy`.

Also unlike what you are doing, we are installing the Python modules with the `pip` tool instead of using homebrew. Given that PyPI (the platform behind `pip`) is providing binary packages, this might save you the trouble of compiling `numpy` and its siblings yourself. Another option is to install them using [Miniconda](https://docs.conda.io/projects/conda/en/latest/user-guide/install/macos.html).

I hope this helps. If it does not, do not hesitate to give us further details about your problems.

JetForMe, 2020-06-15:
So, my Python was clearly broken. After fixing that up (using pyenv), python runs fine. I built using the instructions, but all the test still fail (they take longer to do so, at least). `ModuleNotFoundError: No module named 'jsbsim'`

This is following these steps (after installing all the python dependencies, including cython):

```
$ mkdir build && cd build
$ cmake ..
$ make
$ ctest
```

Maybe I missed a crucial step, but I don't think so.

The linked yaml file does a lot more to build and test than what is described in the "Developer documentation".

bcoconni, 2020-06-15:
That's weird. Can you check that the Python module is compiled in `build/tests` ? The module is a dynamic library with a name like `jsbsim.cpython-36m-darwin.so` (36 stands for the Python version, 3.6 in this example)

Or may be Mac OSX cannot locate the `JSBSim`library ? Check that you can run the executable `JSBSim` in `build/src` for example with the command `JSBSim --help`. If the executable is unable to locate the dynamic library then you might need to define the environment variable `LD_LIBRARY_PATH`.

JetForMe, 2020-06-16:
Ah, it looks like the python module is not built.

```
4822 17:56: build/ (master) $ cmake ..
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.8.15") found components: doxygen dot 
-- Found PythonInterp: /Users/rmann/.pyenv/shims/python (found version "3.8.2") 
-- Found Cython: /Users/rmann/.pyenv/shims/cython  
-- Found PythonLibs: /usr/local/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7m.dylib (found version "3.7.7") 
Notice: Output directory `documentation' does not exist. I have created it for you.
Parsing layout file /Users/rmann/Projects/Personal/RocketSim/jsbsim/doc/DoxygenLayout.xml...
Searching for include files...
Searching for example files...
Searching for images...
Searching for dot files...
Searching for msc files...
Searching for dia files...
Searching for files to exclude
Searching INPUT for files to process...
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/models
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/initialization
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/input_output
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/models/flight_control
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/models/atmosphere
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/models/propulsion
Searching for files in directory /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/math
Reading and parsing tag files
Parsing files
Preprocessing /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/FGFDMExec.cpp...
Parsing file /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/FGFDMExec.cpp...
Preprocessing /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/FGFDMExec.h...
Parsing file /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/FGFDMExec.h...
Preprocessing /Users/rmann/Projects/Personal/RocketSim/jsbsim/src/FGJSBBase.cpp...
...
Patching output file 120/121
Patching output file 121/121
lookup cache used 17278/65536 hits=285703 misses=18190
finished...
-- Could NOT find CxxTest (missing: CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE) 
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Could NOT find LSB_Release (missing: LSB_RELEASE_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rmann/Projects/Personal/RocketSim/jsbsim/build
```

bcoconni, 2020-06-20:
There seems to be some mismatch in the versions. CMake reports having found a Python interpreter 3.8.2 but Python libraries are 3.7.7. Might be the cause of your problem ?
```
-- Found PythonInterp: /Users/rmann/.pyenv/shims/python (found version "3.8.2") 
-- Found Cython: /Users/rmann/.pyenv/shims/cython  
-- Found PythonLibs: /usr/local/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7m.dylib (found version "3.7.7")
```

bcoconni, 2020-10-19:
@JetForMe Has your problem been fixed ? Or do you need further support ?

JetForMe, 2020-10-19:
I have not been able to fix it, but I also haven't been able to look much more at it; too many other things got in the way.

bcoconni, 2020-11-02:
@JetForMe No problem. So I suggest we close this issue.
You can reopen it (or open a new one) as soon as you are able to investigate that issue again.
Thanks.

closed 2020-11-02 by bcoconni