JSBSim-Team/jsbsim #305: Aircraft falling down despite all correct setting

DheerendraTomar, 2020-06-30:
Hi all,

I am not able to understand why the aircraft is falling down from sky. I am taking control inputs from joystick (which is working fine). I am visualising the simulation in FlightGear. I am attaching my python code below. Also, I am not able to figure out which is the variable to monitor vertical velocity of the aircraft. 

`import jsbsim
    import time

    fdm = jsbsim.FGFDMExec('C:\JSBSim', None)
    fdm.load_model('A320')
    fdm.load_ic('basic_ic.xml', True)
    fdm['propulsion/set-running'] = -1
    fdm['ic/alpha-deg'] = 3.2758
    fdm['ic/theta-deg'] = 3.2787
    fdm['ic/psi-true-deg'] = 307
    fdm['gear/gear-cmd-norm'] = 0
    fdm['ic/q-rad_sec'] = 0.0
    fdm['ic/p-rad_sec'] = 0.0
    fdm['ic/r-rad_sec'] = 0.0
    fdm['ic/roc-fps'] = 0.00001
    fdm['fcs/pitch-trim-cmd-norm'] = -0.2
    fdm['fcs/roll-trim-cmd-norm'] = 0.0
    fdm['fcs/yaw-trim-cmd-norm'] = 0.0
    fdm['fcs/flap-cmd-norm'] = 0
    fdm['fcs/elevator-cmd-norm'] = 0.0
    fdm['fcs/roll-cmd-norm'] = 0.0
    fdm['fcs/aileron-cmd-norm'] = 0.0
    fdm['fcs/throttle-cmd-norm'[-1]] = 0.8
    fdm['fcs/mixture-cmd-norm'[-1]] = 0.8
    fdm['simulation/dt'] = 0.017
    fdm.run_ic()
    fdm.set_output_directive('data_output/flightgear.xml' )
    

    fdm.print_simulation_configuration()
    frame_duration = fdm.get_delta_t()
    sleep_nseconds = (frame_duration if True else sleep_period) * 1E9
    current_seconds = initial_seconds = time.time()
    result = fdm.run()

    print('*'*8 + 'Initial Conditions' + '*'*8 )
    print(fdm['ic/vc-kts'])
    print(fdm['ic/h-sl-ft'])
    print(fdm['ic/u-fps'])
    print(fdm['ic/p-rad_sec'])
    print(fdm['ic/q-rad_sec'])
    print(fdm['ic/r-rad_sec'])
    print(fdm['ic/roc-fps'])
    print(fdm['ic/alpha-deg'])
    print(fdm['ic/theta-deg'])

    while result and fdm.get_sim_time() <= 120.0:
        current_seconds = time.time()
        actual_elapsed_time = current_seconds - initial_seconds
        sim_lag_time = actual_elapsed_time - fdm.get_sim_time()
        for _ in range(int(sim_lag_time / frame_duration)):
            elevator_cmd, roll_cmd, aileron_cmd, throttle_cmd = read_joystick()
            fdm['fcs/elevator-cmd-norm'] = elevator_cmd
            fdm['fcs/roll-cmd-norm'] = roll_cmd
            fdm['fcs/aileron-cmd-norm'] = aileron_cmd
            fdm['fcs/throttle-cmd-norm'[-1]] = throttle_cmd
            fdm['fcs/mixture-cmd-norm'[-1]] = throttle_cmd
            result = fdm.run()
            print('*'*8 + 'Current Conditions' + '*'*8 )
            print('Current Alpha: %f'%(fdm['aero/alpha-deg']))
            print('Current Altitude: %f'%(fdm['position/h-sl-ft']))
            print('Current speed: %f'%(fdm['velocities/vc-kts']))
            print('Pitch degree: %f'%(fdm['attitude/pitch-rad']*57.2958))
            print('Roll degree: %f'%(fdm['attitude/roll-rad']*57.2958))
            print('Heading degree: %f'%(fdm['attitude/psi-rad']*57.2958))
            print('Engine 1 N2: %f'%(fdm['propulsion/engine[0]/n2']))
            print('Engine 2 N2: %f'%(fdm['propulsion/engine[1]/n2']))
            print('Engine 1 thrust-lbs: %f'%(fdm['propulsion/engine[0]/thrust-lbs']))
            print('Engine 2 thrust-lbs: %f'%(fdm['propulsion/engine[1]/thrust-lbs']))
            current_seconds = time.time()

`

The aircraft is initialsed using below configuration:

`<?xml version="1.0"?>

    <initialize name="basic_ic">
    <ubody unit="FT/SEC">  421.952  </ubody> 
    <vbody unit="FT/SEC">    0.0  </vbody>
    <wbody unit="FT/SEC">    0.0  </wbody>
    <longitude unit="DEG">   14.093  </longitude> 
    <latitude unit="DEG">   36.16  </latitude>
    <phi unit="DEG">         0.0  </phi>
    <theta unit="DEG">       3.2  </theta>
    <psi unit="DEG">         0.0  </psi>
    <altitude unit="FT">     5000 </altitude>
    <running>               -1 </running> 
   </initialize>`

I have attached the snapshot of the a basic GUI(for illustration purpose only) of for monitoring the variables as well. I am attaching the picture below. 

[image] 

I am attaching a snapshot of the command line to which prints basic configuration of the aircraft. I don't know what is wrong which my code or settings. Why the aircraft is falling down?

[image]

seanmcleod, 2020-06-30:
But your throttle value in your GUI screenshot is 0 and sure enough the thrust being generated by the engines is also pretty much 0 in your GUI screenshot.

DheerendraTomar, 2020-06-30:
That is an old screenshot (I mentioned that screenshot is for illustration purpose only). If you want I can share a new screenshot.

seanmcleod, 2020-06-30:
Without actual data output, e.g. graphing the control positions versus time plus a bunch of useful states like alpha, theta, thrust etc. versus time it's very difficult to help debug.

But a couple of random things I noticed.

Any reason for not using the default `dt` - as opposed to `fdm['simulation/dt'] = 0.017` (58.8Hz)?

Also this looks incorrect: `fdm['fcs/throttle-cmd-norm'[-1]] = throttle_cmd`, it should be `fdm['fcs/throttle-cmd-norm[-1]'] = throttle_cmd`

Your version is the equivalent of: `fdm[m] = throttle_cmd`. You have the same error as part of setting your initial conditions, so I'm guessing you're ending up with no thrust/idle thrust.

No need for setting the mixture for a turbofan `fdm['fcs/throttle-cmd-norm'[-1]] = throttle_cmd`

DheerendraTomar, 2020-06-30:
Thanks you. I will share the graphs tomorrow.

1. The frequency part is a rounding error. It was supposed to be 60Hz. I have fixed it.

2. You have guessed it correct. I am ending up with no thrust/idle thrust. I am using `fdm['fcs/throttle-cmd-norm'[-1]] = throttle_cmd` because it doesn't work like  `fdm['fcs/throttle-cmd-norm[-1]'] = throttle_cmd`. I get an error(uploaded the screenshot with this comment). Any idea why the thrust command doesn't work your way.
[image]

3. Removed the mixture command.

seanmcleod, 2020-06-30:
Any reason not to go with the default frequency of 120Hz which you get if you don't explicitly set `dt`?

Rather paste console output as text as opposed to as a bitmap, e.g. like the following. Makes it indexable for searching etc.

```python
>>> fdm['fcs/throttle-cmd-norm[-1]'] = 0.8
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "jsbsim.pyx", line 998, in jsbsim.FGFDMExec.__setitem__
  File "jsbsim.pyx", line 1148, in jsbsim.FGFDMExec.set_property_value
RuntimeError: unterminated index (looking for ']')
```

I thought the python wrapper would end up passing in -1 for the engine number to `SetThtottleCmd()`, but it must be trying to parse the indexed property name and getting confused.

https://github.com/JSBSim-Team/jsbsim/blob/15fe952905f9b153d05f5879634717bcac57c063/src/models/FGFCS.cpp#L315-L329

Remember the -1 idea is simply a shortcut to apply the throttle value to all engines, particularly useful if the script code doesn't care about the number of engines etc.

But in your case for now since you know you're dealing with a 2 engine aircraft simply pass the throttle command to both engines, i.e.

```python
fdm['fcs/throttle-cmd-norm[0]'] = joystick_throttle
fdm['fcs/throttle-cmd-norm[1]'] = joystick_throttle
```

DheerendraTomar, 2020-06-30:
The thrust part is solved. I used your method to set throttle command to each engine. The frequency is set to 60 Hz to keep the file size small and I believe it will affect the learning of reinforcement agent(which will interact with the simulator at less frequency the simulator frequency) in a good way. I will paste the command line output as mentioned by you.  

I will keep the thread open for now as I will post the graphs tomorrow to seek your advice. Many thanks

DheerendraTomar, 2020-07-01:
@seanmcleod These are the graphs before I made the correction suggested by you. Pleas note that I was trying to stall the aircraft that's why 0 thrust is there. But, idle thrust is 32%, which was not working because of the wrong command for throttle (which is fixed now.) Everything is working as expected now. 

[image]
[image]
[image]

The image below is stick input to control the pitch.
[image]
[image]
[image]
[image]
[image]

Which airbus a320 model do you suggest to simulate aircraft stall more accurately than this? Presently I am using the A320 model provided by JSBSim.

seanmcleod, 2020-07-01:
If you want to accurately simulate the aircraft near, at and possibly just past the stall then you need to find a model which has accurate data for that region, i.e. accurate data for things like the coefficient of lift for those alpha angles, and coefficient of drag around those angles, `Cm-alpha` around those alpha angles etc.

A lot of models won't have accurate data around that region. I'm not familiar with the data for the A320 model that comes with JSBSim.

DheerendraTomar, 2020-07-01:
Okay. Thank you so much.

closed 2020-07-01 by DheerendraTomar