JSBSim-Team/jsbsim #306: Setting/updating atmospheric properties

benbusby, 2020-07-01:
Hi,

I've been working on developing a good way to dynamically set and update atmospheric properties through the Python API, but have been running into a couple of issues. 

It seems that at least some of the documented atmospheric properties in JSBSim (see [here](https://github.com/bstaero/jsbsim/blob/master/properties.md#atmosphere)) might be immutable, since attempts to set new a value for atmospheric properties don't seem to actually update the value. Here's an excerpt of how I'm attempting to set temperature, for example:

```python
self.flight_model.set_property_value('atmosphere/T-R', 540.27)
print(self.flight_model.get_property_value('atmosphere/T-R')

# Prints '518.655...'
```

Are these atmospheric properties intended to accept user modification or are they meant to be immutable? 

In addition, I'd also like to be able to set more properties than what I've come across in the [sourceforge documentation](http://jsbsim.sourceforge.net/JSBSim/classJSBSim_1_1FGAtmosphere.html), namely things like gravity, wind speed, etc, and eventually integrate global reference atmospheric models like the Venus-GRAM into my use case, but the properties listed don't seem to include these. Is there a separate reference for atmospheric properties that I'm overlooking or is this the full list?

Thanks again for all of your work on this project!

seanmcleod, 2020-07-01:
Take a look at this set of properties, which all have setters and getters.

https://github.com/JSBSim-Team/jsbsim/blob/76411ec4d0a095ca031da320c466db3561c31267/src/models/atmosphere/FGStandardAtmosphere.cpp#L671-L698

Then the following set only have getters.

https://github.com/JSBSim-Team/jsbsim/blob/76411ec4d0a095ca031da320c466db3561c31267/src/models/FGAtmosphere.cpp#L271-L286

benbusby, 2020-07-01:
Thanks for the quick response! So despite `FGAtmosphere` properties not having setters, could I not just use the `atmosphere/override/<property name>` nodes to achieve the same effect? I'm referencing this section (although it still looks like a lot of the control I was hoping for is not available):

https://github.com/JSBSim-Team/jsbsim/blob/76411ec4d0a095ca031da320c466db3561c31267/src/models/FGAtmosphere.cpp#L107-L131

seanmcleod, 2020-07-02:
I guess so, I didn't spot those. I've only used `atmosphere/delta-T` in the past.

If they still don't offer the level of control you require then I guess your other option is to implement your own atmosphere model. In addition to the standard atmosphere model you'll see there are 2 other atmosphere models currently in JSBSim, MSIS and Mars.

bcoconni, 2020-07-02:
Properties `atmosphere/override/*` are meant to bypass the ISA1976 atmospheric model, rather than letting JSBSim compute the pressure and temperature from the altitude and the atmospheric properties at sea level. When using these properties, JSBSim assumes your program is taking over the atmospheric properties computations.

As @seanmcleod mentioned, the temperature can be modified by the following properties:
* property `atmosphere/delta-T`  which basically shifts the temperature by the amount specified (in Rankines)
* property `atmosphere/SL-graded-delta-T` which modifies the slope of the temperature as a function of altitude.

When fed with these 2 properties, JSBSim assumes the ISA1976 atmospheric model must be used to assess the pressure and temperature.

closed 2020-12-11 by bcoconni