JSBSim-Team/jsbsim #307: Is it possible to use other aircrafts (not the one supplied with JSBSim)?

DheerendraTomar, 2020-07-01:
Hi all,

I was wondering if we can use other aircrafts (meant to used with FlighGear) with JSBSim? The family of aircraft I am referring to is [Airbus A320](https://github.com/legoboyvdlp/A320-family). It has all the definitions in XML only. 

Thanks and regards,
Dheerendra

seanmcleod, 2020-07-02:
Yes you can in general as long as the aircraft in question uses the JSBSim FDM as opposed to one of the other FDMs supported by FlightGear, see - http://wiki.flightgear.org/Flight_Dynamics_Model

In terms of the particular A320 model you mention you need to make sure you understand what it implements in terms of it's FCS. 

Are you familiar with FBW in general? 

So for example in the A320 in up and away in normal mode the stick position doesn't directly correlate to elevator position in pitch. Rather the stick demands a particular g that the pilot wants, from -1g to +2.5g from full forward stick to full aft stick. So there is a control system which compares the demanded g to the actual g and will drive the elevator and horizontal stabilizer to match the demanded g.

Which may or may not be what you're wanting or expecting for your re-enforcement learning.

Secondly the Airbus FBW system also implements protections, which may or may not be modelled in the particular A320 JSBSim model. For example under normal law it will prevent the pilot from stalling the aircraft.

Again this may or may not be what you want for your particular use.

DheerendraTomar, 2020-07-02:
Hi,
Thank you for the information. I opened an issue and spoke with the developers of [Airbus A320 family](https://github.com/legoboyvdlp/A320-family/issues/162). They have done a great job. The family is meant for FlightGear but since FlightGear uses JSBSim in the background, I thought it would work. The developers of the family told me that they tried to run it and ran into some problems with nasal files (they told that a lot of legacy code of fly-by-wire needs to be rewritten in nasal). The family support stall warning (based on alpha) and different controls laws( normal/alternate/direct) for now but they are improving the stall simulation to that of real world. 

The FlyByWire example you gave further enhances my understanding of FBW systems. 

I have two questions: 
1. If I want to improve the A320 model which comes with JSBSim, how do I start? You mentioned in [this issue](https://github.com/JSBSim-Team/jsbsim/issues/305#issuecomment-652441274) about the `Cm-alpha` values. Is it a good idea to get these values from this [Airbus A320 family](https://github.com/legoboyvdlp/A320-family/issues/162)

2. The current A320 model with JSBSim doesn't implement FBW and operates in direct law, right?

DheerendraTomar, 2020-07-02:
Hi,

I tried to run the Airbus family mentioned in the last comment. I got the flybywire error: 
`FGPropertyManager::GetNode() No node found for /it-fbw/law
Traceback (most recent call last):
  File "environment_experiment.py", line 29, in <module>
    fdm.run_ic()
  File "jsbsim.pyx", line 1012, in jsbsim.FGFDMExec.run_ic
RuntimeError: FGPropertyValue::GetValue() The property /it-fbw/law does not exist.
FGPropertyManager::GetNode() No node found for /engines/engine[0]/n1
Caught error: FGPropertyValue::GetValue() The property /engines/engine[0]/n1 does not exist.`

Any idea to fix this issue?

seanmcleod, 2020-07-02:
If you want to improve the aerodynamic data for the base A320 model in JSBSim then you need to first find better/more accurate data and make sure it covers the full range that you need, e.g. if you want to go past the critical/stall angle of attack then make sure the new data covers that range.

I can't say off-hand if the other A320 model has better aerodynamic data. I did see a comment from the developer in the issue you opened mentioning that he is working on a new aerodynamic model which will have data for stalling.

> I'm working on this in the new aero model. It will take some time to be correct, but it will be done. The current one did not take any stall into account.

So it sounds like you'd need to wait for that.

Then secondly I don't have enough information on what you're trying to achieve with your re-enforcement learning to know whether you want your code to be passing in inceptor commands that directly control the control surfaces, or whether you want to pass in a command law value.

That will determine whether you want or need their FBW command law implementation or not.

Do you want your re-enforcement learning setup to actually have the aircraft stall and then try and recover from the stall? Or do you just need the aircraft to approach the stall?

DheerendraTomar, 2020-07-02:
Alright! thanks for the explanation. Any links to help getting started with aerodynamic data for A320 model. The goal of re-enforcement learning environment is to have a near real world A320 model which means it will have to have FBW system. I will just pass in the side stick input like pilots do which will be processed by FCS. 

For now, I am focusing on recovering from a stall.

seanmcleod, 2020-07-02:
I don't have specific links to aerodynamic data for the A320.

Just remember though in a real-world A320 FBW system in normal law mode the aircraft won't actually stall, so if you want to actually recover from a stall you'll need to be in either alternate mode law or direct mode law.

Take a look at - http://www.airbusdriver.net/airbus_fltlaws.htm

DheerendraTomar, 2020-07-02:
Yes! I know about the different control laws. I used to turn off ADR systems in X-Plane to get into alternate law before using JSBSim.

DheerendraTomar, 2020-07-02:
Just curious, if it is possible to use this [Airbus A320 family](https://github.com/legoboyvdlp/A320-family) without FBW? I mean it would be then similar to flying in a direct law, right?

seanmcleod, 2020-07-02:
You could just use their aerodynamic data if you know it's currently better that the aerodynamic data for the model included with JSBSim and use the basic FCS (just direct connection between inceptor and control surface) that is in the model included with JSBSim.

Otherwise you'd need to ask the A320 developers if they have an option to disable FBW completely.

There is one slight difference in the Airbus direct law and that is:

> Control sensitivity depends on airspeed

DheerendraTomar, 2020-07-02:
Alright! Thank you so much @seanmcleod

bcoconni, 2020-07-02:
Hi @DheerendraTomar ,

Just to clarify a couple of things: Nasal is a built-in script language of FlightGear which is not supported by JSBSim. Authors of aircraft models for FlightGear tend to use massively Nasal instead of JSBSim's FCS because XML is not a very convenient language to implement complex logics. As a result, more than often aircraft models specifically designed for FlightGear cannot be run by JSBSim stand alone.

Some authors are taking extra care to make sure their FlightGear models can be run with JSBSim stand alone but that is obviously not the case for the [A320 family models](https://github.com/legoboyvdlp/A320-family) since the authors explicitly mentioned that they are using Nasal. That is most likely the reason why you got the error about the missing node `/it-fbw/law`.

The bottom line is that you will have to code the FCS from the "A320 family" models yourself in Python since there is no tool for automatic translation from Nasal to Python. Alternatively, as @seanmcleod mentioned, you can cherry pick some elements from their model such as the aerodynamics, etc. and use them in your model.

DheerendraTomar, 2020-07-03:
Hi **@bcoconni**,
Thank you for your clarification. Thinks are getting clear day by day. Presently, I am picking aerodynamics data from [A320 family models](https://github.com/legoboyvdlp/A320-family) and will update about the outcome of the results on this thread later.

Any links or getting started tutorial to write FCS for an aircraft.

Thanks and regards,

DheerendraTomar, 2020-07-03:
I found out this tool named as [DATCOM](http://wiki.flightgear.org/DATCOM) which states that it generates aerodynamic data for an aircraft provided that you enter the dimensions of the aircraft. It also generates files to be used by JSBSim. Is this the right thing to move ahead ?

seanmcleod, 2020-07-03:
@DheerendraTomar can you share more details on your requirements and goals for your reinforcement learning using JSBSim. That will make it easier to make suggestions in terms of FCS development, assuming it's actually required, what level of fidelity you need in terms of aerodynamic data etc.

DheerendraTomar, 2020-07-03:
@seanmcleod In terms of fidelity of aerodynamic data, this model [FF A320 Ultimate](https://store.x-plane.org/A320-Ultimate_p_688.html#tab-1) is really good. It is able to simulate stall very precisely. For example, The above mentioned model (In alternate law) at 5000 feet with idle thrust was stalling at around 16 degree AoA and airspeed was around 120-130 knots. 

I would go for a **low or medium** fidelity model. I don't really require FBW, but If it can be done easily then It will be great. But, It will have a stall warning based on high alpha. 

The RL environment will be like: The aircraft will be initialised in a stalled state and then algorithms will recover from stall and will also bring the aircraft to level flight at the altitude at which the aircraft was stalled.

seanmcleod, 2020-07-03:
The difficulty with modeling a stall isn't in simply getting the stall to start at the correct AoA with the correct Qbar. That's the easy part. However if you want your system to handle recovering somewhere in the post stall area then that's where it's a lot more difficult to get good fidelity aerodynamic data.

For example take a look at some of these links.

[Simulation Study of a Commercial Transport Airplane During Stall and Post-Stall Flight](https://www.jstor.org/stable/44738024?seq=1)

[Representative Post-Stall Modeling of T-tailed Regional
Jets and Turboprops for Upset Recovery Training](https://tspace.library.utoronto.ca/bitstream/1807/72860/1/Teng_Tianhang_201606_MAS_thesis.pdf)

I'd suggest starting out without FBW.

DheerendraTomar, 2020-07-03:
Thank you so much. I will read all this and get back to you.

DheerendraTomar, 2020-07-14:
I read the papers. But, with the little knowledge I have in this aviation domain. It is hard for me to use these papers. I will put my efforts to make this [Airbus Family](https://github.com/legoboyvdlp/A320-family) work in JSBSim. Although, I disabled the FBW section and made some changes in FCS but still there are many parameters which are creating problems. Thanks @seanmcleod for your help.

seanmcleod, 2020-07-14:
It'll probably be easier to copy the aerodynamic data from https://github.com/legoboyvdlp/A320-family into the existing JSBSim A320 model.

DheerendraTomar, 2020-07-14:
Already tried that but the model behaved weirdly. I think it is because of that model have many features like, lift due to slats/spoilers etc. I tried to copy only those parameters which we have but still didn't work :(

jonsberndt, 2020-08-24:
> @jonsberndt <https://github.com/jonsberndt>  I initially noticed this slide during one of the snippets of the video from the final day that I watched.
>
>  <https://user-images.githubusercontent.com/824856/90973898-83bde680-e526-11ea-8951-3b42ffe668cf.png>
>
> Also during the Q&A with the winning team they mentioned JSBSim a number of times, although just in passing.
>
> Also the DARPA news report mentions JSBSim - https://www.darpa.mil/news-events/2019-10-21
>
> All eight teams will compete in each of the three scheduled trials. They will use the JSBSim open-source flight dynamics model for simulation environment
>
> They also mention the use of the F-15C aircraft from FlightGear. Although during the video they mention they switched from the F-15 to an F-16 without mentioning any specific reasons.
>
> The simulated aircraft for the Trials is the publically available, unclassified FlightGear version of an F-15C fighter aircraft.

This would be a cool story to add to the JSBSim Facebook page. But, I don’t know enough about it to make a decent post. Anyone interested in writing up a brief paragraph about this? Than I can post it.

 

JB

bcoconni, 2020-08-25:
I do not know the details about this test but I watched [this Youtube video from a former USAF fighter pilot](https://youtu.be/ziCQqmEllZo) and that was quite interesting as he explained why the AI performance is really a great achievement but still there is a long way to go before fighter pilots will be replaced by AIs. He really makes some good points, in particular about the rules of engagement which were simplified for the AI (complete 360° awareness of the opponent position, velocity, etc. - actually it should be 720° because it is a [solid angle](https://en.wikipedia.org/wiki/Solid_angle) but I'm quibbling 😄).

seanmcleod, 2020-09-30:
@DheerendraTomar can we close this issue?

DheerendraTomar, 2020-09-30:
Hi @seanmcleod,
 I would like to leave it opened for now. Just in case, if someone is interested to work on nasal to python script conversion. Also, I have to write back to you. I will do it soon :)
Thanks.

bcoconni, 2020-10-02:
Hi @DheerendraTomar,
>  I would like to leave it opened for now. Just in case, if someone is interested to work on nasal to python script conversion.

I don't think this is a good enough reason to keep this issue open. Your problem is addressed so I am closing this issue. Would you need some further support, you can open a new issue and we will happily respond to your questions.

Thanks.

closed 2020-10-02 by bcoconni