JSBSim-Team/jsbsim #309: Possible iterator use after call of erase on element

sthagen, 2020-08-04:
In `jsbsim/src/initialization/FGTrimAnalysis.cpp` the iterator `iControls` may be used after calling `erase`on the element. `cppcheck 1.82` warns on that one ... but maybe this is not a problem?

```cpp
bool FGTrimAnalysis::RemoveControl( TaControl control ) {

  FGTrimAnalysisControl* tac;
  bool result=false;

  mode = taCustom;
  vector <FGTrimAnalysisControl*>::iterator iControls = vTrimAnalysisControls.begin();
  while (iControls != vTrimAnalysisControls.end()) {
      tac=*iControls;
      if( tac->GetControlType() == control ) {
        delete tac;
        vTrimAnalysisControls.erase(iControls);  // conditional erase here
        result=true;
        continue;
      }
      iControls++;  // iterator use here
  }

  return result;
}
```

seanmcleod, 2020-08-04:
Yes, the iterator `iControls` needs to be updated with the return value from `erase()`, and the subsequent `iControls++` needs to be in an `else` statement, e.g.

```cpp
for ( ; it != res.end(); ) {
  if (condition) {
    it = res.erase(it);
  } else {
    ++it;
  }
}
```

However, it turns out that `FGTrimAnalysis.cpp` isn't currently built into JSBSim, see:

https://github.com/JSBSim-Team/jsbsim/blob/a9fb5200c176953003253344f3ae5274897d4f6b/src/initialization/CMakeLists.txt#L1-L14

bcoconni, 2020-08-05:
Another possible option since we are using C++11 would be to use `std::remove_if`
```c++
  iControls.erase(std::remove_if(iControls.begin(), iControls.erase(),
                                 [control](FGTrimAnalysisControl* tac){ return tac->GetControlType() == control; }),
                  iControls.end());
```
This is untested code :wink: 
> However, it turns out that `FGTrimAnalysis.cpp` isn't currently built into JSBSim

Indeed, however experience taught us that disabled code might not remain disabled forever (see issue #227 in which we are considering to re-enable parts of `FGLinerization.cpp`). I would then suggest to alter the code so that compilation fails when (not if :wink:) someone will re-activate the code, using the `#error` pragma:

```c++
  #error TODO: Iterator is used after erase(). See GH issue #309

  vector <FGTrimAnalysisControl*>::iterator iControls = vTrimAnalysisControls.begin();
  while (iControls != vTrimAnalysisControls.end()) {
```

seanmcleod, 2020-08-05:
Yep, I wasn't clear enough in my comment. I wasn't suggesting we ignore the issue, it was more a reply to why we don't see any crashes in JSBSim given this bug for the question - "but maybe this is not a problem?"

I mentioned needing to put the iterator increment `iControls++` inside an `else` statement, but I didn't notice the `continue` statement in the original code, so we don't need to, we could just update the iterator with the return value from `erase()`.

```cpp
iControls = vTrimAnalysisControls.erase(iControls);
```

Looks like a typo in the second argument to `std::remove_if`, should be `iControls.end()` instead?

In the `std::remove_if` case we would need to add some additional logic in order to be able to return `true` if one or more elements were removed to match the original code.

bcoconni, 2020-08-05:
> Yep, I wasn't clear enough in my comment. I wasn't suggesting we ignore the issue, it was more a reply to why we don't see any crashes in JSBSim given this bug for the question - "but maybe this is not a problem?"

OK. In addition, since the code is currently not compiled, we have no way to test that the discussed modifications are actually fixing the problem. So I am suggesting to insert an `#error` pragma and hold up the bug fix until the code is re-activated in JSBSim.
> Looks like a typo in the second argument to `std::remove_if`, should be `iControls.end()` instead?

Oops, you are right it should indeed be:
```c++
  iControls.erase(std::remove_if(iControls.begin(), iControls.end(),
                                 [control](FGTrimAnalysisControl* tac){ return tac->GetControlType() == control; }),
                  iControls.end());
```
It's the reason why I wrote a disclaimer about it being untested code :smile:. I have added this option because I have a bias towards lesser lines of code. Not sure if that would be an improvement to the code legibility however ?!? 
> In the `std::remove_if` case we would need to add some additional logic in order to be able to return `true` if one or more elements were removed to match the original code.

The return value could easily be computed by comparing the vector length before and after running `erase()`.

seanmcleod, 2020-08-05:
One other thing, the use of naked pointers `FGTrimAnalysisControl*` means that the 1 line erase-remove idiom won't call the destructor for `FGTrimAnalysisControl` and free the memory etc.

bcoconni, 2020-08-05:
> One other thing, the use of naked pointers `FGTrimAnalysisControl*` means that the 1 line erase-remove idiom won't call the destructor for `FGTrimAnalysisControl` and free the memory etc.

Yes you are correct. Indeed the vector `iControls` should be modified to use smart pointers before using `std::remove_if`. Your proposal is definitely much simpler than mine. But I am still in favour of using `#error` pragmas. What about both of you ?

seanmcleod, 2020-08-05:
I'm happy with your proposal of adding the `#error` pragma with a reference to this issue. Even though it's in theory a trivial 1 line fix we're not actually compiling the code and having it pass any of our tests etc. so let's add the `#error` and if we do ever end up adding the source file back we can fix it then.

PR #313 (2020-08-06): Implemented iterator update after erase (suggested by Sean McLeod)
Details at #309 discussion esp. [latest comment from Sean](https://github.com/JSBSim-Team/jsbsim/issues/309#issuecomment-669171296)

Closes #309
closed 2020-08-06 by bcoconni